jasonacox/Powerwall-Dashboard #2: Generating and String Power do Not Equal

cwagz, 2022-02-19:
This is a really neat tool. Thank you for making it. I received PTO last week and wanted to see more about how my system was setup. I have 2 powerwall+ each with a 7.6kW inverter. I have three arrays and three strings. I am supposed to have a 12.4kW system but I have not seen more than 8kW yet.

The problem I am seeing is that if I add the power value for string A and string B I get 5,091W. But my system shows that it is generating 6.09kW on the dashboard and in the app. I am pretty sure the dashboard is not pulling data for my third string. Is this something I need to fix in a config file?

jasonacox, 2022-02-20:
Hi @cwagz - thanks for opening the issue!  And, congrats on getting to PTO!  I'm in Los Angeles and our 8.5kW system is just now starting to reach 5.9kW peak.   It has been steadily growing since the middle of December as it should.   There are a lot of variables (angle and direction of panels) but I suspect you will see higher output as we get to summer.  

As to the mismatch on the String vs. Total output, I think I know what is happening.  With two inverters, you likely have two sets of strings and the current dashboard.json setup only includes one set (easy to fix which I'll show below).  Before we do that, can we confirm what the Powerwall is reporting via device vitals?

Can you paste your draw string data?  Here is how (change "localhost" to the computer that you are running the docker containers on): http://localhost:8675/strings

It will look something like this:
```json
{
    "A": {
        "Connected": true,
        "Current": 2.72,
        "Power": 635.0,
        "State": "PV_Active",
        "Voltage": 235.5
    },
    "B": {
        "Connected": false,
        "Current": 0.01,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -1.799999999999999
    },
    "C": {
        "Connected": true,
        "Current": 4.6000000000000005,
        "Power": 1206.0,
        "State": "PV_Active",
        "Voltage": 264.2
    },
    "D": {
        "Connected": true,
        "Current": 4.53,
        "Power": 1193.0,
        "State": "PV_Active_Parallel",
        "Voltage": 264.40000000000003
    }
}
```

And also look at the raw data here:  http://localhost:8675/vitals

Look for these sections:

```json
        "PVAC_PVCurrent_A": 2.22,
        "PVAC_PVCurrent_B": 0.02,
        "PVAC_PVCurrent_C": 3.6,
        "PVAC_PVCurrent_D": 3.5500000000000003,
        "PVAC_PVMeasuredPower_A": 577.0,
        "PVAC_PVMeasuredPower_B": 0.0,
        "PVAC_PVMeasuredPower_C": 1067.0,
        "PVAC_PVMeasuredPower_D": 1053.0,
        "PVAC_PVMeasuredVoltage_A": 236.5,
        "PVAC_PVMeasuredVoltage_B": -1.799999999999999,
        "PVAC_PVMeasuredVoltage_C": 262.7,
        "PVAC_PVMeasuredVoltage_D": 262.90000000000003,
        "PVAC_Pout": 2440.0,
        "PVAC_PvState_A": "PV_Active",
        "PVAC_PvState_B": "PV_Active",
        "PVAC_PvState_C": "PV_Active",
        "PVAC_PvState_D": "PV_Active_Parallel",
...
        "PVS_StringA_Connected": true,
        "PVS_StringB_Connected": false,
        "PVS_StringC_Connected": true,
        "PVS_StringD_Connected": true,
```

Alternatively you can use the pyPowerwall python library directly:
```python
import pypowerwall

# Update with your details
password='password'
email='your@email.com'
host = "10.0.0.0"
timezone = "America/LosAngeles"

# Connect to Powerwall
pw = pypowerwall.Powerwall(host,password,email,timezone)

print("String Data: %s\n" % pw.strings(True))
print("Vitals Data: %s\n" % pw.vitals(True))

```

I only have one Powerwall+ and a single 7.6kW inverter so I only have one set up A, B, C or D strings.  You likely have two sets.  If so, to see that 2nd stress, you will need to adjust the query in Grafana to include A1, B1, etc. strings.  That can be adjusted via the GUI (the drop down above the graphs for the strings) using the Edit command. 

[image]. 

Click "Edit" and it will open up the queries:

[image]

Click the "Add Query" button to the right and it should show a space for query E, add the next string (e.g. A1_Power) and make sure you add the Alias name (e.g. A1):

[image]

Repeat for the other strings.

Based on what you discover, I can also update dashboard.json to include the other strings and then anyone with multiple Powerwall+ inverters can use that updated dashboard.json to add the additional strings.

cwagz, 2022-02-20:
Thanks for the help! I know enough to be dangerous, but that is about it. I am a little worried that they have more than 7.6kW on one inverter. That is what led me to your dashboard. Tesla changed their offerings so much by the time of installation my house does not match the line drawing they gave me up front at all. I am not unhappy as I ended up with the 420w panels and two powerwall+ but I just want to make sure they did the strings right. I have a feeling they should have split the larger array into two strings to even out the loading on the two powerwalls.

I might try to fix the dashboard myself with your directions or wait for an updated dashboard.json.

Let me know if you need any other data!

Here is the raw string data:

{
    "A": {
        "Connected": true,
        "Current": 5.23,
        "Power": 1636.0,
        "State": "PV_Active",
        "Voltage": 311.90000000000003
    },
    "A1": {
        "Connected": true,
        "Current": 0.16,
        "Power": 95.0,
        "State": "PV_Active",
        "Voltage": 400.90000000000003
    },
    "B": {
        "Connected": true,
        "Current": 1.48,
        "Power": 573.0,
        "State": "PV_Active",
        "Voltage": 380.0
    },
    "B1": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -2.1999999999999993
    },
    "C": {
        "Connected": false,
        "Current": 0.09,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -0.5999999999999996
    },
    "C1": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -1.0999999999999996
    },
    "D": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -2.1999999999999993
    },
    "D1": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -2.0
    }
}

Vitals Section:

{
[... 129 lines cut ...]
        }
    },
    "PVS--1538100-00-F--CN321278C0F62W": {
        "PVS_EnableOutput": true,
        "PVS_SelfTestState": "PVS_SelfTestOff",
        "PVS_State": "PVS_Active",
        "PVS_StringA_Connected": true,
        "PVS_StringB_Connected": false,
        "PVS_StringC_Connected": false,
        "PVS_StringD_Connected": false,
        "PVS_vLL": 245.0,
        "alerts": [
            "PVS_a018_MciStringB",
            "PVS_a019_MciStringC",
            "PVS_a020_MciStringD"
        ],
        "componentParentDin": "PVAC--1538100-00-F--CN321278C0F62W",
        "firmwareVersion": "0eb61d6819b2b3",
        "lastCommunicationTime": 1645315754,
        "manufacturer": "TESLA",
        "partNumber": "1538100-00-F",
        "serialNumber": "CN321278C0F62W",
        "teslaEnergyEcuAttributes": {
            "ecuType": 297
        }

jasonacox, 2022-02-20:
Thanks!  These are your only active strings:

```json
    "A": {
        "Connected": true,
        "Current": 5.23,
        "Power": 1636.0,
        "State": "PV_Active",
        "Voltage": 311.90000000000003
    },
    "A1": {
        "Connected": true,
        "Current": 0.16,
        "Power": 95.0,
        "State": "PV_Active",
        "Voltage": 400.90000000000003
    },
    "B": {
        "Connected": true,
        "Current": 1.48,
        "Power": 573.0,
        "State": "PV_Active",
        "Voltage": 380.0
    }
```

A and B are on one inverter.  A1 is on the other inverter.  It doesn't look like you are getting much out of A1 so getting that on your graph would be good.  In your case, it would be easy.  Just edit the graph and replace the "C" string with "A1" since your system is not using "C" or "D" inputs.  Let me know if that works for you.  Worse case, you can re-import the dashboard.json to wipe out your changes. :)

I'll work on a new dashboard.json to support two inverter systems.  Thanks for your help!

jasonacox, 2022-02-20:
If you wish, you can try to import this dashboard that has the added strings: https://github.com/jasonacox/Powerwall-Dashboard/blob/main/dashboard-2.json

I'm going to run a few more tests and unless I see issues, I will use this for the main dashboard.json file.

cwagz, 2022-02-20:
I imported it and I see all the new strings, but I cannot get them to display on the graphs. I am seeing A B C and D on the graphs. If I disable the queries and move "E" (A1) up it still does not display.

Might be something simple I am missing.

jasonacox, 2022-02-20:
Interesting.  Can you edit it again and click on the "field( ... )" box of the SELECT row and see if you see the drop down for the "A1" values?  I don't have A1 values in my strings so I can't replicate it, but it should be similar to this (except you should see A1_ fields) but let me know if you see something different:

[image]

cwagz, 2022-02-20:
I assume this means the database is not storing the other strings?

[image]

jasonacox, 2022-02-20:
Ugh!  Yes, that's correct.  Let me see if I can replicate it using my Powerwall simulator to mimic your setup.

cwagz, 2022-02-20:
Thanks! Maybe this is relevant.
I had to make a couple attempts to get this running. During troubleshooting I used the command:

docker logs -f pypowerwall

I noticed red in the output. Everything seemed to be working so I ignored it.

[image]

jasonacox, 2022-02-20:
I found it!  😊 

The issue is that InfluxDB needs to create those fields in the "strings" table.  You will find "A1_Power" under the "raw" table and then it will work.  But that table's retention is only 3 days so we want it to move into the "strings" table which has 2yr retention to give us longer trending.  You can run this to have InfluxDB fill in the missing fields in the "strings" table:

Run an interactive shell in InfluxDB:
```bash
docker exec -it influxdb influx
```

Copy and paste this (enter and enter ^D to exit):

```sql
CREATE CONTINUOUS QUERY cq_strings1 ON powerwall BEGIN SELECT mean(A1_Current) AS A1_Current, mean(A1_Power) AS A1_Power, mean(A1_Voltage) AS A1_Voltage, mean(B1_Current) AS B1_Current, mean(B1_Power) AS B1_Power, mean(B1_Voltage) AS B1_Voltage, mean(C1_Current) AS C1_Current, mean(C1_Power) AS C1_Power, mean(C1_Voltage) AS C1_Voltage, mean(D1_Current) AS D1_Current, mean(D1_Power) AS D1_Power, mean(D1_Voltage) AS D1_Voltage INTO powerwall.strings.:MEASUREMENT FROM (SELECT A1_Current, A1_Power, A1_Voltage, B1_Current, B1_Power, B1_Voltage, C1_Current, C1_Power, C1_Voltage, D1_Current, D1_Power, D1_Voltage FROM raw.http) GROUP BY time(1m), month, year fill(linear) END
```

Go back to Grafana and you should soon see this:

[image]

Let me know how it work for you.

I'll be updating the influxdb.sql setup scrip to include this.   Thanks for your help!

cwagz, 2022-02-20:
We are in business now!

I see all eight strings now. I will just hide the ones that are not in use.

Very glad I could help. Always fun setting something new like this up.

cwagz, 2022-02-20:
[image]

I will be moving to ethernet on the gateway tomorrow and then I will be getting good, continuous data.

jasonacox, 2022-02-20:
Wise move!  I did the same thing.  Despite "great signal" on WiFi, I would often seen those drop-outs at least once a day.  Now that it is on ethernet, I have had zero data loss for several weeks.  The cable was a pain to run, but worth it! 

Thanks again for the help!  I updated the project to have built-in support for systems with up to 3 PW+ inverters.

cwagz, 2022-02-20:
I assume proper etiquette is for me to close this issue. Hopefully. Take care.

closed 2022-02-20 by cwagz

nhasan, 2022-02-20:
This was so timely. I have 2 PW+ and just set up the dashboard last night. Originally I noticed that only one PW was supported and then magically support for additional PW appeared by the time I got to doing this. I had made local changes to add the CQ for additional strings but then discarded them once I saw this. I have A, B, A1, B1 strings active and showing nicely on the dashboard.

Oh, one more thing. The setup.sh failed for me when it tried to substitute the PW ip and password. Unfortunately I do not have the output of the script. I am running on RPI 4b+.

jasonacox, 2022-02-20:
Thanks @nhasan - that's great confirmation on the work @cwagz and I did.  I only have 1 PW+ and a PW 2 (8.5kW system) so I only have A, B, C and D strings.  I rewrote the dashboard and CQ for up to 3 PW+ inverters, but I suspect there is someone out there with 4 or more PW+ systems.  I'm thinking about updating the setup script to ask how many PW+ systems you have and building the dashboard and InfluxDB setup script dynamically.  I'll add that to my backlog.

On the setup script.  Thanks for that note.  If you don't mind, I'll open a new issue on that.  I have an idea what may be the cause but want to track it in case someone else has the same experience.

I'm happy to hear the in-place upgrade worked!  Like the [mihailescu2m](https://github.com/mihailescu2m/powerwall_monitor) version, the InfluxDB container uses a local `influxdb` directory to store the data (on purpose) to make it easier to upgrade.  The same is true for `grafana` to preserve your setup.