jasonacox/Powerwall-Dashboard #3: Strange issue with Telegraf and InfluxDB

nhasan, 2022-02-20:
I am seeing the following errors in Telegraf log:

```
$ docker logs -f telegraf
2022-02-20T05:02:11Z I! Starting Telegraf 1.21.4
2022-02-20T05:02:11Z I! Using config file: /etc/telegraf/telegraf.conf
2022-02-20T05:02:11Z I! Loaded inputs: cpu disk diskio http kernel mem processes swap system
2022-02-20T05:02:11Z I! Loaded aggregators:
2022-02-20T05:02:11Z I! Loaded processors: date (2x)
2022-02-20T05:02:11Z I! Loaded outputs: influxdb
2022-02-20T05:02:11Z I! Tags enabled: host=telegraf
2022-02-20T05:02:11Z I! [agent] Config: Interval:5s, Quiet:false, Hostname:"telegraf", Flush Interval:10s
2022-02-20T05:02:11Z W! [outputs.influxdb] When writing to [http://influxdb:8086]: database "powerwall" creation failed: Post "http://influxdb:8086/query": dial tcp 172.22.0.2:8086: connect: connection refused
2022-02-20T05:02:21Z E! [outputs.influxdb] When writing to [http://influxdb:8086]: failed doing req: Post "http://influxdb:8086/write?db=powerwall&rp=raw": dial tcp 172.22.0.2:8086: connect: connection refused
2022-02-20T05:02:21Z E! [agent] Error writing to outputs.influxdb: could not write any address
```
However, I am seeing data getting updated in InfluxDB as well as in Grafana. The log above stops at that point and then does not move.

I see the following in the InfluxDB log and it keeps moving:
```
[httpd] 172.22.0.5 - - [20/Feb/2022:05:16:01 +0000] "POST /write?db=powerwall&rp=raw HTTP/1.1 " 204 0 "-" "Telegraf/1.21.4 Go/1.17.7" 31a367b2-920c-11ec-8172-0242ac160002 12960
[httpd] 172.22.0.5 - - [20/Feb/2022:05:16:11 +0000] "POST /write?db=powerwall&rp=raw HTTP/1.1 " 204 0 "-" "Telegraf/1.21.4 Go/1.17.7" 37999496-920c-11ec-8173-0242ac160002 25396
[httpd] 172.22.0.5 - - [20/Feb/2022:05:16:21 +0000] "POST /write?db=powerwall&rp=raw HTTP/1.1 " 204 0 "-" "Telegraf/1.21.4 Go/1.17.7" 3d8f7af5-920c-11ec-8174-
```
Similarly I see the following in the pypowerwall log:
```
172.22.0.5 - - [20/Feb/2022 05:17:10] "GET /aggregates HTTP/1.1" 200 -
172.22.0.5 - - [20/Feb/2022 05:17:10] "GET /soe HTTP/1.1" 200 -
172.22.0.5 - - [20/Feb/2022 05:17:10] "GET /strings HTTP/1.1" 200 -
DEBUG:Found 17 devices.
```
Not sure what is going on here.

jasonacox, 2022-02-20:
Hi @nhasan  - Are you seeing issue with the graph or data?  Those logs look "normal" from my experience but please let me know if you see anything odd in the dashboard.

To help with what you are seeing in the logs:
* telegraf - that log happens the first time it runs and before InfluxDB is fully up and running - it will keep retrying until InfluxDB responds so that is why you don't see further errors.  Thankfully it becomes quite unless it has errors.
* InfluxDB - the database is a bit noisy when it comes to logging data - it records all the telegraph updates as well as the internal updates - I haven't discovered how to disable these logs and record only errors.  
* pypowerwall - this logs shows all the requests hitting it, the calls to the Powerwall, as well as debug data.  You can [turn off this debug data](https://github.com/jasonacox/Powerwall-Dashboard#docker-containers) but this can be helpful is troubleshooting Powerwall connectivity issues if you see those.

nhasan, 2022-02-20:
Hi @jasonacox , thanks for the quick reply. The data and graph look all good and I am seeing updates. So this is just timing issue it seems. Makes sense. Looks like telegraf is really quiet in logging as it does not log any successful update. That's what spooked me. All good now. Thanks again. Closing this issue now.

closed 2022-02-20 by nhasan