jasonacox/Powerwall-Dashboard #4: Setup.sh failed on Powerwall IP and password substitution

jasonacox, 2022-02-20:
This was so timely. I have 2 PW+ and just set up the dashboard last night. Originally I noticed that only one PW was supported and then magically support for additional PW appeared by the time I got to doing this. I had made local changes to add the CQ for additional strings but then discarded them once I saw this. I have A, B, A1, B1 strings active and showing nicely on the dashboard.

Oh, one more thing. The setup.sh failed for me when it tried to substitute the PW ip and password. Unfortunately I do not have the output of the script. I am running on RPI 4b+.

Also, I was previously running the [mihailescu2m](https://github.com/mihailescu2m/powerwall_monitor) version and I was able to seamlessly migrate to this dashboard without losing my historic data.

_Originally posted by @nhasan in https://github.com/jasonacox/Powerwall-Dashboard/issues/2#issuecomment-1046269018_

jasonacox, 2022-02-20:
@nhasan On the setup.sh failure - the [setup.sh](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/setup.sh) script is using simple substitution with `sed` :

```bash
sed -i .bak "s/password/${PASSWORD}/g" powerwall.yml
sed -i .bak "s/email@example.com/${EMAIL}/g" powerwall.yml
sed -i .bak "s/192.168.91.1/${IP}/g" powerwall.yml
```
If the user inputs for PASSWORD, EMAIL or IP contain the "/" character, it will fail.  Other escape characters could also cause problem.  I need to implement a more robust substitution method. Any thoughts?

Thanks for the feedback and raising the issue.

nhasan, 2022-02-20:
I tried to recreate this and found the root cause.

"sed -i .bak" fails due the space between option and the suffix. "sed -i.bak" works flawlessly.

As per the man page below, there is no space expected as well.

-i[SUFFIX], --in-place[=SUFFIX]

cwagz, 2022-02-20:
I believe this happened to me as well. I had to correct the file after running the script.

jasonacox, 2022-02-20:
Thank you!  You are right.  Surprisingly it works on MacOS (with space or not) which is my test platform but fails on linux.  I'll push the fix.

nhasan, 2022-02-20:
Most people are going to use this on a Raspberry Pi. So it would make more sense to make that your test platform. Thanks for pypowerwall and this dashboard BTW. Very useful tools.

jasonacox, 2022-02-20:
I agree @nhasan .  I now have a project today to spin up a spare RPi as a test node. :) 

Thanks for feedback and kind note.  By the way, with pypowerwall proxy in your stack, you can query it to get raw data (read only) from the Powerwall API including some aggregate functions you might find useful for other projects (I use this for ESP32 driven displays for example).  Replace localhost with your RPi address:

* pyPowerwall stats:  http://localhost:8675/stats
* Powerwall firmware version and uptime: http://localhost:8675/api/status
* Powerwall temperatures: http://localhost:8675/temps
* Powerwall device vitals: http://localhost:8675/vitals
* Powerwall strings: http://localhost:8675/strings
* Key power data in CSV format (grid, home, solar, battery, batterylevel): http://localhost:8675/csv

I should add these to the readme.

cwagz, 2022-02-20:
> I agree @nhasan . I now have a project today to spin up a spare RPi as a test node. :)
>
> Thanks for feedback and kind note. By the way, with pypowerwall proxy in your stack, you can query it to get [...]

This is awesome!

cwagz, 2022-02-20:
I am using Ubuntu 20.04.4 LTS in a VM on my home server. All out of RPi's and they are really hard to get right now!

cwagz, 2022-02-21:
I erased my virtual machine and installed from scratch last night with your latest updates. I am using Ubuntu Server 20.04.4 LTS.

The updated install script worked perfect.

The only issue I had was if you run git clone as 'sudo' you end up having permission errors with Grafana. I started over and ran git without 'sudo' and everything worked perfect. The install script need 'sudo' but not the initial git command. This may be obvious to the more experienced, but I am always surprised sometimes when 'sudo' ends up breaking something.

The dashboard updates are really nice. Thanks for all the work you have done on this.

[image]

jasonacox, 2022-02-21:
Thank for testing this @cwagz !   I'm so glad you got it working and love the screengrab!  FWIW, I also did testing on my Raspberry Pi 4 (32-bit Debian Bullseye), Ubuntu and MacOS (M1).  

On the issue you mention, you should not need to run 'sudo' for any of the steps.  I suspect your user is not part of the 'docker' group which is why the setup script required it.  If you run `docker ps` and get a permission error, that means your local user needs to be added to the docker group which can be fixed by: 
```bash
sudo usermod -aG docker $USER
```
You may need to log out or reboot for that to take effect (see [docker setup notes](https://docs.docker.com/engine/install/linux-postinstall/)).  I'll add a note to the readme to help with that.

I do have a favor to ask... and ok to say no! :)  I would love to showcase your dashboard (the same section you posted above) on the readme to highlight the Inverter Power graph with two inverters.  Would you be willing to get a snapshot of a full 24hr cycle and upload it here for me to post? 

Thanks again for your help.

cwagz, 2022-02-21:
I will upload some clean screenshots sometime tomorrow afternoon. They should be free of gaps now that I am on ethernet. I also will add my user to the docker group. Thank you.

jasonacox, 2022-02-21:
Awesome!  Thanks again.  And good job on getting the ethernet run.  I put a note in the readme about 'sudo' and the docker tips in the troubleshooting section.  Good find.

jasonacox, 2022-02-22:
[image]

I made a minor adjustment to [dashboard.json](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/dashboard.json) to group the "string" graphs into what Grafana calls a "Row" that I labeled "Powerwall+".  There are now 3 rows: Live Monitoring, Monthly Analysis and Powerwall+.  It allows for collapsing and moving them around a bit easier.  If anyone wants to try it, you should be able to import just the dashboard.json. 

BTW, in the jagged graph I posted you can tell we had good cloud cover here in Valencia.  The spikes to 6.5kW were due to the intriguing "cloud lensing" effect.  I'm thoroughly enjoying having solar and tweaking our Tesla Solar system. ;)

cwagz, 2022-02-22:
Cloudy day out here in Temecula as well. Should I wait for a solid 24-hour period with a sunny day before taking a screen grab for you?

[image]

jasonacox, 2022-02-22:
Oh yes!  No rush.  It would be better to get a typical full graph.  If you can, when you do have a good example to screenshot, make the browser window narrow so that it is a bit easier to read (e.g. text size - similar to the one I posted above)?   Thanks again for your help!

cwagz, 2022-02-25:
The dashboard is working great. Thank you again.

I just disabled the strings that are not active on my system:

[image]

It turns out I have 7.56kW of panels installed on Inverter1 and 4.62kW of panels installed on Inverter2. I assume this will work out fine. I did hit 7.4kw on Inverter1 due to cloud lensing the other day. The chart above was for today and is much more normal. I still have some large trees that are shading my roof during different parts of the day. This should be gone soon as the sun ends up higher in the sky. Today is also the first day that I have produced more power than we will use.

I am going to move everything over from my VM server to an APU2 I had laying around.

cwagz, 2022-02-25:
[image]

[image]

jasonacox, 2022-02-25:
Thank you @cwagz !   I updated the main page to include your example graphs but left my monthly graphs to give people an idea of how it will eventually populate: https://github.com/jasonacox/Powerwall-Dashboard

Thanks so much!!  This looks good. 

I am wondering what other trending data may be interesting to the community.  I am currently pulling in all the vitals data in a splunk instance, specifically frequency, current, voltage.   Do you think any of these would be interesting to add to the dashboard?  Or perhaps others?

[image]
[image]

cwagz, 2022-02-25:
Before we got PTO my tankless water heater errored out several times with a code that basically said check power condition. Or loose PCB connection. I believe this was related to the battery in the Powerwall filling up and increasing the frequency on the house to 65Hz to turn off any non-Tesla solar inverters. Now that we have PTO the system just exports power, so this has not been an issue.

Even with that said I would actually find it very interesting to track all this data down in the area where the strings are. Or in its own panel below the strings. It could definitely come in handy when trying to troubleshoot a strange problem like I described above.

On another note: What do you think the storage requirements will end up being for all of this data? I have a 500GB SSD in my APU2 so I am not that worried about it, but I would like to keep several years of data if possible. It will be great for seeing how the system is aging.

jasonacox, 2022-02-25:
Thanks!   That's a great example.  I'll see about getting those added. 

As for space, I have data for 5 months and current InfluxDB is at 102MB.  Assuming linear, that means 1 year would be ~250MB/yr, so you will be in good shape.  However, that does make we wonder.  The default setup has InfluxDB retaining data for 2 years (730d).  We may want to update that to 10y or 25y to align with the expected life of the system considering the data footprint is so reasonable. 

```bash
docker exec -it influxdb influx
```
```sql
ALTER RETENTION POLICY autogen ON powerwall duration 9125d
ALTER RETENTION POLICY strings ON powerwall duration 9125d replication 1
ALTER RETENTION POLICY pwtemps ON powerwall duration 9125d replication 1
```

jasonacox-sam, 2026-08-31:
This long thread covered the original `setup.sh` failures around Powerwall IP/password substitution and grew into a useful exchange on multi-Powerwall string support, extra trending graphs, and InfluxDB sizing (~250MB/year). The setup issues and string support were resolved in the repo at the time.

Closing this out due to age. If setup fails on a current version, please open a new issue with the error output.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam