jasonacox/Powerwall-Dashboard #5: String Power Y-Axis Scale

cwagz, 2022-02-20:
I am noticing that string A is correctly scaled on the chart with the y-axis. It appears the other strings, although the values are right, plot out at a much higher wattage on the Y-Axis.

I will see if I can figure this one out on my own.

[image]

cwagz, 2022-02-20:
I believe turning off stacking fixes this:

[image]

Stacking off:

[image]

nhasan, 2022-02-20:
Yes, this is due to the stacking in the graph. What I did was to turn off the stacking and add another query for "total" string power.

```
SELECT mean("A_Power")+mean("A1_Power")+mean("B_Power")+mean("B1_Power") FROM "strings"."http" WHERE $timeFilter GROUP BY time($__interval) fill(null)
```

jasonacox, 2022-02-20:
Good feedback.  Stack or unstack, that is the question.  I went back and forth on that before I landed on "stacked" but I can see how that is confusing since the voltage and current graphs are unstacked.  I like the idea @nhasan has of adding a total query to the same graph and just leave it unstacked.  Thoughts?

cwagz, 2022-02-20:
I was confused by the stacking, but I see your reasoning now. One of my main goals here is to make sure my individual inverters are not overloaded. I am concerned I have three strings on a 12.4 system. Would it be possible to programmatically create a total query that will chart the full output from each inverter? This way I can see if my 2-string inverter starts clipping at 7.6kW. If this does not make sense to automate, I will probably modify @nhasan idea but make a total query for each inverter.

nhasan, 2022-02-20:
This is how my graph looks now:

[image]

jasonacox, 2022-02-20:
@cwagz Great idea!  I would be curious if you would want to see these in the same graph panel or multiple panels in grafana?  In other words:
* Option 1:  All on one graph - The addition would be similar to what @nhasan posted but with additional lines for inverters (Inverter-1, Inverter-2, etc.) that are sum of the relevant strings (A...D, A1...D1, A2...D2, etc.).  
* Option 2: Separate graphs for each Inverter with string breakdown (A...D) and a total line.

Naturally, as you have seen, once set up, Grafana is easy to edit to create different cuts of the data you want to see, but for "out of the box" let me know what you think makes the most sense.

cwagz, 2022-02-20:
@jasonacox I really like your idea of another pane. Keep the string power the way it is, but unstacked. Add a 4th panel that will have the total output from each inverter. The total system output could chart on this pane as well. This would keep the individual string data pane less cluttered.

cwagz, 2022-02-20:
I may have been thinking of an option 3. Basically, just a totals pane in kW.

jasonacox, 2022-02-20:
That makes sense.  I'll add that to my list.  

Unrelated to this issue, but do you think there would be any interest in having Powerwall temperature graphs?   My PWs are in my garage, but it may be helpful for winter / summer extreme conditions to understand temp impact to the Powerwalls:

[image]

cwagz, 2022-02-20:
Temperature graphs would be great. Mine had to be installed outside (Temecula changed the building code 2 weeks before my install) and they are going to get a lot of sun for a good portion of the year.

cwagz, 2022-02-20:
This is what I ended up coming up with for an Inverter Power panel:

[image]

closed 2022-02-26 by cwagz