jasonacox/Powerwall-Dashboard #7: Folder to Backup

cwagz, 2022-02-26:
I am very unfamiliar with Docker. Is there a folder I can backup periodically that would basically contain the entire working system and database? This way I could restore to a new system if needed down the road.

jasonacox, 2022-02-27:
Back up the Powerwall-Dashboard folder.  In that folder are two important folders:

* influxdb - This is the folder for the database that stores the metrics. 
* grafana - This is the folder for the dashboard which holds your setup and customization.

You can stop, upgrade or destroy the docker containers, but as long as you rebuild them using these same folders as source volumes (see the [powerwall.yml](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/powerwall.yml) file), the data will still be there. 

I have moved my data around between instances quite often without stopping the docker containers (simple `tar` backup and restore).  However, there is still a risk that you get a snapshot that has some inconsistency (e.g influxdb did not fully write an update to disk during backup).  That would be the same problem observed if you lost power or your VM hosting the containers shut down without stopping the containers.  For that reason, you may also want to have a process to get golden backups by doing a graceful shutdown before the backup, something like this:

```bash
# Go to directory hosting dashboard
cd ~/Powerwall-Dashboard

# Stop Docker Containers
docker-compose -f powerwall.yml stop

# Create Backup (root required for protected files)
sudo tar -zcvf /some-backup-location/dashboard.tgz influxdb grafana

# Restart Docker Containers
docker-compose -f powerwall.yml start
```

Perhaps a less disruptive approach would be to have influxdb backup the data for you and you archive that along with the snapshot:

```bash
# Go to directory hosting dashboard
cd ~/Powerwall-Dashboard

# Run this once to create a place for influx to store backups
mkdir influxdb/backups

# Ask InfluxDB to create a full backup 
docker exec -it influxdb influxd backup -database powerwall /var/lib/influxdb/backups

# Create Backup (root required for protected files)
sudo tar -zcvf /some-backup-location/dashboard.tgz influxdb grafana
```

Naturally, whatever you do, make sure you test it.  Copy the backup to another VM or box, install Powerwall-Dashboard and restore the backup to see if it all comes back up without any data loss.

Would love to hear any other ideas too.

cwagz, 2022-02-27:
This is really helpful. I used Webmin and mounted a file share from my Windows server to use as a backup target.

Webmin's Filesystem Backup module lets you send a command before and after backup.

For now, I am going to have the docker instance stop, backup and then restart. Currently this is taking about 1 minute and will happen at midnight. I was getting a few errors periodically related to a file being changed during backup when trying to just snapshot it while running.

[image]

cwagz, 2022-02-27:
Nice - I think I like the script you posted even better than what I did. Thanks again!

cwagz, 2022-03-01:
I think the influxdb backups will just keep accumulating in the influxdb\backups folder. Is this correct? If so, I am going to make this change to your script to delete them after each backup:

```
# Cleanup Old Backups
echo "Cleaning up old backups"
rm -rf ${DASHBOARD}/influxdb/backups/*
find ${BACKUP_FOLDER}/Powerwall-Dashboard.*tgz -mtime +${KEEP} -type f -delete
echo "Done"
```

jasonacox, 2022-03-01:
Yes, that's a great idea!   You should submit a pull request.  I would love to add you as a contributor to our project. ;)

cwagz, 2022-03-01:
I will try to figure it out. Thanks.

cwagz, 2022-03-01:
Hopefully I didn't mess that up too bad. I read a tutorial and created a fork. This created a public repository under my name. I guess I wasn't expecting that. Is it a problem? Should I delete it? I don't want to create any confusion.

jasonacox, 2022-03-01:
You did it!!   Good job.  I really appreciate the help and glad to welcome you as an OSS contributor!  :) 

[image]

closed 2022-09-25 by jasonacox