jasonacox/Powerwall-Dashboard #10: All Time Max and Past 12 Months Max

cwagz, 2022-03-03:
In Grafana, would it be possible to add a panel that would look at historic maximums? This would be very useful for monitoring system performance over time.

Like a table that would show the date and time of the max production of each string in the last 12 months and then also show the date and time of the max production of each string for all time.

jasonacox, 2022-03-04:
Interesting.  You can create a table panel and have it do some queries.  To get what you want, I think we would have to use that query to downsample the data in InfluxDB so that you get a per-day max resulting table and then use that table to get the max for all time.  I'm not sure how to capture the timestamp of that max.

No exactly what you want, but here is a per-day-max table it made for my setup:

```sql
SELECT max("A_Power") FROM "strings"."http" WHERE $timeFilter GROUP BY time(1d) fill(0) tz('America/Los_Angeles')
SELECT max("C_Power") FROM "strings"."http" WHERE $timeFilter GROUP BY time(1d) fill(0) tz('America/Los_Angeles')
SELECT max("D_Power") FROM "strings"."http" WHERE $timeFilter GROUP BY time(1d) fill(0) tz('America/Los_Angeles')
SELECT max("Inverter1") FROM "strings"."http" WHERE $timeFilter GROUP BY time(1d) fill(0) tz('America/Los_Angeles')
```

[image]

cwagz, 2022-03-04:
I think I am getting somewhere. I will try to post some screenshots later.

cwagz, 2022-03-04:
[image]

cwagz, 2022-03-04:
[image]

jasonacox, 2022-03-04:
Nice!  Can you post the queries too?

cwagz, 2022-03-04:
```
SELECT max("A_Power") FROM "strings"."http" WHERE $timeFilter fill(0) tz('America/Los_Angeles')
SELECT max("B_Power") FROM "strings"."http" WHERE $timeFilter fill(0) tz('America/Los_Angeles')
SELECT max("A1_Power") FROM "strings"."http" WHERE $timeFilter fill(0) tz('America/Los_Angeles')

SELECT max("Inverter1") FROM "strings"."http" WHERE $timeFilter fill(0) tz('America/Los_Angeles')
SELECT max("Inverter2") FROM "strings"."http" WHERE $timeFilter fill(0) tz('America/Los_Angeles')
```
The queries are just to cover my system so I don't have all the other strings and such. I moved them down to the bottom of the Powerwall+ group:

[image]

jasonacox, 2022-09-25:
Should we add a dashboard or panel section for these all time max tables?  Maybe we call it "Greatest Hits" or "HIgh Score" or "Leaderboard"? 😀

cwagz, 2022-09-28:
This idea didn't work out too well for me. I think "cloud lensing" ends up creating false maximums. I ended up abandoning the concept. I like the ideas for names though!

jasonacox, 2022-09-28:
That makes sense.  I'll close this for now and we can re-open if we ever think of a way to use it. :)

closed 2022-09-28 by jasonacox