jasonacox/Powerwall-Dashboard #11: Invalid Interval String

dmayo305, 2022-03-04:
I had previously installed on an older RPi 2B and all was well. Purchased a new RPi 400 (solid purchase, btw) and set up everything the same way, but now only the Live Monitoring "Energy Usage" panel is showing data. InfluxDB appears to be functioning. Here's what I get for an error:
Invalid interval string, expecting a number followed by one of "Mwdhmsy"
[image]
The only difference I can think of was I was running a headless config before, and I'm running a desktop on the RPi 400.

dmayo305, 2022-03-04:
All stats appear normal when looked at through pyPowerwall proxy. Digging into InfluxDB now...

jasonacox, 2022-03-05:
Hi @dmayo305 - what OS are you using?   It does look like an InfluxDB issue. I have a good test running with an RPi 4 with 32-bit Bullseye.

dmayo305, 2022-03-05:
This has happened on 32 bit and 64 bit bullseye, both with desktop environment. The only time it didn't happen was on headless 2B running 32 bit bullseye

dmayo305, 2022-03-05:
I noticed I don't get a GUI when attempting to access port 8086 from either local or across the network, is that a settings difference from the installation script?

jasonacox, 2022-03-05:
Port 8086 is the InfluxDB port so there won't be a GUI there.   Can you check your InfluxDB setup in Grafana to make sure you put "5s" and not just "5" for the time interval?

Here is how:  In Grafana:
Click the "Settings" wheel ->  "Configuration" -> "Data Sources" 

[image]

Scroll to the bottom to make sure you put in 5s (the s is important):

[image]

dmayo305, 2022-03-05:
> Port 8086 is the InfluxDB port so there won't be a GUI there. Can you check your InfluxDB setup in Grafana to make sure you put "5s" and not just "5" for the time interval?
> 
> Here is how: In Grafana: Click the "Settings" wheel -> "Configuration" -> "Data Sources" -> "InfluxDB"
> 
> [image]
> 
> [image]
> 
> Scroll to the bottom to make sure you put in 5s (the s is important):
> 
> [image]

You nailed it. Not sure how I missed that! That said, reading the influx man makes one think there is a UI:
https://docs.influxdata.com/influxdb/v2.1/visualize-data/dashboards/

jasonacox, 2022-03-05:
I'm glad that was the fix!  I should add that as a troubleshooting tip.

closed 2022-03-05 by dmayo305