jasonacox/Powerwall-Dashboard #12: Any way to import all historical data?

mikkyo9, 2022-03-04:
It seems the DB is populated from the time of install?
Any way to pull in all the history?

jasonacox, 2022-03-05:
Hi @mikkyo9  - good question.  The Powerwall doesn't keep that history itself.  When you install the "Powerwall-Dashboard" stack, the Telegraf service starts polling the Powerwall to get the metrics.  The data is then stored in the DB for Grafana to display.  The Tesla App has this historical data because it (Tesla Cloud) started polling the Powerwall when it was commissioned.   I'm not aware of a way to pull this historical data into the database and even if we could, it would be missing some data like the String and Temperature data.

dmayo305, 2022-03-05:
You could probably manually download each day's data and then write a script to transfer that information into the influx DB. I wish I could do an aggregate data download with fine granularity. Best I've found is to do a daily download in the phone app, then upload that to my dropbox and then view the info on my desktop.

dmayo305, 2022-03-05:
@jasonacox - Your abilities greatly outpace mine, so I'm bringing an idea without specific testing. On https://www.teslaapi.io/powerwalls/state-and-settings and https://www.teslaapi.io/energy-sites/state-and-settings there appears to be an API call for historical data using a query period of day, week, month, year (similar to the granularity offered in the app). My thoughts are, could one develop a script with a start date, and then increment by day to download historical data?

jasonacox, 2022-03-07:
Thanks @dmayo305 - I have been looking at those Tesla cloud APIs (used by the Tesla app) to add features to pyPowerwall  (e.g. update battery reserve setting). However, it seems Tesla keeps making breaking changes so I haven't found a good example/library to use TeslaPy seems to be close but I haven't been able to get it to work yet.  If we can figure it out, it wouldn't be too difficult to do what you mention.

fracai, 2022-03-11:
I have a sample project that uses TeslaPy to download Powerwall and Solar site data as json files. I imagine it could be an example of how to get the data for adding to the Dashboard, but I don't know if it would need to be reformatted before loading in to the database.

This example uses the "refresh token" method for logging in, so you'll need to generate that first with a 3rd party app or one of the other examples floating around.

fracai, 2022-03-11:
https://github.com/fracai/tesla-history

The sample project in question

jasonacox, 2022-03-12:
Thanks @fracai !  Nice work.  Yes I think we could use that to port the data into InfluxDB.  What method did you use to get the "refresh token"?   If we extend your project to drop that data into the Dashboard DB, we could include instruction on how to make it work.  

Ideally, I want to find an automated way to get the token so I can include that in the [pyPowerwall](https://github.com/jasonacox/pypowerwall) library.

fracai, 2022-03-12:
I used "Auth for Tesla", an iOS app. There are a few apps listed on this [TeslaMate FAQ](https://docs.teslamate.org/docs/faq/).

I think an automated method is ideal, but I've gotten the impression that manually retrieving the refresh token is the best method as Tesla frequently requires entering a captcha in order to login.

Fortunately, I think the refresh token is valid for a very long time and, unless you need to fill in data gaps, it shouldn't be necessary to request historical data. It might not be ideal, but I think it'd be acceptable to use a manual process to fill in old data.

Then again, if the refresh token is stored, it'd be possible to make a daily request of historical data to regularly fill in any missing gaps.

fracai, 2022-03-12:
There's also this python script for retrieving the token, though it does require manual intervention.

https://github.com/bntan/tesla-tokens

jgleigh, 2022-04-25:
Any more work being done for historical data? It'd be really cool to pull in all the old data.

jasonacox, 2022-04-26:
I agree. It's not an easy fix, unfortunatelyl.  I'm doing more investigation and open to any help. ;)

dvanderb, 2022-08-01:
What I am most interested in is my NEM year data(which started a month ago). So even if I could just import the basic data(to grid, from grid, home usage) as a single point in history it would be useful to fill in the gap from when I started to use powerwall-dashboard.

mcbirse, 2022-09-21:
I have some good news regarding this! 😀 

Thanks to @jasonacox for pointing out the TeslaPy module and also @fracai for the sample project using that. I started testing the module and have worked out how to import historical data via the Tesla Owner API (cloud) directly into InfluxDB. 🎉 

Having >1yr of historical data that I would certainly love to see in my dashboard was a good motivator. 😊 

I'm still working on the program but hopefully will be ready to release in the next week or few. Full time day job doesn't help 😞 - also I've had to brush up on my python knowledge, as these days I'm primarily coding in a language probably most people have never heard of!

So, I have developed this as a command line tool that accepts a start & end datetime range to retrieve history from the Tesla Owner API and import into InfluxDB. @jasonacox I'm thinking submitting a PR to add it to the tools directory seems appropriate?

I'm using the staged authorization process of TeslaPy as it seemed convenient and easy in my opinion. The login to your Tesla account is only required once, as after login a token is cached/saved and it appears to never expire (similar to the Tesla app which stays logged in).

Obviously not all data that the Powerwall-Dashboard logs is available for import. But, the main data shown in the "Energy Usage" and "Power Meters" panels is.

I think I can get the "Monthly Analysis" data working too (still testing). This is stored in the kwh retention policy, but, it is populated from a CQ based on the data in autogen. So I just need to run a query to refresh it after importing history data.

Also, I've been able to extrapolate the grid status from the backup event history, which is nice. 😃 

In case there are problems with an import, obviously backups are highly recommended. But, the program will also tag all the data points with a "source=ownerapi" tag in InfluxDB. This makes it easy to reverse what was imported with a query like `DELETE WHERE source='ownerapi'` if something went wrong (and could be further limited to a specific time period if required). This would only delete the imported data, and not affect data that came from your gateway via the dashboard. Also note original dashboard data is never overwritten or changed if you were to enter a datetime range that overlapped with the dashboard data (although I would not recommend - it is best to just fill periods of missing data only).

The history data available from the Tesla cloud is in 5 minute intervals only (15 min for charge percentage), but it seems good enough. Here are some example comparisons.

Below is showing data logged from Powerwall-Dashboard as normal (i.e. every 5 seconds or so):
[image]

Compared with below where data was imported from the Tesla cloud - the graph and values (e.g. Power Meters) are pretty close:
[image]

Example showing imported history data with a backup event - the grid status value is extrapolated from backup event history:
[image]

If you have a data gap / missing data your graph might look like below:
[image]

This can fixed now by importing the history from Tesla cloud for the specific time period:
[image]

So it seems to be working well so far as I can tell, and useful for importing old data from before you installed Powerwall-Dashboard, as well as if your system went down for some reason!

While I'm still working on this, if anyone has any feature requests, questions, tips etc. please send them my way. 😊  

(side note: I'm already thinking about fetching historical weather data as well now, which is available from the OpenWeather One Call API 3.0 "timemachine" API call 😉 )

jasonacox, 2022-09-21:
Hi @mcbirse  Great work here!!!  Please feel free to submit a PR in the /tools directory.  I'm sure others would love this too.

jgleigh, 2022-09-22:
I'm looking forward to testing this. I have several small data dropouts as well as several months of data before I got the dashboard setup.

fracai, 2022-09-22:
Same here. Take your time to get it where you're happy, but I'm really looking forward to getting a look as well.
Thanks for working on it.

mcbirse, 2022-10-13:
Update finally - I have been working on the script to import historical data from Tesla cloud over the past few weeks, and believe it is now good to go! :crossed_fingers:

I will submit a PR to add it to the /tools directory shortly.

Hoping for some feedback from the community on this, as so far all testing has been with my system only.

Please see the below instructions for how to use the script (will also add this to the tools README).

The tool can be used to import historical data from Tesla cloud, for instance from before you started using Powerwall-Dashboard. It can also be used to fill in missing data (data gaps) for periods where Powerwall-Dashboard stopped working (e.g. system down, or lost communication with the Local Gateway). By default, the script will search InfluxDB for data gaps and fill gaps only, so you do not need to identify time periods where you have missing data, as this will be done automatically.

NOTE:
- The available data from Tesla cloud is limited to 5 minute intervals of power usage, and backup events only. It will not be as accurate as data logged by Powerwall-Dashboard, however is still sufficient to provide approximate usage history. The historical data is obtained via the same API as used by the Tesla mobile app, so should be very similar in accuracy to what is displayed in the app.
- Data imported by this tool will not overwrite existing data, and will update the following panels only: Energy Usage, Grid Status, Power Meters, and Monthly Analysis

To use the script:
- Install the required python modules: `pip install python-dateutil teslapy influxdb`
- Follow the steps below

### Setup and logging in to Tesla account

On first use, it is recommended to use the `--login` option. This will create the config file, save an auth token so you will not need to login again, and then display the energy site details associated with your Tesla account.

```bash
# Login to Tesla account
python3 tesla-history.py --login
```

It will run in an interactive mode. The example below shows the config creation:

```
Config file 'tesla-history.conf' not found

Do you want to create the config now? [Y/n] Y

Tesla Account Setup
-------------------
Email address: your@email.address
Save auth token to: [tesla-history.auth]

InfluxDB Setup
--------------
Host: [localhost]
Port: [8086]
User (leave blank if not used): [blank]
Pass (leave blank if not used): [blank]
Database: [powerwall]
Timezone (e.g. America/Los_Angeles): Australia/Sydney

Config saved to 'tesla-history.conf'
```

In most cases, the `[default]` values will be correct and can be accepted by pressing Enter, however these can be changed if you have a custom setup.

Generally, only your Tesla account `email address` and your `timezone` will be required.

After the config is saved, you will be prompted to login to your Tesla account.

This is done by opening the displayed URL in your browser and then logging in:

```
----------------------------------------
Tesla account: your@email.address
----------------------------------------
Open the below address in your browser to login.

<copy URL to browser> e.g.: https://auth.tesla.com/oauth2/v3/authorize?response_type=code...etc.

After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login: <paste URL from browser> e.g.: https://auth.tesla.com/void/callback?code=...etc.
[... 217 lines cut ...]

### InfluxDB notes and Analysis Data

I am recording some technical notes below regarding InfluxDB and the Analysis data updates for anyone interested - although mainly because I'm likely to forget why I have done something a certain way in the program if I need to look at this again in the future. :blush:

Importing data:
- When writing large amounts of data to InfluxDB, I found it is *significantly faster* to write data points in Line Protocol format, instead of JSON
- The power usage data retrieved from Tesla cloud is imported into autogen.http however with an added tag of `source='cloud'`
- Any existing data points will not be affected at all, as the tags don't match (refer [here](https://docs.influxdata.com/influxdb/v1.8/troubleshooting/frequently-asked-questions/#how-does-influxdb-handle-duplicate-points) for an explanation of how InfluxDB handles duplicate data points)
- This also means, it is easy to remove any imported data by simply specifying `WHERE source='cloud'` in a `DELETE` query - only imported data will be removed, as original data will not have this tag
- When deleting data in InfluxDB, it is important to note it is not possible to remove data from a specific retention policy, instead a DELETE will remove data from *all retention policies*. However, due to the imported data being tagged with `source='cloud'` this will not be an issue as the data in the other retention policies will not have this tag

For updating analysis data of InfluxDB (kwh, daily, and monthly retention policies):
- After data is imported into autogen, the script will currently replicate the SELECT INTO queries used in the CQ's for kwh, daily, and monthly
- To update the measurement values of existing data points however, it is important to ensure the data points have the exact same timestamp and tags (refer [here](https://docs.influxdata.com/influxdb/v1.8/troubleshooting/frequently-asked-questions/#how-does-influxdb-handle-duplicate-points) again), therefore points cannot be tagged with source='cloud' as this would add points (NOTE: some of the CQ's are a sum() of data poin [... cut ...] dorssers/TeslaPy) module `Be aware that Tesla may temporarily block your account if you are hammering the servers too much`).

However, this did only occur when trying to request data as quickly as possible. As such, a default 1 second delay between requests has been added to the script (can be changed by adding `DELAY = x` in the config file), after which I was able to retrieve a full year of historical data in one run without issue.

Please let us know if you find any issues or have questions.

fracai, 2022-10-14:
This looks fantastic. I haven't had a chance to test it yet, but I want to soon. I love that you have it designed to search for and fill gaps. That's great.
Thanks for your work. I look forward to testing.

jgleigh, 2022-10-14:
AMAZING!!! I imported over six months of data and filled a ton of small data gaps. No issues so far. THANKS!!!

mikkyo9, 2022-10-14:
Cool! I can't wait to try it out. I assume your work is based on the most current release schema, so I should upgrade first?

mcbirse, 2022-10-15:
> Cool! I can't wait to try it out. I assume your work is based on the most current release schema, so I should upgrade first?

@mikkyo9 - You will need to be running at least version 2.3.0 I believe, as that was the release when the CQ for grid status was added (released July). I'd still recommend upgrading to the latest version if you can though, as there has been many fixes.

> AMAZING!!! I imported over six months of data and filled a ton of small data gaps. No issues so far. THANKS!!!

@jgleigh - 

Awesome to hear, thanks for the feedback! 

Yes, you may see a number of small data gaps identified even if you think your system had been logging non-stop without issues.

This is more likely with the grid status data as the gap threshold is 1 minute for that, as apposed to 5 minutes for power usage. 

Any time you do an upgrade, for instance, you probably lose a few minutes of data but generally not more than 5 mins in my experience. 

I considered accepting longer gaps for grid data - but then looking at my own backup history, I've had a number of events that range from a few seconds to a few minutes... so this way these will still be filled in if the system did not log data during that time for whatever reason.

mikkyo9, 2022-10-15:
Great job on this!
Worked very well filling dozens of gaps and a missing year of data.

youzer-name, 2022-10-15:
Works like a charm. Thanks for putting this together @mcbirse

DMBlakeley, 2022-10-15:
Worked quite well for me also, thanks for the all your efforts!

oralallen82, 2022-10-16:
This worked like a charm. It's amazing what a community can do. I have all the date from system install August 2021 thanks @mcbirse and @jasonacox

mcbirse, 2022-10-17:
Thanks everyone for the feedback - great to know it is working! 😊 

I had one last TODO item for the program, which was to confirm the home power usage calculation is correct during VPP events (it *should be* 🤞 ).

@cwagz - as discussed earlier, you have some data from a VPP event which should help us to confirm this.

If possible, it would be great if you could run the program with the below arguments when you have a chance:

```bash
python3 tesla-history.py --start "2022-08-17 00:00:00" --end "2022-08-17 23:59:59" --force --debug --test
```
Using the `--test` option will ensure no data is imported so it won't affect your system.

`--debug` will show the responses from Tesla cloud. I am interested in seeing the power usage responses only, like below: 
(i.e. not the other responses, which could include personal information about your powerwall)

```json
        {
            "timestamp": "2022-08-17T18:00:00+10:00",
            "solar_power": 0,
            "battery_power": 567.1428571428571,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T18:05:00+10:00",
            "solar_power": 0,
            "battery_power": 565.7142857142857,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T18:10:00+10:00",
            "solar_power": 0,
            "battery_power": 557.8333333333334,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
```

Specifically I would like to see the values around 6pm-8pm (and before/after) when the VPP event occurred, so we can see in the response how the `grid_services_power` value is used and whether that affects how home usage power is calculated:

[image]

Appreciate the help, thanks.

cwagz, 2022-10-20:
@mcbirse - Here you go. Thanks for working on this!

        {
            "timestamp": "2022-08-17T17:55:00-07:00",
            "solar_power": 1727.142857142857,
            "battery_power": 0,
            "grid_power": 1146,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T18:00:00-07:00",
            "solar_power": 1592.857142857143,
            "battery_power": 11457.142857142857,
            "grid_power": -6196.857142857143,
            "grid_services_power": -10547.370702489794,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T18:05:00-07:00",
            "solar_power": 1501.6666666666667,
            "battery_power": 11460,
            "grid_power": -7958.333333333333,
            "grid_services_power": -11602.287772738773,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T18:10:00-07:00",
            "solar_power": 1405.7142857142858,
            "battery_power": 11454.285714285714,
            "grid_power": -10301.714285714286,
            "grid_services_power": -11602.387772738773,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T18:15:00-07:00",
            "solar_power": 1308.5714285714287,
            "battery_power": 11455.714285714286,
            "grid_power": -10424.285714285714,
            "grid_services_power": -11602.337772738774,
            "generator_power": 0
        },
		{
            "timestamp": "2022-08-17T19:30:00-07:00",
            "solar_power": 0,
            "battery_power": 11460.140845070422,
            "grid_power": -3353.760563380282,
            "grid_services_power": -10665.09230304092,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T19:35:00-07:00",
            "solar_power": 0,
            "battery_power": 11461.346153846154,
            "grid_power": -2332.6923076923076,
            "grid_services_power": -11600.851106072108,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T19:40:00-07:00",
            "solar_power": 0,
            "battery_power": 11455.757575757576,
            "grid_power": -7867.212121212121,
            "grid_services_power": -11600.841106072108,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T19:45:00-07:00",
            "solar_power": 0,
            "battery_power": 10071.919191919193,
            "grid_power": -7377.080808080808,
            "grid_services_power": -10440.758995464897,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T19:50:00-07:00",
            "solar_power": 0,
            "battery_power": 0,
            "grid_power": 2383.746835443038,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T19:55:00-07:00",
            "solar_power": 0,
            "battery_power": 0,
            "grid_power": 6388.857142857143,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T20:00:00-07:00",
            "solar_power": 0,
            "battery_power": 0,
            "grid_power": 6504.666666666667,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2022-08-17T20:05:00-07:00",
            "solar_power": 0,
            "battery_power": 0,
            "grid_power": 3018.285714285714,
            "grid_services_power": 0,
            "generator_power": 0

cwagz, 2022-10-20:
@mcbirse - I just loaded all my history. This is amazing! Thank you again for doing this.

cwagz, 2022-10-20:
Does anyone else have some crazy data in the Tesla cloud? This is mine:

[image]

The sun must have really been blasting!

Would this be easy to clean up manually?

cwagz, 2022-10-20:
> > The sun must have really been blasting!
> 
> It's hard to read your tone here - I take it that you are having a chuckle about unexpected and unlikely milliwatt hours values in the data (and that was probably not a heavy cloud/rain day with minimal generation, also not Megawatt hours?)
> 
> You can use TeslaPy to take a look historical data - used by mcbirse in his history tool, and I'm having a putter with it on the historical cumulatives in the cloud.

Actually, I mistakenly thought this was the day I produced Megawatt hours. I should have checked the app better and done a sanity check on my math. The weird data day was actually December 14, 2021. The Tesla App shows .136 MWh that day. It seems my memory was off by an order of magnitude.

Tesla App:
[image]

Downloaded csv From Tesla App:
[image]

Interestingly, the anomaly did not get picked up when I imported the pre-Powerwall Dashboard data from the Tesla cloud:

Imported Data:
[image]

mcbirse, 2022-10-20:
@cwagz - thanks for the output! 

That's terrific, I believe this confirms all the calcs tally up with your graph correctly for home usage during the VPP.

So home usage power calculation is correct as below, no change required in the program (except to remove the comment):
```py
# Calculate power usage values (TODO: check 'home' power usage calculation is correct during VPP event - NOTE: response includes 'grid_services_power')
home = d['solar_power'] + d['battery_power'] + d['grid_power']
```

E.g. at 6pm
```json
    {
        "timestamp": "2022-08-17T18:00:00-07:00",
        "solar_power": 1592.857142857143,
        "battery_power": 11457.142857142857,
        "grid_power": -6196.857142857143,
        "grid_services_power": -10547.370702489794,
        "generator_power": 0,
    },
```
So at 6pm home usage in being calculated in the program as below:
```
"home" = "solar_power": 1592.86 + battery_power": 11457.14 + "grid_power": -6196.86 = 6853.14
```
This looks about right according to the graph at that time. Cheers!

P.S. @cwagz - Mass Effect fan? 😊

TechCrazi, 2022-10-20:
Getting this error.
ERROR: Failed to retrieve SITE_DATA - string indices must be integers

BuongiornoTexas, 2022-10-20:
> The Tesla App shows .136 MWh that day. It seems my memory was off by an order of magnitude.

The app and Tesla cloud do some odd things - you can get slightly different outcomes depending on the time range and the data you are pulling, and there are also a set of rules for "correcting" weird data that are not applied consistently (e.g. I have a day with a misconfigured CT - it can be reported with truncated data, with 'fixes" applied, or not reported at all, depending on the format you ask for).

I believe the history that @mcbirse extracts is reconstructed from 5 minute power readings, which I think are be a different data set to that used for the app energy reporting (as noted above - these can contain some weirdness). Lots of guesswork in closed APIs though!

mcbirse, 2022-10-21:
> Getting this error.
> ERROR: Failed to retrieve SITE_DATA - string indices must be integers

@TechCrazi - strange, not sure why you would be getting this error (first error reported!).

Can you do some tests to help diagnose this issue?

First off, I'm just wondering if this was a temporary problem and it has subsequently started working at a later time, or is this happening all the time?

The script is failing on this API call:
```py
# Retrieve site current time
data = battery.api('SITE_DATA')['response']
```
The response returned is a string instead of the expected JSON response. The string may contain some clues as to why it is failing, however I did not account for this so we can't see the message.

Can you test the script with the below command, and confirm you see some JSON output with your site details etc.
```py
# Login to Tesla cloud and print debug output
python3 tesla-history.py --login --debug
```
If it is still failing with an error *after the site details*, perhaps try deleting `tesla-history.auth` and then re-running the script with `--login` to re-create the auth token, and see if that resolves it?

If not, there are some changes I can make to the script which might help, as getting the current site time is not critical to it's operation. I originally did this to confirm the system time of my Powerwall was valid and correct (I'm quite paranoid about date/times and ensuring they are right!!).

Just speculating...(feedback welcome) the API call for SITE_DATA is requesting the "live status" of the system, as below:
```json
  "SITE_DATA": {
    "TYPE": "GET",
    "URI": "api/1/energy_sites/{site_id}/live_status",
    "AUTH": true
  },
```
Therefore I wonder: if a site had not reported "live status" for a period (e.g. loss of connectivity) this API call may return an error string?

It's basically guesswork with the Tesla cloud API unfortunately, since nothing is documented. I'll make some changes to work around this potential issue regardless, since it appears to have happened at least once, as in this case.

TechCrazi, 2022-10-21:
> > Getting this error.
> > ERROR: Failed to retrieve SITE_DATA - string indices must be integers
> 
> @TechCrazi - strange, not sure why you would be getting this error (first error reported!).
> 
> Can you do some tests to help diagnose this issue?
> 
> First off, I'm just wondering if this was a temporary problem and it has subsequently started working at a later time, or is this happening all the time?
> 
> The script is failing on this API call:
> 
> ```python
> # Retrieve site current time
> data = battery.api('SITE_DATA')['response']
> ```
> 
> The response returned is a string instead of the expected JSON response. The string may contain some clues as to why it is failing, however I did not account for this so we can't see the message.
> 
> Can you test the script with the below command, and confirm you see some JSON output with your site details etc.
> 
> ```python
> # Login to Tesla cloud and print debug output
> python3 tesla-history.py --login --debug
> ```
> 
> If it is still failing with an error _after the site details_, perhaps try deleting `tesla-history.auth` and then re-running the script with `--login` to re-create the auth token, and see if that resolves it?
> 
> If not, there are some changes I can make to the script which might help, as getting the current site time is not critical to it's operation. I originally did this to confirm the system time of my Powerwall was valid and correct (I'm quite paranoid about date/times and ensuring they are right!!).
> 
> Just speculating...(feedback welcome) the API call for SITE_DATA is requesting the "live status" of the system, as below:
> 
> ```json
>   "SITE_DATA": {
>     "TYPE": "GET",
>     "URI": "api/1/energy_sites/{site_id}/live_status",
>     "AUTH": true
>   },
> ```
> 
> Therefore I wonder: if a site had not reported "live status" for a period (e.g. loss of connectivity) this API call may return an error string?
> 
> It's basically guesswork with the Tesla cloud API unfortunately, since nothing is documented. I'll make some changes to work around this potential issue regardless, since it appears to have happened at least once, as in this case.

@mcbirse Here is the debug output of my login. I have removed some of my personal info for privacy. I have 2 sites under my account. One of them was the wrong one that Tesla set it up on my account but I can not figure out how to delete it. I also tested the "--site" parameter but the site ID that is getting logged in debug is not the right one and I am not sure how to get the right site ID. 

root@PowerWall:/Powerwall-Dashboard/tools# python3 tesla-history.py -l -d 
----------------------------------------
Tesla account: abcd@gmail.com
----------------------------------------
{
    "id": "1538000-25-F--TG121182002WXC",
    "site_name": "Ghar",
    "backup_reserve_percent": 35,
    "default_real_mode": "self_consumption",
    "installation_date": "2021-08-11T19:00:00-05:00",
    "user_settings": {
        "storm_mode_enabled": true,
        "powerwall_onboarding_settings_set": true,
        "sync_grid_alert_enabled": true,
        "breaker_alert_enabled": false
    },
    "components": {
        "solar": true,
        "solar_type": "pv_panel",
        "battery": true,
        "grid": true,
        "backup": true,
        "gateway": "teg",
        "load_meter": true,
        "tou_capable": true,
[... 54 lines cut ...]
        "car_charging_data_supported": false,
        "off_grid_vehicle_charging_reserve_supported": false,
        "vehicle_charging_performance_view_enabled": false,
        "vehicle_charging_solar_offset_view_enabled": false,
        "battery_solar_offset_view_enabled": true,
        "solar_value_enabled": true,
        "energy_value_header": "Energy Value",
        "energy_value_subheader": "Estimated Value",
        "energy_service_self_scheduling_enabled": true,
        "show_grid_import_battery_source_cards": true,
        "backup_time_remaining_enabled": true,
        "rate_plan_manager_supported": true,
        "battery_type": "solar_powerwall",
        "configurable": true,
        "grid_services_enabled": false,
        "ITC_cliff": 1
    },
    "battery_count": 2,
    "installation_time_zone": "America/Chicago",
    "max_site_meter_power_ac": 1000000000,
    "min_site_meter_power_ac": -1000000000,
    "geolocation": {
        "latitude": 32.9701219,
        "longitude": -96.5766098
    },

mcbirse, 2022-10-21:
@TechCrazi - Interesting!

Thanks, I can resolve this issue and should have an update to the program this weekend.

I suspect it is failing due to the additional (incorrect) site on your account. The "live_status" data retrieval to get current system time is probably failing, as the invalid site is not live / reporting data.

No problem - I'll make some changes which should allow it to report the sites without failure, and you will probably need to use the `--site` option to select the correct site (once fixed, it will output the site id correctly that you will need to use).

TechCrazi, 2022-10-21:
@mcbirse I called Tesla and they fixed my issue with duplicate sites. Now the script is working as expected...

mcbirse, 2022-10-22:
> @mcbirse I called Tesla and they fixed my issue with duplicate sites. Now the script is working as expected...

No worries @TechCrazi - good to hear it was fixed and thanks for the update!

I'll make a couple of small changes to the script regardless to try to work around this potential problem for any future users. If it requires contacting Tesla or an agent to get an invalid site removed, that could be a pain for some people.

jasonacox, 2022-10-23:
New home (URL) for this great tool: 
* GitHub: https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history 
* GH Pages: https://jasonacox.github.io/Powerwall-Dashboard/tools/tesla-history/

longzheng, 2022-10-25:
Just a heads up I had issues running the `python3 tesla-history.py --login` script on Windows using Git Bash. After I pasted the callback URL after login it would just get stuck/not do anything.

I used Windows command prompt instead and worked correctly. In case anyone else is trying to run this on Windows.

cwagz, 2022-10-29:
It seems like I have a period of time where my data is weird. Specifically, 2/25/22 to 2/28/22. This may have been a time when I had backed up and then reinstalled my dashboard. I am not really sure how to fix it, but I thought it would be appropriate to ask here. Basically, the usage / production that is logged for the day does not match the "realtime" usage. This created this couple of days where it looks like I was offline on the 12-month chart:

[image]

You can see on 2/26 a good example of the two measurements being off. I checked a range of time and there seems to be a problem with my data from 2/25 to 2/28. Both 2/24 and 3/1 match within reason.

[image]

How can I fix this in my data?

mcbirse, 2022-10-30:
@cwagz - I think I follow the issue you are seeing. Please correct me if I have misunderstood.

Your "realtime" data (as shown in the Energy Usage panel) appears to be correct, however the corresponding cumulative data that should be shown for those days in Home Usage, Solar Energy, and Solar Energy Year is NOT correct?

The data for the "Monthly Analysis" panels is queried from the kwh retention policy. The kwh data is extrapolated from the your "realtime" data, i.e. from autogen.

So if your data in autogen is all okay, you can run the queries that populate kwh, daily, and monthly, manually for that period (these are normally run automatically from the CQ definitions - continuous queries). I wonder how this issue has occurred?

Try the following to correct the data:

```sh
# Run the InfluxDB shell console
docker exec -it influxdb influx -database 'powerwall'
```

This allows you to connect to InfluxDB and manually run queries. You should see the below:

```
Connected to http://localhost:8086 version 1.8.10
InfluxDB shell version: 1.8.10
>
```

At the `>` prompt, then run the following queries (they need to be run in the order shown below - that is into kwh, then daily, then monthly).

Note - you need to edit the below queries for your timezone!!

Please replace the timezone strings with your own. And replace the timezone offset in the `WHERE time` clauses with your own offset for those dates (you may need to check what it was on those dates due to DST etc. For Los Angeles, I guess it was -8 on those dates, I think).

Paste queries at the `>` prompt, and press Enter.

SELECT from autogen into kwh:
```sql
SELECT integral(home)/1000/3600 AS home, integral(solar)/1000/3600 AS solar, integral(from_pw)/1000/3600 AS from_pw, integral(to_pw)/1000/3600 AS to_pw, integral(from_grid)/1000/3600 AS from_grid, integral(to_grid)/1000/3600 AS to_grid INTO kwh.:MEASUREMENT FROM autogen.http WHERE time >= '2022-02-25T00:00:00-08:00' AND time < '2022-03-01T00:00:00-08:00' GROUP BY time(1h), month, year tz('America/Los_Angeles')
```

SELECT from kwh into daily:
```sql
SELECT sum(home) AS home, sum(solar) AS solar, sum(from_pw) AS from_pw, sum(to_pw) AS to_pw, sum(from_grid) AS from_grid, sum(to_grid) AS to_grid INTO daily.:MEASUREMENT FROM kwh.http WHERE time >= '2022-02-25T00:00:00-08:00' AND time < '2022-03-01T00:00:00-08:00' GROUP BY time(1d), month, year tz('America/Los_Angeles')
```

SELECT from daily into monthly:
```sql
SELECT sum(home) AS home, sum(solar) AS solar, sum(from_pw) AS from_pw, sum(to_pw) AS to_pw, sum(from_grid) AS from_grid, sum(to_grid) AS to_grid INTO monthly.:MEASUREMENT FROM daily.http WHERE time >= '2022-02-01T00:00:00-08:00' AND time < '2022-03-01T00:00:00-08:00' GROUP BY time(365d), month, year
```

After running each query you should get a response like below. The "written" value will be different for each due to the different group by times.

Response example:
```
name: result
time written
---- -------
0    96
```

Hopefully, after doing this, when you check the graphs in Grafana the Analysis Data values will be corrected? 🤞

cwagz, 2022-10-30:
@mcbirse - Thank you for helping with this! You understood the problem perfectly. I ran your commands as written. The time zone and offset were correct for that time of year. The days now look good in Grafana.

I do not recall when I first started using the dashboard, but it was fairly close to where this bad data existed. I changed hardware several times in the beginning until I ended up with a setup I like that is also fast enough. I am guessing I had an incomplete backup or something from that time period that got moved onto new hardware. I really don't know though, this is my first experience with InfluxDB and I need to read up on how it works.

I was using a PC Engines APU2C4 for a few weeks, but it would struggle really bad if I tried to pull up multiple days' worth of data in Grafana. Would something like the problem I had occur with InfluxDB if the CPU was maxed out?

[image]

[image]

mcbirse, 2022-10-30:
@cwagz - Great to hear this fixed the issue!

> Would something like the problem I had occur with InfluxDB if the CPU was maxed out?

I suppose it is possible, but I'm not sure. It's a strange issue to have occurred, since the real time data was still logging okay. 

There is a difference with the CQ's for kwh, daily, monthly, in that compared to other CQ's they use a "RESAMPLE EVERY 1m" clause - maybe this resampling failed for some reason, like the high CPU load?

Actually, I just Googled for that and found reference to this possibility. If the query is taking longer than the resample time (1m in this case) then it will fail silently and the data in kwh etc. would not have been updated.

You mentioned it struggled to pull data into Grafana on that system, so it sounds like queries were taking a long time. So it seems highly likely the resample queries were failing and caused this issue.

mcbirse, 2022-10-31:
@cwagz or for anyone else's interest:

There is a way to check if CQ's are failing by querying InfluxDB's internal database. This could be useful for anyone setting up a new host and wanting to check if it is handling the load okay?

Unfortunately it appears to only store a week of history - so @cwagz we cannot confirm for sure what caused your issue.

```sh
# Connect to InfluxDB internal database
docker exec -it influxdb influx -database '_internal' -precision 'rfc3339'
```

Query first 5 cq records:
```sql
SELECT * FROM cq LIMIT 5 tz('Australia/Sydney')
```
Response showing queryFail and queryOk counts:
```
name: cq
time                      hostname queryFail queryOk
----                      -------- --------- -------
2022-10-23T11:00:00+11:00 influxdb 0         139012
2022-10-23T11:00:10+11:00 influxdb 0         139031
2022-10-23T11:00:20+11:00 influxdb 0         139031
2022-10-23T11:00:30+11:00 influxdb 0         139031
2022-10-23T11:00:40+11:00 influxdb 0         139031
```

Query last 5 cq records (most recent):
```sql
SELECT * FROM cq ORDER BY DESC LIMIT 5 tz('Australia/Sydney')
```

Response:
```
name: cq
time                      hostname queryFail queryOk
----                      -------- --------- -------
2022-10-31T11:02:30+11:00 influxdb 0         192
2022-10-31T11:02:20+11:00 influxdb 0         192
2022-10-31T11:02:10+11:00 influxdb 0         192
2022-10-31T11:02:00+11:00 influxdb 0         175
2022-10-31T11:01:50+11:00 influxdb 0         175
```

Maybe I should open another discussion thread at some point where we can document various InfluxDB tips & tricks.

cwagz, 2022-10-31:
> Maybe I should open another discussion thread at some point where we can document various InfluxDB tips & tricks.

This is probably a good idea. All of this information is very helpful. Thanks again.

chrisheninger, 2023-01-01:
@mcbirse thank you so much for the script and instructions– I ran it to backfill the entire past year and everything came through exactly as intended first try. I've wanted to run this dashboard for a while– and figured I would set it up for 2023. The fact I was just able to backfill everything with a single script is amazing. 🙇 🙌 

Thank you too for the dashboard @jasonacox, this is very neat! 🙇

mcbirse, 2023-01-01:
@chrisheninger - Thanks for the feedback, awesome to hear! Happy New Year! 🥳 🎉

jurassic410, 2023-01-12:
Hi Like @TechCrazi, i have registered on my account 2 powerwalls, one is an old system still registered in the account, (offline for months now), when listing the sites, the old one is listed correctly, when the script need to read the second powerwall info (the one that i currently own and that is running) i get this error:

`Get SITE_DATA for Site ID 2352913
{
    "response": ""
}
ERROR: Failed to retrieve SITE_DATA - string indices must be integers`

someone have some idea on what to try?

mcbirse, 2023-01-12:
Hi @jurassic410 - I have not seen this occur before, but it appears the request for SITE_DATA is returning an empty response, which is a bit odd.

This hopefully shouldn't be too difficult to account for, and that request is only used to get the current system time of your Powerwall (which, when not available, should fall back to the current system time of your host server instead).

I'll try to submit a PR for the program this weekend that should fix this issue.

jurassic410, 2023-01-14:
@mcbirse thank you, 8 will look forward to it, and let you know if the fix works!

mcbirse, 2023-01-14:
@jurassic410 - I have submitted a fix for this (hopefully). Once @jasonacox has merged the changes and you have pulled the update, let me know how it goes! 🤞 

Essentially the request for SITE_DATA has been changed, so if any exception at all occurs with this request the script will no longer exit.

The request was used simply to get the system time from the Powerwall, so is not very critical (and when unable to get this, the script would use the host system current time instead anyway).

jasonacox, 2023-01-14:
Changes merged!  Thanks for the fix here and in the pypowerwall `set-mode.py` and `set-reserve.py` tools!

jurassic410, 2023-01-14:
@mcbirse @jasonacox now works like a charm!
it will now asks correctly wich site to use.
thank you

mcbirse, 2023-03-30:
Hi @Jot18 - for solar only sites, this is a work in progress at the moment. There is a modified version of the dashboard under the "tools/solar-only" folder [here](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/solar-only) as well as modified tesla-history script that will work with solar sites.

Please give it a try - I would welcome some more feedback! Check/join the discussion #183 and let us know if it works for you. 🙏

Jot18, 2023-03-30:
> Hi @Jot18 - for solar only sites, this is a work in progress at the moment. There is a modified version of the dashboard under the "tools/solar-only" folder [...]

Tried that as well. Couldn't really understand how to use that one. I only had success getting data with c\[https://auth.tesla.com/oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=P5hztmHgtWMFMwv7mObCdIivp7ASFx&code_challenge=LJjPtbFTRO_tsq7ZP7yxJKMtORnP6IPA-sqBNMmwvm0&code_challenge_method=S256&login_hint=Gurmailsaini12%40yahoo.com](url) but that also couldn't download data for dec 2022, Jan/Feb 2023. It only downloads data for Oct/Nov 2022 & March 2023. I got solar in late oct 2023.

mcbirse, 2023-03-30:
@Jot18 - the script in the solar-only tools folder works pretty much the same. Make sure to read the [usage](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/solar-only/tesla-history) instructions for the script.

Can you post some output/details of what you are running, errors received, etc.?

Try starting with below to ensure your config is set up correctly and the script finds your Tesla energy site correctly.

```bash
# Run tesla-history login
python3 tesla-history.py --login
```

Also, you may need run setup.sh from the "tools/solar-only" directory to set up Powerwall-Dashboard for a solar only site first (rather than the main project). Please post in #183 for further help with the solar-only testing.

Jot18, 2023-03-31:
> 

> @Jot18 - the script in the solar-only tools folder works pretty much the same. Make sure to read the [usage](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/solar-only/tesla-history) instructions for the script.
> 
> Can you post some output/details of what you are running, errors received, etc.?
> 
> Try starting with below to ensure your config is set up correctly and the script finds your Tesla energy site correctly.
> 
> ```shell
> # Run tesla-history login
> python3 tesla-history.py --login
> ```
> 
> Also, you may need run setup.sh from the "tools/solar-only" directory to set up Powerwall-Dashboard for a solar only site first (rather than the main project). Please post in #183 for further help with the solar-only testing.

After pasting the URL, I get no tesla energy sites found. I did run from Solar-Only>Tesla-Hitory.

clukas1967, 2023-04-10:
THANK YOU to @mcbirse and the community so much for pulling this historical tool together.  I got PW Dashboard up and running a week ago, and over this past weekend I successfully fought through pulling in all my history since 2019.

I’m writing to share the extra step I had to do **given that I’m using a Synology DS1618+** to host my Docker instances.  Just like with PW Dashboard itself, there is some funkiness to using the Tesla history app.  Hopefully I can save someone the trouble I went through to pull my old data.

As a recap for other aspiring Synology hosters, definitely make sure you follow this procedure posted by the Community ([Docker issue on Synology NAS · jasonacox/Powerwall-Dashboard · Discussion #169 · GitHub]()).  In particular, make sure you:

```python
sudo synogroup --add docker
sudo chown root:docker /var/run/docker.sock
sudo synogroup --member docker <username>
``` 

And finally it is vital that you **hand modify the Grafana section of the file powerwall.yml** that spins up all the Docker instances.  If you do not do this step, what happens is that the Grafana container gets stuck in a perpetual restart loop.

- [ ] Edit “user 1000:1000” to change the default value of 1000 to whatever ID the username is
- [ ] For example, in my case it became “user 1026:1026”

After that it’s a breeze and the whole package comes up nicely and starts populating the Grafana dashboard.  So…. now on to pulling historical data from Tesla.

I can confirm that the current iteration of the tesla-history.py tool runs well, thanks again to @mcbirse for doing this.  However, the Synology Python install is abnormal in several respects that breaks the procedure:

- Synology bundles an old Python 2 with the OS, and then upgrades it automatically in the background with DSM updates.  It cannot be de-installed and you don’t want to modify it.
- This Python instance does not include PIP, which is required for the tesla-history.py tool.

_Note I am running latest DSM (DSM 7.1.1-42962 Update 4) which includes Python 3.8.12._

I attempted and had to roll back the following workarounds to get PIP running after they failed:

1. Install PIP as root (via both sudo -i and direct login as root)

    - This successfully installs PIP but due to Synology implementation will not run as anything other than root.  
    - Exiting sudo and checking for PIP results in this error:
[image]
    - Since I don’t run anybody’s code as root that doesn’t require it, this was a dead end for me.  Had to back out PIP completely with “sudo pip uninstall pip”
    - NOTE – **Synology specifically disclaims and warns again doing anything in Python as root** and throws lots of warnings about it.  You’ll see this error every time:
[image]

2.	Repeat above with various PIP installation techniques

    - Attempted “python3 -m ensurepip --default-pip”.  Failed and rolled it back.
    - Attempted “wget https://bootstrap.pypa.io/get-pip.py” followed by invoking “python3 get-pip.py”.   Failed and rolled it back.
    - When I say it failed, I mean that after leaving root and returning to my normal user CLI any attempt to invoke PIP failed with some version of the traceback error above.

3.	Install Python 3.9 separately and then manually install PIP in that instance.

    - Python 3.9 is available through the Synology Package Center for automatic download & install.  This works fine, and when it’s done you can do invoke either “python3” or “python3.9” from CLI depending on which instance you want to hit.
    - However, this package also does not include PIP.  Attempting to install it as a non-root user always failed.  Attempting as root repeated the experience above with built-in Python bundle.
    - Since Py3.9 is a fully optional bundle, I attempted to see if I could play around with the directly structure directly and then ran into a variety of other problems.  The two biggest were that the DSM Package Center installs everything as root only (everything in site-packages directory is 700).  And python3.9 is not added to the user $PATH and it seems to require an act of god to modify this.
    - So I backed out Py3.9 altogether since I don’t need it for anything else and the Synology folks periodically update their version so want to avoid conflicts.

I finally stopped beating my head against the wall and went with a Python virtual instance.  The reasons I didn’t do this out of the gate are (1) it seemed like having a working PIP would be useful for other things; (2) I haven’t any experience with Python VENVs yet; and finally (3) I did not appreciate that the tesla_history.py process is a one-and-done.  

So what ultimately worked just fine on my DS1618+ to pull my history was the following:

```
sudo -i
python3 -m venv <Virtual_Env_Name>
cd <Virtual_Env_Name>
source bin/activate
python3 -m pip install --upgrade pip
pip install python-dateutil teslapy influxdb
``` 

And from there you can run the entire tesla-history login and download procedure as laid out.  Note the whole thing is running inside the Python venv, but it doesn’t  matter since the way the script works is that it’s essentially a temporary buffer for data pulled via HTTPS.  Once the download is done for the specified time interval, it writes out everything it has to InfluxDB so there is no residual state required.  You can blow away the venv after you verify the history is showing up in Grafana.

I now have history all the way back to April of 2019 when my PW got installed, and I’m geeking out studying trends in the data.

[image]

Thanks again!

jasonacox, 2023-04-11:
GREAT writeup! Thanks @clukas1967 !

> sudo synogroup --add docker
> sudo chown root:docker /var/run/docker.sock
> sudo synogroup --member docker <username>
> 

I wonder if we should update the README.md instructions for Dashboard installation on Synology systems ([here](https://github.com/jasonacox/Powerwall-Dashboard#synology-nas-and-rootless-docker)).  Would you want to submit a PR for that so you can get credit?

One change:

> modify the Grafana section of the file powerwall.yml

That will break the upgrade.sh process.  Instead, you should be able to edit compose.env to do the same thing:

```
GRAFANAUSER="1000:1000"
```

Thanks!! 🙏

[46 more replies]