jasonacox/Powerwall-Dashboard #18: How to adjust Savings graph?

jgleigh, 2022-04-08:
What's the best way to change the settings for cost savings? I buy energy at ~$0.12 and sell it back at ~$0.05.

jasonacox, 2022-04-09:
HI @jgleigh  - you can edit the panel - click on the header to "Savings" and the dropdown has "Edit".

[image]

There are four queries below that have `0.19` (representing $0.19/kWh) - change all of them to `0.12` except for the ones that show `sum("to_grid") * 0.19` and instead, make those `0.05`.

[image]

After you finish the edit, save the dashboard by using the "save" icon at the top.

[image]

jgleigh, 2022-04-09:
Excellent, thanks!

closed 2022-04-09 by jgleigh

jasonacox, 2022-04-09:
Hi @jgleigh - There is one error in my instruction to you.  You should adjust the "to_grid" for the "B" query (the "Solar" savings = cost avoidance of powering the house by solar only) to be the same rate as the "from_grid":  `sum("to_grid") * 0.12`.  This is because it is not computing the revenue you are getting by selling back to the grid but subtracting out the represented power that is not going to the house.  Only "C" query should be changed to the `0.05` value.

[image]

Sorry for the confusion - I'll edit my instructions above for anyone else finding this.

jgleigh, 2022-04-11:
I realized it's a little more complicated because that same calculation gets used in D. I ended up just putting them all at $0.11 which is close enough for rough numbers.

jasonacox, 2022-04-12:
Great catch @jgleigh !   The "D" query would need to be updated to have both (one at 0.12 for subtraction and one at 0.05 for revenue).  I created `tgg` for the 0.05 value:

SELECT tgg + (s - tp - tg ) + fp FROM (SELECT sum("to_pw")  * 0.12 AS tp, sum("solar") * 0.12 AS s, sum("to_grid") * 0.05 AS tgg, sum("to_grid") * 0.12 AS tg, sum("from_pw") * 0.12 AS fp FROM kwh.http WHERE  $timeFilter tz('America/Los_Angeles'))