jasonacox/Powerwall-Dashboard #21: Telegraf Errors in new version

mcg88, 2022-04-19:
Hi-

The new version of Powerwall-Dashboard isn't working properly for me and telegraf has a bunch of these errors. Pypowerwall looks to be working correctly.

```
2022-04-19T14:29:30Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/freq]: invalid character 'E' looking for beginning of value

2022-04-19T14:29:35Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/freq]: invalid character 'E' looking for beginning of value

2022-04-19T14:29:35Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/temps/pw]: invalid character 'E' looking for beginning of value
```

jasonacox, 2022-04-23:
Hi @mcg88 

That indicates that it is getting the "ERROR!" response from pypowerwall.  What do you see when you browse over to these URLs (replace localhost with the address of the server running the dashboard):

http://localhost:8675/stats

This should show the version of pypowerwall running. It should be `{"pypowerwall": "0.4.0 Proxy t8"`.  If it is not, you can upgrade pypowerwall with this:

```bash
docker stop pypowerwall
docker rm pypowerwall
docker images | grep pypowerwall | awk '{print $3}' | xargs docker rmi -f
docker-compose -f powerwall.yml up -d
```

Also all of these (even if some error):

http://localhost:8675/freq
http://localhost:8675/temps/pw
http://localhost:8675/aggregates

If all of them error, it could also be that the credentials are wrong.   Re-run the `setup.sh` and make sure the password is entered correctly.

mcg88, 2022-04-26:
Thanks for the response @jasonacox 

Based on your recommendation, I found that pypowerwall was running 0.3.0 and I went ahead and upgraded it it. It is now on the correct version. Still having the same issue:

Stats endpoint is pasted below. Freq end point doesn't work. Temps and aggregates both appear to work. 

```
{
"pypowerwall": "0.4.0 Proxy t8",
"gets": 80,
"errors": 28,
"timeout": 0,
"uri": {
"/stats": 25,
"/soe": 13,
"/strings": 13,
"/temps/pw": 14,
"/aggregates": 15
},
"ts": 1651000438,
"start": 1651000344,
"clear": 1651000344
}
```

jasonacox, 2022-04-27:
Hi @mcg88 can you see if you can get to any of these (replace localhost with the IP address of the dashboard)?

```
http://localhost:8675/version
http://localhost:8675/vitals
```

I'm starting to suspect that the error for http://localhost:8675/freq could be due to some missing vital data that the proxy uses to get the frequencies.

You can also check for logs on the host:

```bash
docker logs pypowerwall
```

mcg88, 2022-04-27:
@jasonacox  

I can access both vitals and version endpoints.

Logs:

`----------------------------------------
----------------------------------------
Exception happened during processing of request from ('172.18.0.4', 56290)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/local/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "server.py", line 165, in do_GET
    fcv["METER_X_VL1N"] = d['METER_X_VL1N']
KeyError: 'METER_X_VL1N'
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('172.18.0.4', 56302)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/local/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "server.py", line 165, in do_GET
    fcv["METER_X_VL1N"] = d['METER_X_VL1N']
KeyError: 'METER_X_VL1N'
`

jasonacox, 2022-04-28:
@mcg88 Would you be able to paste part of your vitals?  I'm looking for the details of the TESYNC section. I suspect your system is using something else to report voltages.

```json
    "TESYNC--1493315-01-F--x": {
        "ISLAND_FreqL1_Load": 60.0,
        "ISLAND_FreqL1_Main": 60.0,
        "ISLAND_FreqL2_Load": 60.0,
        "ISLAND_FreqL2_Main": 60.0,
        "ISLAND_FreqL3_Load": 0.0,
        "ISLAND_FreqL3_Main": 0.0,
        "ISLAND_GridConnected": true,
        "ISLAND_GridState": "ISLAND_GridState_Grid_Compliant",
        "ISLAND_L1L2PhaseDelta": -256.0,
        "ISLAND_L1L3PhaseDelta": -256.0,
        "ISLAND_L1MicrogridOk": true,
        "ISLAND_L2L3PhaseDelta": -256.0,
        "ISLAND_L2MicrogridOk": true,
        "ISLAND_L3MicrogridOk": false,
        "ISLAND_PhaseL1_Main_Load": 0.0,
        "ISLAND_PhaseL2_Main_Load": 0.0,
        "ISLAND_PhaseL3_Main_Load": -256.0,
        "ISLAND_ReadyForSynchronization": true,
        "ISLAND_VL1N_Load": 120.5,
        "ISLAND_VL1N_Main": 120.5,
        "ISLAND_VL2N_Load": 120.5,
        "ISLAND_VL2N_Main": 120.5,
        "ISLAND_VL3N_Load": 0.0,
        "ISLAND_VL3N_Main": 0.0,
        "METER_X_CTA_I": 3.1485,
        "METER_X_CTA_InstReactivePower": -140.0,
        "METER_X_CTA_InstRealPower": 268.0,
        "METER_X_CTB_I": 2.7720000000000002,
        "METER_X_CTB_InstReactivePower": -90.0,
        "METER_X_CTB_InstRealPower": -275.0,
        "METER_X_CTC_I": 0.0,
        "METER_X_CTC_InstReactivePower": 0.0,
        "METER_X_CTC_InstRealPower": 0.0,
        "METER_X_LifetimeEnergyExport": 1707894.0,
        "METER_X_LifetimeEnergyImport": 3233722.0,
        "METER_X_VL1N": 119.84,
        "METER_X_VL2N": 120.11,
        "METER_X_VL3N": 0.0,
        "METER_Y_CTA_I": 0.0,
        "METER_Y_CTA_InstReactivePower": 0.0,
        "METER_Y_CTA_InstRealPower": 0.0,
        "METER_Y_CTB_I": 0.0,
        "METER_Y_CTB_InstReactivePower": 0.0,
        "METER_Y_CTB_InstRealPower": 0.0,
        "METER_Y_CTC_I": 0.0,
        "METER_Y_CTC_InstReactivePower": 0.0,
        "METER_Y_CTC_InstRealPower": 0.0,
        "METER_Y_LifetimeEnergyExport": 0.0,
        "METER_Y_LifetimeEnergyImport": 2.0,
        "METER_Y_VL1N": 119.85000000000001,
        "METER_Y_VL2N": 120.11,
        "METER_Y_VL3N": 0.0,
        "SYNC_ExternallyPowered": false,
        "SYNC_SiteSwitchEnabled": false,
        "alerts": [
            "SYNC_a001_SW_App_Boot"
        ],
        "componentParentDin": "STSTSM--1232100-00-E--x",
        "firmwareVersion": "bce98cade591cb",
        "lastCommunicationTime": 1651120300,
        "manufacturer": "TESLA",
        "partNumber": "1493315-01-F",
        "serialNumber": "x",
        "teslaEnergyEcuAttributes": {
            "ecuType": 259
        }
    },
```

mcg88, 2022-04-28:
Here is the TESYNC section. To note- I've been using the Dashboard for a while without any problems until it suddenly stopped working and throwing these errors. No hardware changes and no software updates aside from what may be automatic / over the air

```
"TESYNC--1118431-00-L--TG120040000M7S": {
"ISLAND_FreqL1_Load": 60,
"ISLAND_FreqL1_Main": 60,
"ISLAND_FreqL2_Load": 60.010000000000005,
"ISLAND_FreqL2_Main": 60.010000000000005,
"ISLAND_FreqL3_Load": 0,
"ISLAND_FreqL3_Main": 0,
"ISLAND_GridConnected": true,
"ISLAND_GridState": "ISLAND_GridState_Grid_Compliant",
"ISLAND_L1L2PhaseDelta": -256,
"ISLAND_L1L3PhaseDelta": -256,
"ISLAND_L1MicrogridOk": true,
"ISLAND_L2L3PhaseDelta": -256,
"ISLAND_L2MicrogridOk": true,
"ISLAND_L3MicrogridOk": false,
"ISLAND_PhaseL1_Main_Load": 0,
"ISLAND_PhaseL2_Main_Load": 0,
"ISLAND_PhaseL3_Main_Load": -256,
"ISLAND_ReadyForSynchronization": true,
"ISLAND_VL1N_Load": 119.5,
"ISLAND_VL1N_Main": 119.5,
"ISLAND_VL2N_Load": 118.5,
"ISLAND_VL2N_Main": 118.5,
"ISLAND_VL3N_Load": 0,
"ISLAND_VL3N_Main": 0,
"SYNC_ExternallyPowered": false,
"SYNC_SiteSwitchEnabled": true,
"componentParentDin": "STSTSM--1118431-00-L--TG120040000M7S",
"firmwareVersion": "bce98cade591cb",
"lastCommunicationTime": 1651154738,
"manufacturer": "TESLA",
"partNumber": "1118431-00-L",
"serialNumber": "TG120040000M7S",
"teslaEnergyEcuAttributes": {
"ecuType": 259
}
},
```

jasonacox, 2022-04-29:
Thank you @mcg88 !  The issue is that your payload doesn't include the "METER" data points that were assumed to be there for the /freq macro in the proxy.  I am going to adjust the proxy to ignore the missing data points and keep proceeding without error.  

Just to clarify, you said the dashboard stopped working.  The errors you are getting from pypowerwall should only impact the frequency and voltage graphs in the dashboard.  The other graphs should still be working.  Is that wrong?

mcg88, 2022-05-02:
@jasonacox 

My stats page is still showing errors

```
{
"pypowerwall": "0.4.0 Proxy t9",
"gets": 118,
"errors": 33,
"timeout": 0,
"uri": {
"/stats": 26,
"/soe": 17,
"/temps/pw": 17,
"/strings": 17,
"/freq": 24,
"/aggregates": 17
},
"ts": 1651500306,
"start": 1651500204,
"clear": 1651500204
}
```

Here is a screenshot of what my dashboard looks like. Note the top graph is populated but no subsequent graph is. 
[image]

jasonacox, 2022-05-03:
Thanks @mcg88  - Can you paste some of the telegraf errors you are seeing?   It really seems like there are some InfluxDB issues as some of those graphs below are just down-sampling of the raw data showing up in the main graph.  Check the Influxdb logs too.

mcg88, 2022-05-03:
Influxdb logs

```
ts=2022-05-03T04:10:01.121123Z lvl=info msg="Snapshot for path written" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE1byuG000 op_name=tsm1_cache_snapshot path=/var/lib/influxdb/data/powerwall/daily/39 duration=59.266ms
ts=2022-05-03T04:10:01.121232Z lvl=info msg="Cache snapshot (end)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE1byuG000 op_name=tsm1_cache_snapshot op_event=end op_elapsed=59.373ms
ts=2022-05-03T04:10:01.266142Z lvl=info msg="Cache snapshot (start)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE1bzhW000 op_name=tsm1_cache_snapshot op_event=start
ts=2022-05-03T04:10:01.353006Z lvl=info msg="Snapshot for path written" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE1bzhW000 op_name=tsm1_cache_snapshot path=/var/lib/influxdb/data/powerwall/monthly/6 duration=86.876ms
ts=2022-05-03T04:10:01.353067Z lvl=info msg="Cache snapshot (end)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE1bzhW000 op_name=tsm1_cache_snapshot op_event=end op_elapsed=86.935ms
ts=2022-05-03T04:10:04.276062Z lvl=info msg="TSM compaction (start)" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 op_event=start
ts=2022-05-03T04:10:04.276084Z lvl=info msg="Beginning compaction" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 tsm1_files_n=3
ts=2022-05-03T04:10:04.276088Z lvl=info msg="Compacting file" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 tsm1_index=0 tsm1_file=/var/lib/influxdb/data/powerwall/raw/36/000000001-000000001.tsm
ts=2022-05-03T04:10:04.276092Z lvl=info msg="Compacting file" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 tsm1_index=1 tsm1_file=/var/lib/influxdb/data/powerwall/raw/36/000000003-000000001.tsm
ts=2022-05-03T04:10:04.276095Z lvl=info msg="Compacting file" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 tsm1_index=2 tsm1_file=/var/lib/influxdb/data/powerwall/raw/36/000000004-000000001.tsm
ts=2022-05-03T04:10:04.834935Z lvl=info msg="Compacted file" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 tsm1_index=0 tsm1_file=/var/lib/influxdb/data/powerwall/raw/36/000000004-000000002.tsm.tmp
ts=2022-05-03T04:10:04.834975Z lvl=info msg="Finished compacting files" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 tsm1_files_n=1
ts=2022-05-03T04:10:04.834981Z lvl=info msg="TSM compaction (end)" log_id=0aDHAezW000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0aE1cAT0000 op_name=tsm1_compact_group db_shard_id=36 op_event=end op_elapsed=558.924ms
ts=2022-05-03T04:33:06.405200Z lvl=info msg="Retention policy deletion check (start)" log_id=0aDHAezW000 service=retention trace_id=0aE2wXPG000 op_name=retention_delete_check op_event=start
ts=2022-05-03T04:33:06.406512Z lvl=info msg="Retention policy deletion check (end)" log_id=0aDHAezW000 service=retention trace_id=0aE2wXPG000 op_name=retention_delete_check op_event=end op_elapsed=1.326ms
ts=2022-05-03T05:03:05.195171Z lvl=info msg="Retention policy deletion check (start)" log_id=0aDHAezW000 service=retention trace_id=0aE4eJvl000 op_name=retention_delete_check op_event=start
ts=2022-05-03T05:03:05.195388Z lvl=info msg="Retention policy deletion check (end)" log_id=0aDHAezW000 service=retention trace_id=0aE4eJvl000 op_name=retention_delete_check op_event=end op_elapsed=0.284ms
ts=2022-05-03T05:10:00.685134Z lvl=info msg="Cache snapshot (start)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE52fwG000 op_name=tsm1_cache_snapshot op_event=start
ts=2022-05-03T05:10:00.754709Z lvl=info msg="Snapshot for path written" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE52fwG000 op_name=tsm1_cache_snapshot path=/var/lib/influxdb/data/powerwall/daily/39 duration=69.601ms
ts=2022-05-03T05:10:00.754910Z lvl=info msg="Cache snapshot (end)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE52fwG000 op_name=tsm1_cache_snapshot op_event=end op_elapsed=69.792ms
ts=2022-05-03T05:10:00.901015Z lvl=info msg="Cache snapshot (start)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE52gmG000 op_name=tsm1_cache_snapshot op_event=start
ts=2022-05-03T05:10:00.972859Z lvl=info msg="Snapshot for path written" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE52gmG000 op_name=tsm1_cache_snapshot path=/var/lib/influxdb/data/powerwall/monthly/6 duration=71.858ms
ts=2022-05-03T05:10:00.972890Z lvl=info msg="Cache snapshot (end)" log_id=0aDHAezW000 engine=tsm1 trace_id=0aE52gmG000 op_name=tsm1_cache_snapshot op_event=end op_elapsed=71.887ms
ts=2022-05-03T05:10:01.907075Z lvl=info msg="TSM compaction (start)" log_id=0aDHAezW000 engine=tsm1 tsm1_level=1 tsm1_strategy=level trace_id=0aE52khl000 op_name=tsm1_compact_group db_shard_id=6 op_event=start
ts=2022-05-03T05:10:01.907122Z lvl=info msg="Beginning compaction" log_id=0aDHAezW000 engine=tsm1 tsm1_level=1 tsm1_strategy=level trace_id=0aE52khl000 op_name=tsm1_compact_group db_shard_id=6 tsm1_files_n=8
[... cut ...]
2022-04-23T21:31:20Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/temps/pw]: invalid character 'E' looking for beginning of value
2022-04-23T21:31:25Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/freq]: invalid character 'E' looking for beginning of value
2022-04-23T21:31:25Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/temps/pw]: invalid character 'E' looking for beginning of value
2022-04-23T21:31:30Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/temps/pw]: invalid character 'E' looking for beginning of value
```

jasonacox, 2022-05-04:
Hi @mcg88 - Thanks for these!  The errors in telegraf and what you see in the grafana dashboard has me puzzled. Was it working before the upgrade to the new version?

If you can, I would recommend setting up a clean install on another system (if you have one).  Otherwise, you can archive and try a new build on your current host.  Your installation may vary, but I would do something like this:

```bash
# Clean up old containers and file system
echo "Stopping and Cleaning up old containers..."
docker-compose -f powerwall.yml down
docker-compose -f powerwall.yml rm
docker rm pypowerwall
docker images | grep pypowerwall | awk '{print $3}' | xargs docker rmi -f
docker rm telegraf
docker images | grep telegraf | awk '{print $3}' | xargs docker rmi -f
docker rm grafana
docker images | grep grafana | awk '{print $3}' | xargs docker rmi -f
docker rm influxdb
docker images | grep influxdb | awk '{print $3}' | xargs docker rmi -f

# Archive Old Setup
cd ..
mv Powerwall-Dashboard Powerwall-Dashboard.old

# Pull New Install
echo "Cloning from github..."
git clone https://github.com/jasonacox/Powerwall-Dashboard.git
cd Powerwall-Dashboard

# Setup
echo "Starting setup..."
./setup.sh
```

jasonacox, 2022-09-25:
Closing this for now - feel free to re-open if the issue persists and we can troubleshoot further.

closed 2022-09-25 by jasonacox