jasonacox/Powerwall-Dashboard #23: pypowerwall error: Max Retries...

InstigatorX, 2022-04-26:
Ran both setup methods and getting an error with pypowerwall. System is Ubuntu 21.10. Hostname not set?

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "server.py", line 66, in <module>
    pw = pypowerwall.Powerwall(host,password,email,timezone)
  File "/usr/local/lib/python3.7/site-packages/pypowerwall/__init__.py", line 114, in __init__
    self._get_session()
  File "/usr/local/lib/python3.7/site-packages/pypowerwall/__init__.py", line 121, in _get_session
    r = requests.post(url,data = pload, verify=False, timeout=self.timeout)
  File "/usr/local/lib/python3.7/site-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/sessions.py", line 529, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/sessions.py", line 645, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='hostname', port=443): Max retries exceeded with url: /api/login/Basic (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fde0e77cad0>: Failed to establish a new connection: [Errno -3] Try again'))

jasonacox, 2022-04-27:
Hi @InstigatorX , you are right, the issue is that it is not getting the right address for your Powerwall
```
HTTPSConnectionPool(host='hostname', port=443):
```

What did you use for the "IP Address" of your powerwall during the setup?  Look at the `pypowerwall.env` file to make sure it includes a valid IP address for your powerwall:

```python
PW_EMAIL=email@example.com
PW_PASSWORD=password
PW_HOST=10.0.1.2
PW_TIMEZONE=America/Los_Angeles
PW_DEBUG=no
```

I see errors in the README instructions that has the wrong format for the env file.  I will update that but the `setup.sh` should have built the right env file.

jasonacox, 2022-06-11:
Closing this for now.  Please re-open if there is more to add or fix.  Thanks for your help!

closed 2022-06-11 by jasonacox