jasonacox/Powerwall-Dashboard #24: TZ setting issue in dashboard.json

jaydkay, 2022-05-01:
Hi @jasonacox,

not sure if this is an actual issue or if I just don't understand how the dashboard actually works: I installed and set up the dashboard the usual way via _./setup.sh_. This included setting the correct timezone, _Europe/Berlin_ in my case. I checked that _tz.sh_ acually changed the timezone in the four referenced files.

But when I look at the values of the dashboard, e.g. _Solar Energy_ and click on _Edit_, the TZ shown in the following settings menu fr this signal is _America/Los Angeles_. This is also true for every other value on the dashboard. This is not obvious from todays values, but correcting the timezone right inside the dashboard for every signal shown slightly changes the values e.g. for the bar graphs in the monthly analysis.

Is there some problem with the definition of the signals inside _dashboard.json_ or is this working as intended?

jasonacox, 2022-05-01:
Hi @jaydkay - thanks for the feedback.   Some thoughts and questions for clarification:

1) Are you say saying that the dashboards look *correct* but when you edit them you see "America/Los Angeles" timezones listed?

2) Can you check your Grafana setup to see what it is set to use?  Here is how:

Click on settings wheel: 
[image]

Under general check to see if it is set to use browser time:

[image]

I would be interested to know what it shows for you.

3) When you check the dashboard.json file, do you see that the timezone data is correct (Europe/Berlin)?

jaydkay, 2022-05-03:
Hi @jasonacox, sorry, I checked again and it was my mistake. We can close the issue, just a case of not following the instructions to the word...

closed 2022-05-03 by jaydkay