jasonacox/Powerwall-Dashboard #29: Power Flow Animation setup

jgleigh, 2022-05-24:
I'm trying out the new animation dashboard, but the Power Flow panel is just black. Is there something special needed to get this new dashboard working?

jgleigh, 2022-05-25:
Looks like there is an "ERROR!" in the box. It was black text so kind of hidden.

jasonacox, 2022-05-27:
Hi @jgleigh that sounds like you are not running the latest pyPowerwall container.   Check the version:  http://10.0.0.10:8675/stats (replace 10.0.0.10 with the address of the host running the dashboard). 

It should show: `"pypowerwall": "0.4.0 Proxy t13`

Also, you should see the animation here:  http://10.0.0.10:8675/ and an iframe version here: http://10.0.0.10:8675/example.html

You can force an upgrade with:

```bash
# Delete pyPowerwall for Upgrade
echo ""
echo "Delete and Upgrade pyPowerwall to Latest"
docker stop pypowerwall
docker rm pypowerwall
docker images | grep pypowerwall | awk '{print $3}' | xargs docker rmi -f

# Restart Stack
echo "Restarting Powerwall-Dashboard stack..."
docker-compose -f powerwall.yml up -d
```

jgleigh, 2022-05-27:
Much better. Guess I didn't notice that pypowerwall had been upgraded since it was still on 0.4.0. One small graphics glitch with wide browser windows.
[image]

jasonacox, 2022-05-27:
The graphic glitch you see is a scroll-bar.  This is fixed by this PR https://github.com/jasonacox/pypowerwall/pull/15 (connecting this issue to that fix).  I don't see it in Grafana, but I plan to release a new "Proxy t14" container later today or tomorrow.  Thanks for raising the issue, @jgleigh

jgleigh, 2022-05-27:
You rock...thanks! I'll pull in t15 when it's released.

closed 2022-05-27 by jgleigh

jgleigh, 2022-05-27:
t14 is working. Thanks again.