jasonacox/Powerwall-Dashboard #31: Gaps in data

Homer-Rivera, 2022-06-02:
This is a great project!  I am using powerwall-dashboard in docker.  It seems that I have gaps in my data. 

[image]

Checking the logs from pypowerwall I see this...

`Traceback (most recent call last):
  File "/usr/local/lib/python3.8/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.8/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.8/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.8/http/server.py", line 415, in handle_one_request
      
    method()
  File "server.py", line 172, in do_GET
    vitals = pw.vitals()
  File "/usr/local/lib/python3.8/site-packages/pypowerwall/__init__.py", line 262, in vitals
    stream = self.poll('/api/devices/vitals')
  File "/usr/local/lib/python3.8/site-packages/pypowerwall/__init__.py", line 193, in poll
    payload = r.raw.data
  File "/usr/local/lib/python3.8/site-packages/urllib3/response.py", line 303, in data
    return self.read(cache_content=True)
  File "/usr/local/lib/python3.8/site-packages/urllib3/response.py", line 544, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/usr/local/lib/python3.8/contextlib.py", line 131, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.8/site-packages/urllib3/response.py", line 446, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='10.XX.XX.XX', port=443): Read timed out.`

Looking at the doc from the pypowerwall do you think that turning on the proxy might solve this issue?

Homer-Rivera, 2022-06-02:
Well, it looks like the proxy is trying to run but I get this error, and then the container restarts.

`pyPowerwall [0.4.0] Proxy Server [t14] Started - HTTP Port 8675
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/pypowerwall/__init__.py", line 109, in __init__
    f = open(self.cachefile, "r")
FileNotFoundError: [Errno 2] No such file or directory: '.powerwall'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "server.py", line 88, in <module>
    pw = pypowerwall.Powerwall(host,password,email,timezone)
  File "/usr/local/lib/python3.8/site-packages/pypowerwall/__init__.py", line 114, in __init__
    self._get_session()
  File "/usr/local/lib/python3.8/site-packages/pypowerwall/__init__.py", line 125, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.8/site-packages/requests/cookies.py", line 328, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.8/site-packages/requests/cookies.py", line 399, in _find_no_duplicates
    raise KeyError('name=%r, domain=%r, path=%r' % (name, domain, path))
KeyError: "name='AuthCookie', domain=None, path=None"
`

jasonacox, 2022-06-02:
Hi @Homer-Rivera - the error messages you are seeing are caused by pyPowerwall having trouble reaching the IP address of your Powerwall and eventually times out. 

> HTTPSConnectionPool(host='10.XX.XX.XX', port=443): Read timed out.

Check the network connection between the host running  the dashboard and the Powerwall itself (e.g. long running `ping`).  I suspect it is intermittent which is why you are getting some of the data for the graph but have gaps.  I had this same issue. 
In my case the Powerwall was on WiFi and showing a strong signal, but because it was at the edge of the WiFi range, it would intermittently disappear.  I fixed it by hardwiring the Powerwall gateway into my network.  Since then, I have had zero dropouts.

If running an ethernet cable to the Powerwall is not possible in your circumstance, there may be another option.  I have thought about adding additional retry logic and configurable timeout for the proxy.  It won't be 100% but would likely give you better coverage.  Let me know if you are interested in trying that and I'll give you a way to test.

Homer-Rivera, 2022-06-02:
You nailed it!  I ran an ethernet cable to the Powerwall and now everything is consistent with no gaps. 

Thanks!