jasonacox/Powerwall-Dashboard #32: Solar generation giving negative values.

externalz, 2022-06-02:
Been running this setup for 5 days now and i noticed my solar saving values and self powered values being zero or negative in the savings graph.

Not sure if a bug or a configuration error

[image]

jasonacox, 2022-06-02:
Hi @externalz  - Nice graph!  Lots going on here.  Technically what you are seeing is a "bug".  You had something like a _Storm Watch_ charge event that used grid power to charge up your Powerwalls at about 21:00.  

The cost calculation doesn't know how to factor in a `grid>powerwall` flow.  What happens is that it basically subtracts it out from the `solar>home` bucket (since it only knows about "to_pw" and assumes it was solar doing that).  Technically it's not wrong (you paid for that grid usage) but makes it a bit confusing by subtracting it from `solar>home`.

Here are the equations used to calc the savings with variable we can pull from history (sum):
* tp = to powerwall
* tg = to grid
* fp = from powerwall
* s = from solar

```sql
# powerwall > home - assumes all power from PW (charged from solar) = savings
SELECT fp FROM (SELECT sum("from_pw")  * 0.19 AS fp FROM kwh.http WHERE  $timeFilter tz('America/Los_Angeles'))

# solar > home - assumes all solar not going to PW or grid is going to the home = savings
SELECT s - (tp + tg) FROM (SELECT sum("solar") * 0.19 AS s,  sum("to_pw")  * 0.19 AS tp, sum("to_grid") * 0.19 AS tg FROM kwh.http WHERE  $timeFilter tz('America/Los_Angeles'))

# solar > grid - assumes all power going to grid = savings
SELECT tg FROM (SELECT sum("to_pw")  * 0.19 AS tp, sum("solar") * 0.19 AS s, sum("to_grid") * 0.19 AS tg FROM kwh.http WHERE  $timeFilter tz('America/Los_Angeles'))

# total - sum of the three above
SELECT tg + (s - tp - tg ) + fp FROM (SELECT sum("to_pw")  * 0.19 AS tp, sum("solar") * 0.19 AS s, sum("to_grid") * 0.19 AS tg, sum("from_pw") * 0.19 AS fp FROM kwh.http WHERE  $timeFilter tz('America/Los_Angeles'))
```

I'm not sure how to accomodate a _Storm Watch_ type charge events.  I'm open to ideas.  

Thanks for sending!

PS -  separate item.  I noticed that you have a slight older dashboard-animation.json running where the alert threshold was too short which is causing some markers on the grid graph:
[image]

It's fine, but if you want to fix going forward so that it only places markers when you have an actual power outage, you can download and import the latest [dashboard-animation.json ](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/dashboard-animation.json)or edit that panel with these settings:
[image]

externalz, 2022-06-02:
Thank you so much for this detailed reply.

Now that you have explained how this works, i have a better understanding of the calculations. The pulling from the grid was from my time of use settings as its winter here with a low solar output.
Adding this info to the wiki might be a good idea?

Lastly this may require a seperate thread but the grid flow animation when using a reverse proxy times out like this picture.
Is this something that can be fixed?
[image]

Viewing it locally is fine.

jasonacox, 2022-06-03:
Thanks @externalz !   

I agree, we should add a comment about how the "savings" is calculated.  I'll see what I can do to get that added to the README.

On the disappearing animation, two thoughts:

1) The only time I have seen that is if you are loading the same animation in another browser window on the same machine.  It seems that the iFrames do not like to share.  Only one will work.  Load the dashboard in just one window and see if that helps.

2) If that is not your case, what reverse proxy are you using or did you mean pyPowerwall?  The iFrame renders a javascript animation from the Powerwall web portal that makes many API calls directly back to the pyPowerwall proxy (those API calls are how it determines power flow values, etc.).  It could be that some connectivity issues are happening causing a timeout.  If you force a refresh, does it start work?  Also, what browsers are you using?  I tested in Chrome, Safari and Brave.  Sorry for all the questions, but would be helpful to know for any others seeing the same thing.

Thanks again for the feedback.

externalz, 2022-06-03:
Thanks for the tips Jason, i use HAproxy on pfsense to reverse proxy into my VM's running on seperate VLANS.
Everything else works correctly except the powerflow meter, (took too long to respond) error.
Tried all the usual suspects, different browsers, cache clearing etc
I guess  its a problem with my HAProxy setup, when i get time to fix and troubleshoot i will reply back here if i find anything.

Thanks again.

closed 2022-06-03 by externalz