jasonacox/Powerwall-Dashboard #36: How to create an alert when powerwall reaches a certain percentage.

Homer-Rivera, 2022-06-12:
I see that the alert tab shows up on the new grid status panel.  How would one go about adding an alert tab to the powerwall percentage panel?

[image]

jasonacox, 2022-06-12:
Hi @Homer-Rivera - That Grafana panel doesn't have an alert function, unfortunately.  I believe alerts need to be added to time series data graphs (allowing for warning, alert, recovery markers).  I'm not an expert on Grafana alerts so there may still be a way.

Here is an alternative.  The main *Energy Usage* graph has a charge level ("E") that you could use for alerting.

[image]

Tip: One thing I do know about alerting, you need to "Save" the dashboard before alerts activate.

Homer-Rivera, 2022-06-13:
That will work.  Thanks!

closed 2022-06-13 by Homer-Rivera