jasonacox/Powerwall-Dashboard #37: Powerwall dashboard shows lower battery level then gateway animation from powerwall direct.

Homer-Rivera, 2022-06-12:
I noticed today that the tesla imbedded powerflow grid was showing 51% full on backup battery and when I switched to the powerwall default panel with the gradient powerwall the backup battery was reading 48%.  I have 3 powerwalls and I was not sure if the value provided by pypowerwall was 1 specific battery or an average of all of them.  Is there any reason why the numbers would be different?

[image]

[image]

jasonacox, 2022-06-12:
Hi @Homer-Rivera - Thanks for spotting this issue.   The percentages are for "all of the Powerwalls" in aggregate.  The issue, which I didn't even think about until you raised it, is that the animation is coming direct from the Powerwall gateway and it is not adjusted for the 5% buffer Tesla reserves of the battery for the app view.

### Background - Tesla 5% Reserve
 
Tesla reserves 5% of the battery so that it will never go to 0%.  They recompute the % based on the top 95% and this is how it is calculated:

```python
battery_for_user = ( (real_battery_level / 0.95) - (5 / 0.95) )
```

The 51% is the actual real value (real_battery_level) and the 48% in your example is the adjusted data (battery_for_user). If you check the Tesla App (smartphone app), you will see that it matches the battery_for_user value (48%).

### The Fix

There are some ways we could address this:
1. Switch to "real value" numbers everywhere so that they are consistent in the graphs.  This however will mean that it won't match the Tesla App.  
2. Figure out a way to adjust the pass-through animation to represent the adjusted value. I have an ideas on how I could intercept the calls and recompute the values.  It would mean that it isn't consistent with the gateway data but since everything else in the dashboard is base on the 95% at least we would have consistency. 

I assume we would want to keep the data consistent with the app so I'll start working on that.   Let me know if you have any other thoughts.

jasonacox, 2022-06-12:
Update: I have a possible fix.  I'm going to test it tonight and if all goes well, release it tomorrow.

jasonacox, 2022-06-12:
The fix looks good.  @Homer-Rivera  you can upgrade to the latest pypowerwall proxy (t15) by running the `upgrade.sh` script in the Powerwall-Dashboard directory or you can do it manually (also from Powerwall-Dashboard directory):

```bash
# Delete old pyPowerwall for Upgrade
docker stop pypowerwall
docker rm pypowerwall
docker images | grep pypowerwall | awk '{print $3}' | xargs docker rmi -f

# Restart Stack
docker-compose -f powerwall.yml up -d
```

The battery levels between the two may still be slightly off (1% or so) at times because the *Flow Animation* is pulling live data and the Grafana elements can be a minute or so behind.

Thanks again for raising this.  Please let me know how this is working for you.

Homer-Rivera, 2022-06-13:
Looking good.  The powerwall values are currently matching.  Thanks for the explanation on the 5% reserve.

closed 2022-06-13 by Homer-Rivera