jasonacox/Powerwall-Dashboard #38: Upgrade.sh isn't updating telegraf

jgleigh, 2022-06-12:
I noticed that upgrade.sh now includes a section for upgrading telegraf, but I'm wondering if it's missing something. I've run the newest upgrade.sh version and my telegraf is still on 1.22.0 when the latest is 1.22.4. Does the upgrade.sh script need something like this?

docker images | grep telegraf | awk '{print $3}' | xargs docker rmi -f

jasonacox, 2022-06-13:
Hi @jgleigh - You are right!  Actually the `upgrade.sh` script was only causing telegraf to reload configuration changes (new data points to pull in).  But you are right, if we want to force a telegraf upgrade, removing the old images would be the way to do that.  I'll push an update.

jasonacox, 2022-06-13:
I ran `git pull` to get the latest upgrade.sh script and ran it.  It pulled the latest:

```
pi@piserver:~/Powerwall-Dashboard $ docker logs telegraf
2022-06-13T03:08:53Z I! Using config file: /etc/telegraf/telegraf.conf
2022-06-13T03:08:53Z I! Starting Telegraf 1.22.4
```

Let me know if it works for you too,  Thanks @jgleigh !

jgleigh, 2022-06-13:
Excellent. Thanks for the quick fix.

closed 2022-06-13 by jgleigh