jasonacox/Powerwall-Dashboard #44: Add Tool for Node Red Upload to PVoutput

damclarkd, 2022-06-26:


jasonacox, 2022-07-04:
Thanks @damclarkd - I'm sure others in the community could use this.  

This is the first time using NodeRed but I seem to be doing something wrong importing your JSON.  What am I missing?

[image]

damclarkd, 2022-07-05:
Not doing anything wrong. You just don’t have some of the elements downloaded into your Node Red.

All these are added/managed from the Node Red Pallette Manager

Cronplus - https://flows.nodered.org/node/node-red-contrib-cron-plus <https://flows.nodered.org/node/node-red-contrib-cron-plus>

Moment - https://flows.nodered.org/node/node-red-contrib-moment <https://flows.nodered.org/node/node-red-contrib-moment>

API Current State - https://flows.nodered.org/node/node-red-contrib-home-assistant-websocket <https://flows.nodered.org/node/node-red-contrib-home-assistant-websocket>

This HA Websocket collection is specific to using entities from HA. How I am using them in this case is to get the current state of each entity I want to push up to PVOutput.

Adding these nodes to your instance should then clear all the unknown node types. The entities you use in api-current-state will need to be changed to your data. If your data does not come from HA, you will need to replace those nodes with whatever your data sources are.

HTH,

Cheers
Dennis

jasonacox-sam, 2026-08-31:
This shared a Node-RED flow for uploading dashboard data to PVOutput, adapted from the PVOutput forum example. The import question was resolved (the flow needs palette nodes like cron-plus and moment installed first).

Closing since nothing is outstanding. The flow JSON stays in this thread's history for anyone who wants it — thanks @damclarkd for sharing it.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam