jasonacox/Powerwall-Dashboard #53: iframe powerflow

Flying-Ace, 2022-07-20:
unable to populate the powerflow iframe. 
note my install in non-standard. the docker pypowerwall.yml did not work for me (firewall and VLANs) so I installed each container manually in bridge mode, and edited the telegraf hostname to the ip of the influx and pypowerwall.  everything is working except the iframe. 

pypowerwall_ip:8675 shows the powerflow image
pypowerwall_ip:8675/stats = {"pypowerwall": "0.5.0 Proxy t16", "gets": 5554, "errors": 0, "timeout": 0, "uri"

can i get manual steps to getting the iframe working (ie: what do i need to modify to make it work, where to put my custom ip, etc)
thanks

[image]

Flying-Ace, 2022-07-21:
im literally shaking in the legs with my hand over my mouth. that did the trick! I spent hours troubleshoot all my errors and this was the last one of them.
🙏🙏🙏🙏🙏🙏🙏🙏🙏
-> resolved

closed 2022-07-21 by Flying-Ace

jasonacox, 2022-07-22:
Congrats, Ace!  Thanks for the update.  

In hindsight, is there anything we could add to the instructions that would have helped on setup, or better yet, what could we change that would help the setup script work on your network?