jasonacox/Powerwall-Dashboard #54: Connection refused to port 8675, RPi

maxEntropyProd, 2022-07-25:
Hi 
setup.sh seemed to work fine while installing on a RPi4; however, when I pull up Powerwall Power Flow dashboard, the Power Flow window shows "Unable to connect" to localhost:8675. If I point a browser to that port, I get a similar message about site can't be reached. 
Any suggestions? Note, my powerwall+ and solar was just installed, but the inverter has some issues and so the system is in standby as they try to troubleshoot that. However, if I log into the gateway it shows the Tesla graphic with power being supplied by the grid. So that's working. 
Thanks!

maxEntropyProd, 2022-07-25:
Sorry, I've moved this to discussion, as it's very likely the problem is on my end.

closed 2022-07-25 by maxEntropyProd

jasonacox, 2022-07-25:
Hi @maxEntropyProd thanks for opening the issue!   A few things to try:

1) Ensure you do not have another browser window/tab open that is showing the animation.  Chrome and other browsers prevent the iframe from rendering if it opened more than once (I would like someone to explain this to me at some point :)
2) Several of the browsers do not like iframes on "localhost" so try to use the IP address instead (e.g. http://10.0.1.2:9000/).
3) Try to view the animation directly from the pyPowerwall proxy by going to this: http://10.0.1.2:8675/ - also make sure the latest pypowerwall was installed (e.g. v0.5.1): http://10.0.1.2:8675/stats (of course replace the IP address with your RPi)

jasonacox, 2022-07-25:
Moving to https://github.com/jasonacox/Powerwall-Dashboard/discussions/55