jasonacox/Powerwall-Dashboard #63: Running Dashboard on Windows

jasonacox, 2022-08-13:
** Message from Jeff **

> Very impressed with this dashboard mate – its awesome.
>  
> Had a few challenges to get it all running as I’m not really a dev – more of a lifetime tweaker with a Wintel background but a bit of Un*x – and haven’t really dabbled in Docker before (but thought it might help others if you were interested):
>  
> * Installed Docker Windows
> * Installed Git Windows
>  
> Ran the setup.sh under a bash shell – received an initial error:
>  ```
> Using Australia/Melbourne timezone...
> -----------------------------------------
> Running Docker-Compose...
> Starting pypowerwall ...
> Recreating influxdb  ...
> Starting pypowerwall ... done
> Recreating influxdb  ... done
> Recreating telegraf  ...
> Recreating grafana   ...
> Recreating telegraf  ... done
> Recreating grafana   ... done
> -----------------------------------------
> Waiting for InfluxDB to start...
> . up!
> Setup InfluxDB Data for Powerwall...
> **the input device is not a TTY.  If you are using mintty, try prefixing the command with 'winpty'**
>  ```
> Used Notepad++ to edit the setup.sh file and add winpty to the docker exec line – then received this error:
>  
> ```
> Setup InfluxDB Data for Powerwall...
> ERROR: open C:/Program Files/Git/var/lib/influxdb/influxdb.sql: no such file or directory
>  ```
>
> I got stuck here initially as for the life of me I couldn’t provide the correct path – but then thought I could run the command within the container itself – so started a CLI from the influxdb container in Docker Desktop – found the file in the correct path and run the influxdb command there.
>  
> And BINGO – started working.
>  
> My thoughts are that there is something to do with the path or winpty getting access to the container path correctly?  Seems to prefix the Git path and fail.  Could do a bit more troubleshooting but am enjoying the dashboard too much atm.
>  
> Anyway - thought this might help others if you wanted to put onto the page or you probably can fix up the setup script in 2 seconds anyway.

jasonacox, 2022-08-13:
I don't have a Windows PC to test but I'm incredibly happy to hear that the Dashboard will run on a Windows PC using Docker!

The error that came up during setup was due to this step in the `setup.sh` script:

```bash
echo "Setup InfluxDB Data for Powerwall..."
docker exec -it influxdb influx -import -path=/var/lib/influxdb/influxdb.sql
```

The `-it` switch in the docker command tells it to run interactively and use a TTY term for output.  Apparently this is not supported directly by Windows.  I believe the command will run correctly without that even though output will not be shown.  I'll run some tests and it it works, I'll remove that to help other Windows users wishing to run this Dashboard.

```bash
docker exec influxdb influx -import -path=/var/lib/influxdb/influxdb.sql
```

Jeff - thanks for raising this issue!  Good job on getting it to work!  👍

jasonacox, 2022-08-13:
Thanks @jaffro2k !  

I don't have a Windows 11 PC to test but I'm incredibly happy to hear that the Dashboard will run on a Windows PC using Docker!

The error that came up during setup was due to this step in the `setup.sh` script:

```bash
echo "Setup InfluxDB Data for Powerwall..."
docker exec -it influxdb influx -import -path=/var/lib/influxdb/influxdb.sql
```

The `-it` switch in the docker command tells it to run interactively and use a TTY term for output.  Apparently this is not supported directly by Windows.  I believe the command will run correctly without that even though output will not be shown.  I'll run some tests and it it works, I'll remove that to help other Windows users wishing to run this Dashboard.

```bash
docker exec influxdb influx -import -path=/var/lib/influxdb/influxdb.sql
```

Jeff - thanks for raising this issue!  Good job on getting it to work!  👍

jasonacox, 2022-08-13:
> Thanks Jason – that’s great!  Glad I could be of assistance – these dashboards are very impressive and am now running the containers on a Windows 11 machine that I use as a Plex server so its running 24x7.
>  
> As a side note I found that if I run the docker command in PowerShell / Admin Console (the native Windows shell) or Git command prompt it seems to work as well – just get the error in Git bash for some reason.
>  
> As another side note - I did notice a few of the stats aren’t coming through – have you seen this before anywhere?
> 
> 

[image]

jasonacox, 2022-08-13:
Hi @jaffro2k, yes we have seen this before.  Do you have a Powerwall+ system?  If not, that is the reason. The String and Inverter data is only available on Powerwall+ system, unfortunately.  

On Tesla Solar systems that use a separate inverter like Delta and SolarEdge instead of the integrated Tesla branded inverted in the Powerwall+, the Tesla Gateway has no way to poll the String data or Inverter performance. The Tesla Gateway is how we get all of the data for the dashboard via [pypowerwall](https://github.com/jasonacox/pypowerwall). I would love to figure out a way to pull this data from the others Inverters, but others have reported that they don't have a good API to facilitate that.  If you discover anything, please let us know!

prussell69, 2022-10-22:
I'm in the same boat as jaffro2k. I've tried to follow his instructions of what he did, but seem to get lost on exactly where he ran the command, and what was typed. When I tried to run the command from the "powerwall-desktop" folder, it seemed to accept it (no error shown), but I didn't get any data for the powerwall+ on the grafana webpage. I did start showing the "Charge state" and temperature in the top graph. I can see the data from the powerwall+ in the top-right display, same as if I log into the powerwall itself.
I'm sure there is a simply part I'm missing, and will feel real dumb after I'm shown, but I would love to get this running on my Windows 11 server since it too is running my Plex server.
Any help would greatly be appreciated!

jasonacox, 2022-10-22:
Hi @prussell69 !  I hope I can help.  Can you capture and paste a screenshot of what you are seeing (which parts are working)?  

The dashboard is made up of 5 components (containers): pypowerwall, telegraf, influxdb, weather411, and grafana. 

* If you see anything on the dashboard, that means `grafana` is working.
* If you see the animation happening on the dashboard, that indicates `pypowerwall` is working
* If you see weather data on any of the graphs, that indicates `weather411` is working
* If there is any graph data (not the animation) on the dashboard, that indicates `telegraf` is working.
* I can't tell by your description if `telegraf` or `influxdb` are working but when you did the following steps, if `influxdb` was not working, `grafana` would have shown an error during the 'save and test' part:

```
------------------[ Final Setup Instructions ]-----------------

Open Grafana at http://localhost:9000/ ... use admin/admin for login.

Follow these instructions for *Grafana Setup*:

* From 'Configuration\Data Sources' add 'InfluxDB' database with:
  - Name: 'InfluxDB'
  - URL: 'http://influxdb:8086'
  - Database: 'powerwall'
  - Min time interval: '5s'
  - Click "Save & test" button
```

You can test to see if the container are running:

```bash
docker logs telegraf
docker logs influxdb

# pypowerwall is working based on your description but also good to test
docker logs pypowerwall
```

Let us know what you see.

Also, try to go to this URL to see what pypowerwall says about your Tesla system: 

http://localhost:8675/alerts
http://localhost:8675/version
http://localhost:8675/help

Don't paste it but try this to see if you get a response:

http://localhost:8675/vitals

prussell69, 2022-10-22:
Here's the pic:
[image]
(Hopefully you can see it!)
I have the dashboard setup as you described. I will check the status of the telegraf & influxdb as soon as I get back to the house (traveling today).
Thanks for the quick response!
[image]

prussell69, 2022-10-22:
I checked the pypowerwall links you sent, and they all send back data.
I got this from the docker telegraf:
[outputs.influxdb] When writing to [http://influxdb:8086]: database "powerwall" creation failed: Post "http://influxdb:8086/query": dial tcp 172.18.0.2:8086: connect: connection refused

Hopefully that helps point to the issue...

prussell69, 2022-10-22:
Just a reminder, I too got the:
Setup InfluxDB Data for Powerwall...
**the input device is not a TTY.  If you are using mintty, try prefixing the command with 'winpty'**
I tried to remove the -it, but had to put quotes around the "/var/lib/influxdb/influxdb.sql" for it to not error.

jasonacox, 2022-10-22:
The problem is that InfluxDB is not setup and running correctly.  I would love to see if we can fix this and I can update the main setup script to better work for Windows.  Can you try running this from command line to see what happens?

```bash
# first make sure influx is running
docker logs influxdb

# try to shell into influxdb command prompt
docker exec --interactive --tty influxdb influx
```

If you were to remove -it, can you tell me what error you are getting?  If you used quotes, the command should put quotes around the entire command like this

```bash
# with quotes - doesn't work on my system but perhaps win11?
docker exec --tty influxdb "influx -import -path=/var/lib/influxdb/influxdb.sql"

# without quotes
docker exec --tty influxdb influx -import -path=/var/lib/influxdb/influxdb.sql
```

jasonacox, 2022-10-23:
I think I found it!  When you run that docker command on a Windows PowerShell or Command Prompt, it convert the "/' to a switch instead of passing it into the container for execution.  The Microsoft \ vs / curse.  I was able to get it to work using this:

```bash
docker exec --tty influxdb /bin/sh -c "influx -import -path=/var/lib/influxdb/influxdb.sql"
```

Can you confirm that it works for you too?  If so, I can update the setup.sh to use that which should work on all OS variants. Also, for a quick "how-to" for anyone using Windows 10/11 to install this, I recorded these steps were needed:

## Instructions on installing Powerwall-Dashboard on a Windows 11 Host

Install and Setup (adminstrator PowerShell or Windows Command Prompt):
* Install WSL (wsl --install) with an OS (recommend Ubuntu)
* Install *Docker Desktop* for Windows (https://www.docker.com/)
* Clone repo (`git clone https://github.com/jasonacox/Powerwall-Dashboard.git`)
* Run `cd Powerwall-Dashboard`
* Run `.\setup.sh` (if it prompts how to run - use Git)

Does that seem right?

jaffro2k, 2022-10-23:
Hi Jason,

Will do – am about to re-install on another machine so will let you know!

Cheers,

Jeff

jasonacox, 2022-10-23:
Thanks @jaffro2k !  I appreciate your help! I edited some of my comments - see the details: https://github.com/jasonacox/Powerwall-Dashboard/issues/63#issuecomment-1288003922 .  I pushed the update out and have tested it on Linux as well as Windows OS variants.

Let me know if I'm missing anything in the steps for Windows 11 instructions, or any details/steps we should add.

Thanks!

BJReplay, 2022-10-23:
Just tried again on Windows 11 and got it working.  Now for Azure... :)

prussell69, 2022-10-23:
I ended up deleting everything (too many tweaks!), and installed from scratch. It worked beautifully! Your instructions/steps are great.
Thank you so much for your help and quick response!
I am going to jump in and see if I can add in my solar charging data for my Tesla. I use a great app from "Charge HQ". Of course the trick is going to be only polling the car during the charging periods. Gotta let the car go to sleep sometime!

Again, thanks for all your help!!!

prussell69, 2022-10-23:
One more thing...
The only other thing I can think of that would help others:
You created a directory with different dashboards, but the time zone doesn't get updated with the setup script. It would help others if that could be updated from the setup, or tz scripts, saving to have to do that manually. Not a big deal, just something I think would be helpful.

jasonacox, 2022-10-23:
Thanks @prussell69 !  Good call on the tz.sh fix!  Fixed.  I also tested the `upgrade.sh` script and it works on my Windows 11 test systems.

On your Tesla car data, if you get that working, can you provide the details?  I suspect there are others that would love to add that. :) 

By the way, @mcbirse has created a brilliant tool that allows you to import all the Tesla CLOUD data (what the Tesla App uses) into Powerwall-Dashboard. If you are interested in that please check out https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history - I tested it successfuly on my Windows 11 setup as well and loaded data from the past year.  The instructions @mcbirse put together are excellent.  You might even be able to use a similar approach to get your car data.

prussell69, 2022-10-23:
I am looking at the history recall now. I followed his directions, and windows had me install Python during the 1st steps.
After the install, when I ran the tesla-history.py, I got the following:
PS C:\powerwall-dashboard\tools\tesla-history> python3 tesla-history.py --login
Traceback (most recent call last):
  File "C:\powerwall-dashboard\tools\tesla-history\tesla-history.py", line 41, in <module>
    from dateutil.relativedelta import relativedelta
ModuleNotFoundError: No module named 'dateutil'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\powerwall-dashboard\tools\tesla-history\tesla-history.py", line 45, in <module>
    sys.exit("ERROR: Missing python dateutil module. Run 'pip install python-dateutil'.")
NameError: name 'sys' is not defined

I tried to run the install python-dateutil, it of course replies:
PS C:\powerwall-dashboard\tools\tesla-history> pip install python-dateutil
Requirement already satisfied: python-dateutil in c:\users\parus\appdata\local\programs\python\python310\lib\site-packages (2.8.2)
Requirement already satisfied: six>=1.5 in c:\users\parus\appdata\local\programs\python\python310\lib\site-packages (from python-dateutil) (1.16.0)

Not sure if I'm just running it from the wrong folder or what??
It's never easy, is it...

jasonacox, 2022-10-23:
Ha!  Yes, it is a foreign language for Windows, but here is what I did that worked:

Run "Git Bash" prompt - not PowerShell or Command prompt from the directory with `tesla-history.py`:

```bash
# Install needed python modules
pip install python-dateutil teslapy influxdb

# Login to Tesla account
python tesla-history.py --login
```

If this works, we should add a note to the [tesla-history instructions](https://jasonacox.github.io/Powerwall-Dashboard/tools/tesla-history/) for Windows users to install python then the above steps.

prussell69, 2022-10-23:
Ran from "Git Bash" from the tesla-history folder and it seemed to freeze after asking to create the "tesla-history.conf" file.
To try something, I created the file (blank), and it came back:
ERROR: Config file 'tesla-history.conf' - No section: 'Tesla'

so I created one, [Tesla].
tried again, and got:
ERROR: Config file 'tesla-history.conf' - no option 'user' in section: 'Tesla'

added 'user' then tried again, followed by:
ERROR: Config file 'tesla-history.conf' - no option 'auth' in section: 'Tesla'

then:
ERROR: Config file 'tesla-history.conf' - No section: 'InfluxDB'

added that, then:
ERROR: Config file 'tesla-history.conf' - argument of type 'NoneType' is not itemable

OMG! Not sure why it didn't create the file by itself, nor where to go from the last message.

prussell69, 2022-10-23:
Figured it out!
I used git cmd, instead of git bash.
Worked great!
Gotta love Windows....

youzer-name, 2022-10-23:
> 

I took a quick link at Charge HQ but I'm not sure if it's storing anything locally. 

Are you using TeslaMate? https://github.com/adriankumpf/teslamate 

If not, you should take a look, as it brings all the data about the car into a local database and Grafana dashboards. While it streams all the data in real-time while the car is active, it doesn't keep the car awake through polling.

It wouldn't take too much effort to create dashboards or panels that combine the data from the Powerwall InfluxDB and the TeslaMate database.

prussell69, 2022-10-23:
No I don't, not yet anyway! I will look into it.

Ok, so I ran the script, and it prompted the info, asked for a webpage thats returned from authorizing, and came back with my site ID, name, install date and system time.
I'm not seeing the data, so maybe I answered one of InfluxDB questions wrong. It did create the tesla-history.conf and tesla-history.auth files!

prussell69, 2022-10-23:
Well, once again, got it figured out. Sorry for the last few questions!!

jasonacox, 2022-10-23:
Awesome!  Yes, the next step in case anyone is following this is to run the script again with the date range you want to import (it takes a while but you can list a large range like a year), something like:

```bash
# Get history data for start/end period
python tesla-history.py --start "2021-10-01 00:00:00" --end "2022-10-31 23:59:59"
```

longzheng, 2022-10-24:
Thanks for this. Switched from an Ubuntu VM to Windows. Works great 👍

prussell69, 2022-10-25:
Just wanted to say thanks to [youzer-name](https://github.com/youzer-name) for his Teslamate suggestion. It is a great program and look forward to using it. Hopefully one day, incorporating these into a single container (maybe), and adding in the solar charging functions of Charge HQ. 
Of course, I'm being pretty optimistic!
Thanks to you all!!!!

youzer-name, 2022-10-26:
@prussell69 I expected that anyone who is into Powerwall Dashboard and also drives a Tesla would also find TeslaMate useful. 

If you're using home automation and/or use Node Red, the MQTT interface from TeslaMate opens up a lot of possibilities. I just threw together a flow in Node-Red to notify me if my Model 3 is inside the 'Home' geofence with Sentry Mode on, because even though it is supposed to keep sentry off at home, I noticed it on earlier today. It's also easier to pull up TeslaMate to see where my wife is with the car than to either find the car in the Tesla app or use Google maps. Also my home automation hub (Hubitat) has a user-developed app for connecting to the Tesla API, so while TeslaMate is read-only, I can use the automation hub to send commands to my cars. I have all my charging logic set up in the Hubitat, rather than using the Tesla app.

On another tangent, I'm curious if anyone else is using Portainer with their docker setup? If you're not familiar with it, Portainer runs in Docker and gives you a web GUI for interacting with the other containers. You can view logs, open a command line interface, or update any of the containers from a browser instead of needing to SSH / VNC / RDP into the host machine. I have Portainer running on both my Raspberry Pi and Windows docker environments.

jaffro2k, 2022-10-29:
Hi @Jasoncox - I’ve re-installed from scratch and can verify setup.sh now working on my Windows 11 Hyper-V machine – I ran the commands from Windows Terminal which kicked off a GIT terminal to run the script.  Went through without error.

Only thing not reading data is the weather part of the dashboard – will have a look at that later today (unless you have any thoughts).  I put in location, the weather API key and GPS location all fine but no data as yet.

jaffro2k, 2022-10-30:
Hi – quick update it looks like I had the incorrect API key – I re-ran setup.sh and is now working.

jasonacox, 2022-10-30:
Awesome!  Great news @jaffro2k . Thanks for the update!

BJReplay, 2023-10-21:
> I noted when you install Docker Desktop, this essentially installs a WSL Linux distribution (alpine) called "docker-desktop".
> 
> I'm not sure if this helps anyone, but to me it seems Git for Windows is not really required in the original instructions since Ubuntu is being installed. Or, Ubuntu is not required if deciding on using Git for Windows. Or... I'm just confused by Windows things?!
> 
> What reasons are there for using Git for Windows / Git Bash over the WSL Ubuntu distribution?

Great write up, @mcbirse Michael.

I have to admit that I was a relative WSL newbie when I first started running Powerwall Dashboard, and a relative Git newbie as well.  I'm still a Git newbie, I suppose.  I just followed the instructions.

I suspect, however, that your analysis is spot on, and your two alternatives are spot on, and you can skip Git for Windows or skip the Ubuntu distro.

I also, now believe that there may be a third alternative, which I'm possibly up for trying today, and that is that you can skip Docker Desktop for Windows.

For a couple of months, WSL has supported systemd, which means, I believe, that it supports starting docker as a service, and you can install docker from within a WSL distro.

I'm going to give that a crack - if it works, it might look like this:

- Install WSL
- Enable systemd
- Install Docker and set to start on WSL startup
- From Ubuntu, clone repo
- Run CD Powerwall-Dashboard
- Run ./setup

Stand by...

If that works, it might work around the issue that I've not yet solved; getting Powerwall Dashboard to restart when the host reboots - because Docker Desktop doesn't restart until you log in.

mcbirse, 2023-10-21:
> I also, now believe that there may be a third alternative, which I'm possibly up for trying today, and that is that you can skip Docker Desktop for Windows.
>
> For a couple of months, WSL has [...]

@BJReplay - that would be great if you get that to work!

I had briefly looked into this also, but decided the ROI was not worth it for me as I was only setting up a system for testing, and not a permanent host for the dashboard.

Below are a couple of guides I came across that may help however:
https://dev.to/bowmanjd/install-docker-on-windows-wsl-without-docker-desktop-34m9
https://superuser.com/questions/1343558/how-to-make-wsl-run-services-at-startup

BJReplay, 2023-10-21:
> that would be great if you get that to work!

Looking good so far.

Up and running without Docker Desktop, and docker comes up on system startup.

Just about to install Powerwall Dashboard and see how I go.

I found the first link you supplied, but skipped it because it was skipping past the systemd approach.
The second also looks a bit kludgy.

The systemd approach requires windows 11, so isn't for everyone (that said, I haven't yet found old hardware that won't run it).

My current sources are:

- [Medium Article - requires sign up, but free](https://medium.com/geekculture/run-docker-in-windows-10-11-wsl-without-docker-desktop-a2a7eb90556d) - pretty good, but links to outdated mode of installing docker that doesn't get you docker compose installed as part of the installation.
- [Official Docker installation instructions](https://docs.docker.com/engine/install/ubuntu/#install-using-the-repository) - I'll use this to improve the previous instructions (requires adding docker's apt repository, which I think is fine).
- [A good transition from Docker Desktop reference](https://nickjanetakis.com/blog/install-docker-in-wsl-2-without-docker-desktop) - this has a good note about iptables compatibility that I'll include

BJReplay, 2023-10-22:
I got there in the end (running without Docker Desktop at all), but I wouldn't recommend it!

A few quick notes before I forget:

- Networking is hard: Docker Desktop makes it easy to allow access from other machines on your network.
- You need to enable bridge mode in .wslconfig, and set up a Hyper-V Network Bridge to allow access from other hosts on the network.
- Auto-start is still hard, but I got there too: `wsl -u root -e sh -c "service docker start && touch startup.log && tmux new-session -d -s keepalive"` was the command that I needed on windows startup to bring up WSL and docker before a user logs into the windows host.  It looks like you need all three to get it to start and stay up! Starting a terminal session on it's own won't start docker (even though it is set to start, starting docker is fine, but without the background tmux session, it shuts down).
- I migrated my existing setup from using Docker Desktop to docker on WSL2 pretty easily.  I moved data from a mysql database by finding the unmounted data volume from a mysql instanced started with a docker run, copying it with the docker container stopped, starting a new docker container with the copied data in mounted volume, and mysql came up; that gave me confidence that I could move ephemeral container volumes to host mounted volumes and blow away the docker desktop environment.
- I had to re-authenticate to docker which was a little tricky, but a google of he error message got me there (blow away the left over config in .docker in the Unbuntu distro)

All in all, I wouldn't recommend it (primarily because of the networking aspect), but it does work, and is nicer in many ways; upgrades are faster, startup is faster, and it uses less memory.

jasonacox, 2023-10-22:
That's helpful @BJReplay - Most of our community are new to setting up something like a dashboard, and much less in the weeds on networking dark magic.  If we could script it, then it may become a viable path, but your recommendation is probably right.  Keeping your installation notes would be good to help other experts explore Docker Desktop-less installations.

If the goals is shortest "easy path" (quickstart) for our Windows community, should we update the instructions to not include Git for Windows? @mcbirse I'm looking at your first example? With the optimization being that it requires less 3rd party packages to install and maintain, is that the best instructions set? 

If we provide an option for "experts" for the optimized but "complex path", with lowest memory and higher stability, we can point to your instructions @BJReplay?

BJReplay, 2023-10-22:
> If the goals is shortest "easy path" (quickstart) for our Windows community, should we update the instructions to not include Git for Windows?

I think this is the best bet.  No need for Git for Windows or Git Bash Prompt

> If we provide an option for "experts" for the optimized but "complex path", with lowest memory and higher stability, we can point to your instructions BJReplay?

I will write it up and can create a pull request for a separate windows11-docker-standalone.md that can be linked from the main README.md if you think that it's appropriate, or include it as separate section in the main README.md, as you prefer; it's a lot longer than the standard windows 10/11 install because of a number of additional steps.  Let me know your thoughts.

jasonacox, 2023-10-22:
Thanks @BJReplay !  I think we keep the simple version in README.md but add a note and link for advanced setup (without Docker Desktop).  The advanced instructions could be a WINDOWS.md file or it could point to a Discussion topic if you think it would be helpful for Q&A and refinement.

mcbirse, 2023-10-22:
@BJReplay - Great information. I'm sure this will be useful to Windows users who would like to go down that path, although it sounds like a pain.

I agree with keeping a simple version of instructions in the README similar to my first example, and a link or separate instructions with advanced options for Windows.

jaffro2k, 2023-10-27:
Hi all - had a couple of additional notes / observations that may be of use:

1. You can install Docker Desktop using Hyper-V instead of using WSL if you run a Hyper-V setup.  This is a choice in the installation of the product.  Its not the recommended option but works fine for me as I run other VM's and avoids me installing WSL.  I believe that required you to install Git etc to enable the Linux scripts to work (but not 100% on this).

2.  I use auto logon in Windows 11 so that Docker Desktop starts up after reboot - and run it all under a local account.  Not perfect from a security perspective but it does the trick: https://learn.microsoft.com/en-us/sysinternals/downloads/autologon