jasonacox/Powerwall-Dashboard #74: Weather not in animation dashboard

zikronix, 2022-09-07:
Upgraded using upgrade.sh. It failed, had to run it again and then weather.sh and tz.sh. Imported new animation dashboard, not seeing it. Downloaded dashboard from repository and imported it same result

jasonacox, 2022-09-09:
Hi @zikronix  - thanks for opening this issue!  When upgrade.sh failed, do you recall where it was or any error message?

The non-weather data on the new dashboard is an indication that the weather data is not making it into InfluxDB.  Can you check the output from weather411:

```
docker logs weather411
```
Also, visit these URLs to see if the weather411 service is responding (change 'localhost' to the IP address running your dashboard):

http://localhost:8676/
http://localhost:8676/stats

recrudesce, 2022-12-15:
I have this issue, but it was because my API key was disabled on https://home.openweathermap.org/api_keys

So maybe check that it's enabled :)

jasonacox-sam, 2026-08-31:
This reported weather data missing from the animation dashboard after an upgrade. Resolution found in the thread: the OpenWeatherMap API key was disabled/inactive — once enabled, weather flowed again. Worth checking `https://home.openweathermap.org/api_keys` if weather ever goes blank.

Closing since the cause was identified. Re-open if weather ingestion breaks for a different reason on current versions.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam