jasonacox/Powerwall-Dashboard #75: Backups No longer Delete Older Versions

oralallen82, 2022-09-17:
I am currently using the backup script I am using the default 5 days however i've noticed that the backup has stopped deleting the older versions. Let me know what details you need and I can provide it to assist with troubleshooting.

jasonacox, 2022-09-17:
Hi @oralallen82 - can you provide a listing `ls -l` of the backup directory?

Also, can you run the backup script manually and capture any errors?  The removal is a simple:

```sh
find ${BACKUP_FOLDER}/Powerwall-Dashboard.*tgz -mtime +${KEEP} -type f -delete
```

oralallen82, 2022-09-17:
> Hi @oralallen82 - can you provide a listing `ls -l` of the backup directory?
> 
> Also, can you run the backup script manually and capture any errors? The removal is a simple:
> 
> ```shell
> find ${BACKUP_FOLDER}/Powerwall-Dashboard.*tgz -mtime +${KEEP} -type f -delete
> ```

ls -l 

total 797680
-rw-r--r-- 1 root root 110152658 Sep 11 06:00 Powerwall-Dashboard.2022-09-11.tgz
-rw-r--r-- 1 root root 108119050 Sep 12 06:00 Powerwall-Dashboard.2022-09-12.tgz
-rw-r--r-- 1 root root 117849760 Sep 13 06:00 Powerwall-Dashboard.2022-09-13.tgz
-rw-r--r-- 1 root root 118346288 Sep 14 06:00 Powerwall-Dashboard.2022-09-14.tgz
-rw-r--r-- 1 root root 118532368 Sep 15 06:00 Powerwall-Dashboard.2022-09-15.tgz
-rw-r--r-- 1 root root 123832842 Sep 16 22:02 Powerwall-Dashboard.2022-09-16.tgz
-rw-r--r-- 1 root root 119974711 Sep 17 06:00 Powerwall-Dashboard.2022-09-17.tgz

Error when running: find ${BACKUP_FOLDER}/Powerwall-Dashboard.*tgz -mtime +${KEEP} -type f -delete

find: invalid argument `+' to `-mtime'

jasonacox, 2022-09-17:
Sorry, substitute 5 for ${KEEP} and replace "folder" with the location where those *tgz files are located:

```bash
find folder/Powerwall-Dashboard.*tgz -mtime +5 -type f -delete
```

oralallen82, 2022-09-17:
> Sorry, substitute 5 for ${KEEP} and replace "folder" with the location where those *tgz files are located:
> 
> ```shell
> find folder/Powerwall-Dashboard.*tgz -mtime +5 -type f -delete
> ```

‘/backup/Powerwall-Dashboard.*tgz’: No such file or directory

I can confirm the directory is correct. Any ideas?

oralallen82, 2022-09-17:
Another user error the ${BACKUP_FOLDER} location missed the / in front of home

closed 2022-09-17 by oralallen82