jasonacox/Powerwall-Dashboard #84: Enhancement: Represent grid outages directly on the Energy Usage graph

mp09, 2022-09-26:
It would be nice to have some representation of the grid outage directly on the main energy usage graph to be able to quickly look at information during outage periods.

youzer-name, 2022-09-26:
I was able to do this by adding the following query to the energy usage panel (which I'm running in the updated Time series component)
```
SELECT min("grid_status") +1  FROM "grid"."http" WHERE grid_status = 0 and $timeFilter GROUP BY time($__interval) tz('America/New_York')
```
This gets the grid status where it is zero and adds 1 so that any points with a grid outage return "1" from the query. I aliased this query as "Grid Status"

[image]

Then I set up the following overrides:

Fields with name; Grid status
Axis > Placement: Hidden
Graph styles > Fill opacity: 24
Standard Options > Min : 0
Standard Options > Max: 1
Standard Options > Color scheme: Single color (select red)
Series > Hide in area: select "Tooltip" and "Legend"

[image]

I had to zoom way in to show this outage because it was pretty short, but here is what I get from the above:

[image]

Found a longer outage. Here is a full day's chart:

[image]

mp09, 2022-09-26:
I'm not sure why but my Override UI is missing some of those options 🤔 
I think I'm up to date on everything...

youzer-name, 2022-09-26:
My Grafana is 9.0.7 and says an update is available. I may be lagging behind the latest. Which option(s) are missing? I'll try to update and see if I lose the same items.

youzer-name, 2022-09-26:
I just updated to the project's standard Grafana. I see it did lose some color options that I had used in another post over in the discussions > show & tell section, but I think all the overrides look the same on my Grid Down field on the Energy Usage panel:

[image]

jasonacox, 2022-09-27:
Hey @youzer-name this is awesome!  This only works on the new time series graphs (the project dashboard.json is still using the older version).

@mp09 if you want to upgrade to the newer time series graph, there is a migrate button. 
[image]

Note: There are some zero-point offset errors that occur (especially noticeable is the sun graph).  We need to update the graphs to have the +/- values to force zero-point alignment but I haven't had time to work on that and test it yet.

I do love the idea of pushing the grid up/down into the main energy usage graph so I'll leave this as a TODO enhancement.  Maybe @youzer-name will submit a PR once he gets it tweaked. 😄

youzer-name, 2022-09-27:
@jasonacox I can try to submit a PR, but since I'm not using the install script nor am I using the standard Grafana dashboards. I've tweaked mine a bunch from what's in the project. I think if I import the standard Dashboard and then copy over just that one panel, I can then export the result and it should merge ok. While I'm at it, I'll include the extra queries that fix the zero point issue. Maybe have time this afternoon or tomorrow to give it a try.

youzer-name, 2022-09-27:
@jasonacox Ok, I just submitted a PR The changes don't match up that well in the Github comparison, but I think that may be due to the switch to the time series component. The whole first section of the dashboard ("inputs") wasn't there after I got done editing.

Also, in the latest version of Grafana Enterprise, I found a toggle switch for 'centered zero', so there was no need for my workaround of adding the inverse queries and hiding them from view. Toggling that switch is all that was needed. I first built this on version 9.0, and that switch wasn't available in that version.

mp09, 2023-01-25:
this is great, closing! 
[image]

closed 2023-01-25 by mp09