jasonacox/Powerwall-Dashboard #85: Unable to Upgrade Grafana.env Permission Error

oralallen82, 2022-09-27:
Attempting to upgrade to version 2.6.2 everything upgraded except for Grafana.env error "cp: cannot create regular file 'grafana.env': Permission denied"

jasonacox, 2022-09-27:
Hi @oralallen82 - Thanks for opening this issue.  That's interesting.  The file should be owned by your local user.  Can you run this?

```bash
ls -l grafana.env
```

The update to grafana.env is the addition of the new boomtable code to fix the "No data" bug.  You can manually update:

```bash
# create backup
cp "grafana.env" "grafana.env.bak"

# copy new version
cp "grafana.env.sample" "grafana.env"

# restart grafana
docker stop grafana
docker rm grafana
docker-compose -f powerwall.yml up -d
```

I do see that the upgrade.sh is not forcing a restart of grafana (which I should fix) but that would not cause the error you are seeing.

oralallen82, 2022-09-27:
Running ls -l: -rw-r--r-- 1 root root 926 Aug 30 19:19 grafana.env

I decided to manually update and this task was successful.

Thanks again for your great work and help.

oralallen82, 2022-09-27:
> Running ls -l: -rw-r--r-- 1 root root 926 Aug 30 19:19 grafana.env
> 
> I decided to manually update and this task was successful.
> 
> Thanks again for your great work and help.

I do believe that permission issue may be from me previously running sudo ./upgrade.sh

jasonacox, 2022-09-28:
Ah yes!  That makes sense.  Thanks @oralallen82 !

oralallen82, 2022-10-03:
New issue attempting to upgrade to version 2.6.3

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 37, done.
remote: Counting objects: 100% (37/37), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 30 (delta 17), reused 20 (delta 10), pack-reused 0
error: insufficient permission for adding an object to repository database .git/objects
fatal: failed to write object
fatal: unpack-objects failed

jasonacox, 2022-10-04:
Thanks for the details. I decided to change the upgrade methodology.  The issue with using `git` for updating the files is that you run into situations like you mention above.  There may be an even better approach for upgrading. If anyone has an idea or is interested in helping with that effort (either on approach or coding) please let me know.

@oralallen82 for this specific issue you can try again (**I still recommend you create a backup first**) and it should take you to 2.6.4 now.  Let us know how it works.

oralallen82, 2022-10-04:
> Thanks for the details. I decided to change the upgrade methodology. The issue with using `git` for updating the files is that you run into situations like you mention above. There may be an even [...]

See error:

Upgrade Powerwall-Dashboard from 2.6.2 to 2.6.4
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Upgrade - Proceed? [y/N] y

Resetting Timezone to Default...

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...

Saved working directory and index state WIP on main: 1c5fed6 Add spacing
remote: Enumerating objects: 43, done.
remote: Counting objects: 100% (43/43), done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 35 (delta 20), reused 24 (delta 12), pack-reused 0
error: insufficient permission for adding an object to repository database .git/objects
fatal: failed to write object
fatal: unpack-objects failed

jasonacox, 2022-10-04:
Oh no!  I believe you are still suffering from the `sudo upgrade.sh` before we fixed the upgrade.sh to prohibit running as root.

If you were to `ls -l .git` I bet you would see root owning some things in there.

You probably need to:

```bash
sudo chown -R $USER .git
```

oralallen82, 2022-10-04:
> Oh no! I believe you are still suffering from the `sudo upgrade.sh` before we fixed the upgrade.sh to prohibit running as root.
> 
> If you were to `ls -l .git` I bet you would see root owning some things in there.
> 
> You probably need to:
> 
> ```shell
> sudo chown -R $USER .git
> ```

@jasonacox ls -l .git showed I owned everything in there but I ran sudo chown... and reran the upgrade and I was able to upgrade. I honestly hope this is the last of these issues I've created for myself.

jasonacox, 2022-10-04:
That's great news!  There could have been hidden (dot) files or subdirs that were root owned.  

I do appreciate you discovering this so we could update `upgrade.sh` to prevent running as root. To be fair, running `sudo upgrade.sh` usually seems logical.

oralallen82, 2022-10-16:
@jasonacox my upgrade.sh has reared it's head again attempting to upgrade. I was able to change ownership of the README.sh, I now show I own it and not root but I am still presented with this error:

Upgrade Powerwall-Dashboard from 2.6.4 to 2.6.5
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Upgrade - Proceed? [y/N] y

Resetting Timezone to Default...

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...

No local changes to save
Updating 725bf23..49eccc6
error: unable to unlink old 'tools/README.md': Permission denied
error: unable to create file tools/tesla-history.py: Permission denied

ls -l README.md

-rw-r--r-- 1 oral root 3253 Aug 30 19:14 README.md

jasonacox, 2022-10-16:
It's probably a permission issue on the tools directory:

```bash
# see what they are
ls -ld tools

# fix them
sudo chown -R $USER tools
```

oralallen82, 2022-10-16:
Worked!!

closed 2022-10-25 by oralallen82