jasonacox/Powerwall-Dashboard #88: 22.18.6 Flashing Powerflow Animation

cwagz, 2022-09-28:
This morning I noticed my Powerflow animation is just flashing on the dashboard. I checked my firmware version, and it looks like I got updated to 22.18.6. Is anyone else noticing this issue?

Navigating to port 8675 shows the same behavior. The page just keeps reloading making the animation "flicker"

jasonacox, 2022-09-28:
Ok, the fix is easy!  The issue is that pypowerwall proxy caches the responses from the Powerwall, including the setup JS.  You just need to restart pypowerwall and refresh your browser:

```bash
docker restart pypowerwall
```

Please let me know if this fixes the issue for you too @cwagz.

I'll have to look to see if there is an easy way to have the proxy detect version upgrades and flush the cache.

cwagz, 2022-09-29:
@jasonacox - I restarted pypowerwall and it is no longer flashing. Is that what you were expecting or did Tesla change something back to the way it was?

jasonacox, 2022-09-29:
Awesome!  Thanks for opening the issue and confirming the fix.

Tesla didn't change anything back.  Restarting the pypowerwall proxy clears the local cache of Javascript that contained the old firmware version of the animation.  Once cleared, pypowerwall rebuilds the cache with the new firmware version. 

I'm looking to see if we can detect and auto-clear the pypowerwall proxy cache anytime the firmware upgrades the animation.

jasonacox-sam, 2026-08-31:
This reported the Powerflow animation flashing/reloading after firmware 22.18.6. Cause: the pypowerwall proxy had cached the old firmware's animation JS. Fix: `docker restart pypowerwall` and refresh the browser — confirmed working by the reporter.

Closing since it was resolved. Same advice still applies whenever firmware upgrades make the animation flicker.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam