jasonacox/Powerwall-Dashboard #101: Default dashboard doesn't filter out invalid data from InfluxDB

ibmaster, 2022-10-15:
Somehow the database occasionally records some time entries where all the data fields are 0 / NULL. (Perhaps from network or API timeouts?
Can the default dashboards filter these out so they are ignored/omitted in the graphs?
If not, any suggestions on how to manually modify the database?

jasonacox, 2022-10-16:
Hi @ibmaster  - Can you paste a screenshot of what you are seeing?  

1) I had similar problems with data dropout that was caused by the Powerwall losing WiFi connection for a while. I eventually fixed this by hardwiring the Powerwall gateway into my network.  I highly recommend that.
2) There is a way to fill in those gaps in your data from the Tesla Cloud using a tool that @mcbirse built (see https://github.com/jasonacox/Powerwall-Dashboard/issues/12).

ibmaster, 2022-10-16:
Thanks, I was able to use the tesla_history.py script (after some code modification to not just remove cloud data) to remove appropriate time ranges of invalid entries in the database.

closed 2022-10-16 by ibmaster