jasonacox/Powerwall-Dashboard #104: hard coded upgrade.sh url

BuongiornoTexas, 2022-10-15:
@jasonacox, this issue may just be my inexperience with git, but I can't figure out how to test the upgrade.sh script without changing the hard coded path to the master repo, and even then, I have no idea how to make this work from a branch I'm testing (i.e. how can I force the upgrade to pull updated files from the branch I'm testing rather than the master?). 

This is in relation to updates from #96. If there is no obvious short term fix, I will raise the PR without testing upgrade.sh and will highlight that in the commit notes - there are relatively few changes and it should be straightforward to review these. 

On a related note - the upgrade.sh script runs a `git stash` before doing a rebase pull, but never refers to the stashed elements again. Is this deliberate?

jasonacox, 2022-10-16:
Great questions @BuongiornoTexas .  The intent of `upgrade.sh` was less for developers and more for end users.  In full transparency, I'm not fond of using git as the means of doing upgrades for general users. But if I pivot to having build artifacts (releases) manage state, it would likely create other complications.  But as to your points:

* I think we could update `upgrade.sh` to accommodate branching and variable repository locations, perhaps through environmental settings for developers?  What do you think?
* Thanks for your PR!
* I use `git stash` to record changes the user may have made to tracked files like `powerwall.yml` before I blow them away with `git pull --rebase`.  It means that the user can manually recover those changes `git stash pop` or just see what they are  with `git stash list`.  I debated over adding a `pop` after the rebase, but I'm worried that user changes may break the install since the tracked files should not be changed if we are doing clean upgrades.  I'm open to ideas here, but as I mention, the git method is both a blessing and a curse depending on persona of the user.

BuongiornoTexas, 2022-10-16:
@jasonacox - Your response makes sense. In answer to your own question, I think as long as changes to `upgrade.sh` are infrequent and you are willing to live with the pain of final code review, I'd stick with what you have. From my (limited) dig into, it looks as though making git behave in the way we would like for development may need a great deal of work and deliver fairly limited benefit.

And now I'll get back to trying reconcile the various energy measurements from my systems and meters - which is looking to be a long road.

jasonacox-sam, 2026-08-31:
This asked about testing `upgrade.sh` from a branch instead of the hard-coded master URL. The discussion concluded that the script is deliberately end-user-oriented rather than developer-oriented, and the maintainer chose to stick with the simple git-based approach as long as changes to `upgrade.sh` stay infrequent.

Closing since the discussion reached its conclusion. Re-open if the upgrade path needs rethinking.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam