jasonacox/Powerwall-Dashboard #110: Ubuntu 22.04 Host - Install works but InfluxDB fails to start

jasonacox, 2022-10-22:
>  running host os ubuntu 22.04 everything installed, working, docker, etc - setup script executes, 
> but gets stuck Running Docker-Compose...
>
> pypowerwall is up-to-date
> influxdb is up-to-date
> grafana is up-to-date
> weather411 is up-to-date
> telegraf is up-to-date
>  
> Waiting for InfluxDB to start...
> ..........................................................................................................................................
> It's still trying - after 24 hours I've blown the vm away and tried again, same issue ? might it be permissions ?

Some thoughts:
* Docker-compose is working fine, the step stage is to wait for InfluxDB to start. 
* I recommend that you run `docker logs influxdb` to see what errors it is throwing.

BuongiornoTexas, 2022-10-22:
Stopping Influxdb only and then restarting interactively (`up` without `-d`) might also provide some information about what is going on. Is this on a clean install?

(I've either had this problem or something closely related to it, but I cannot for the life of me remember what the solution was. )

nullgel, 2022-12-23:
Has anyone figured out what the issue is? I'm stuck here too.

jasonacox, 2022-12-23:
Hi @nullgel !  Can you provide details about your setup (OS, processor) so I can try to replicate?  Also, would love to see what you get from `docker logs influxdb` and try what @BuongiornoTexas suggests.

nullgel, 2022-12-23:
Ubuntu 22.04.1 - (New install)
Intel Core i7-3770 @ 3.40GHz

I've also added logs from telegraf in case that's useful.

I had a previous Powerwall-Dashboard installation (from a VM) that I just now untarred into the directory just to see what would happen and after running the upgrade script, compose-dash stop, and finally compose-dash start it "appears" to be working. I couldn't get my old data back for some reason, but just used the history script to pull in what it could.

**#docker logs influxdb:**
ts=2022-12-23T20:39:17.634049Z lvl=info msg="InfluxDB starting" log_id=0exBEBWW000 version=1.8.10 branch=1.8 commit=688e697c51fd
ts=2022-12-23T20:39:17.634066Z lvl=info msg="Go runtime" log_id=0exBEBWW000 version=go1.13.8 maxprocs=8
ts=2022-12-23T20:39:24.638033Z lvl=info msg="Using data dir" log_id=0exBEBWW000 service=store path=/var/lib/influxdb/data
ts=2022-12-23T20:39:24.638217Z lvl=info msg="Compaction settings" log_id=0exBEBWW000 service=store max_concurrent_compactions=4 throughput_bytes_per_second=50331648 throughput_bytes_per_second_burst=50331648
ts=2022-12-23T20:39:24.638258Z lvl=info msg="Open store (start)" log_id=0exBEBWW000 service=store trace_id=0exBEbsW000 op_name=tsdb_open op_event=start
ts=2022-12-23T20:39:24.638337Z lvl=info msg="Open store (end)" log_id=0exBEBWW000 service=store trace_id=0exBEbsW000 op_name=tsdb_open op_event=end op_elapsed=0.083ms
ts=2022-12-23T20:39:24.638402Z lvl=info msg="Opened service" log_id=0exBEBWW000 service=subscriber
ts=2022-12-23T20:39:24.638416Z lvl=info msg="Starting monitor service" log_id=0exBEBWW000 service=monitor
ts=2022-12-23T20:39:24.638432Z lvl=info msg="Registered diagnostics client" log_id=0exBEBWW000 service=monitor name=build
ts=2022-12-23T20:39:24.638449Z lvl=info msg="Registered diagnostics client" log_id=0exBEBWW000 service=monitor name=runtime
ts=2022-12-23T20:39:24.638466Z lvl=info msg="Registered diagnostics client" log_id=0exBEBWW000 service=monitor name=network
ts=2022-12-23T20:39:24.638487Z lvl=info msg="Registered diagnostics client" log_id=0exBEBWW000 service=monitor name=system
ts=2022-12-23T20:39:24.638519Z lvl=info msg="Starting precreation service" log_id=0exBEBWW000 service=shard-precreation check_interval=10m advance_period=30m
ts=2022-12-23T20:39:24.638545Z lvl=info msg="Starting snapshot service" log_id=0exBEBWW000 service=snapshot
ts=2022-12-23T20:39:24.638559Z lvl=info msg="Starting continuous query service" log_id=0exBEBWW000 service=continuous_querier
ts=2022-12-23T20:39:24.638567Z lvl=info msg="Starting HTTP service" log_id=0exBEBWW000 service=httpd authentication=false
ts=2022-12-23T20:39:24.638576Z lvl=info msg="Storing statistics" log_id=0exBEBWW000 service=monitor db_instance=_internal db_rp=monitor interval=10s
ts=2022-12-23T20:39:24.638715Z lvl=info msg="Listening on HTTP" log_id=0exBEBWW000 service=httpd addr=[::]:8086 https=false
ts=2022-12-23T20:39:24.638747Z lvl=info msg="Starting retention policy enforcement service" log_id=0exBEBWW000 service=retention check_interval=30m
ts=2022-12-23T20:39:24.638847Z lvl=info msg="Listening for signals" log_id=0exBEBWW000
ts=2022-12-23T20:39:24.638891Z lvl=info msg="Sending usage statistics to usage.influxdata.com" log_id=0exBEBWW000
ts=2022-12-23T20:54:41.858341Z lvl=info msg="Signal received, initializing clean shutdown..." log_id=0exBEBWW000
ts=2022-12-23T20:54:41.858381Z lvl=info msg="Waiting for clean shutdown..." log_id=0exBEBWW000
ts=2022-12-23T20:54:41.858513Z lvl=info msg="Shutting down monitor service" log_id=0exBEBWW000 service=monitor
[tcp] 2022/12/23 20:54:41 tcp.Mux: Listener at 127.0.0.1:8088 failed failed to accept a connection, closing all listeners - accept tcp 127.0.0.1:8088: use of closed network connection
ts=2022-12-23T20:54:41.858547Z lvl=info msg="Terminating storage of statistics" log_id=0exBEBWW000 service=monitor
ts=2022-12-23T20:54:41.858570Z lvl=info msg="Terminating precreation service" log_id=0exBEBWW000 service=shard-precreation
ts=2022-12-23T20:54:41.858581Z lvl=info msg="Listener closed" log_id=0exBEBWW000 service=snapshot
ts=2022-12-23T20:54:41.858591Z lvl=info msg="Terminating continuous query service" log_id=0exBEBWW000 service=continuous_querier
ts=2022-12-23T20:54:41.858730Z lvl=info msg="Closing retention policy enforcement service" log_id=0exBEBWW000 service=retention
ts=2022-12-23T20:54:41.859126Z lvl=info msg="Closed service" log_id=0exBEBWW000 service=subscriber
ts=2022-12-23T20:54:41.859162Z lvl=info msg="Server shutdown completed" log_id=0exBEBWW000
ts=2022-12-23T20:55:03.546424Z lvl=info msg="InfluxDB starting" log_id=0exC7vUW000 version=1.8.10 branch=1.8 commit=688e697c51fd
ts=2022-12-23T20:55:03.546444Z lvl=info msg="Go runtime" log_id=0exC7vUW000 version=go1.13.8 maxprocs=8
ts=2022-12-23T20:55:11.976665Z lvl=info msg="Using data dir" log_id=0exC7vUW000 service=store path=/var/lib/influxdb/data
ts=2022-12-23T20:55:11.976742Z lvl=info msg="Compaction settings" log_id=0exC7vUW000 service=store max_concurrent_compactions=4 throughput_bytes_per_second=50331648 throughput_bytes_per_second_burst=50331648
ts=2022-12-23T20:55:11.976774Z lvl=info msg="Open store (start)" log_id=0exC7vUW000 service=store trace_id=0exC8RQ0000 op_name=tsdb_open op_event=start
ts=2022-12-23T20:55:11.981947Z lvl=info msg="Reading file" log_id=0exC7vUW000 engine=tsm1 service=cacheloader path=/var/lib/influxdb/wal/_internal/monitor/1/_00001.wal size=263898
[... cut ...]
2022-12-23T20:40:28Z E! [outputs.influxdb] When writing to [http://influxdb:8086]: received error retention policy not found: raw
2022-12-23T20:40:38Z E! [outputs.influxdb] When writing to [http://influxdb:8086]: received error retention policy not found: raw
2022-12-23T20:40:48Z E! [outputs.influxdb] When writing to [http://influxdb:8086]: received error retention policy not found: raw
2022-12-23T20:40:58Z E! [outputs.influxdb] When writing to [http://influxdb:8086]: received error retention policy not found: raw

jasonacox, 2022-12-23:
Interesting!  It is like the influxdb setup script didn't run correctly. Try to run this manually:

```bash
docker exec -it influxdb influx -import -path=/var/lib/influxdb/influxdb.sql
```

BuongiornoTexas, 2022-12-23:
If @jasoncox hasn't already fixed it, can you check which images are up and running using `docker ps`? 

When I've seen this in the past, I've been able to isolate the problem container using this approach. And another check - can you access your powerwall local webserver separately from the dashboard (if this isn't working, pypowerwall will keep failing and restarting) - I've got a problem with the powerwall not responding to ARP after a bit - had to resolve this with a start ARP workaround.

nullgel, 2022-12-24:
@jasonacox I ran your command with no error. (or any output) I also didn't see anything of note in the influxdb log.

@BuongiornoTexas
CONTAINER ID   IMAGE                          COMMAND                  CREATED       STATUS       PORTS                                                 NAMES
3a0ccc5338f9   telegraf:latest                "/entrypoint.sh tele…"   3 hours ago   Up 2 hours   8092/udp, 8125/udp, 8094/tcp                          telegraf
fb3836ddbd02   jasonacox/pypowerwall:latest   "python3 server.py"      3 hours ago   Up 2 hours   0.0.0.0:8675->8675/tcp, :::8675->8675/tcp             pypowerwall
ab05eb2a1f29   jasonacox/weather411:latest    "python3 server.py"      3 hours ago   Up 2 hours   0.0.0.0:8676->8676/tcp, :::8676->8676/tcp             weather411
d2ddb2f6eecf   grafana/grafana:9.1.2-ubuntu   "/run.sh"                4 hours ago   Up 2 hours   3000/tcp, 0.0.0.0:9000->9000/tcp, :::9000->9000/tcp   grafana
d8c49561683d   influxdb:1.8                   "/entrypoint.sh infl…"   4 hours ago   Up 2 hours   0.0.0.0:8086->8086/tcp, :::8086->8086/tcp             influxdb

The whole thing is certainly weird, but my instance is running smoothly now. Other than un-tarring another installation I had and running the scripts I mentioned, I don't know what made it ultimately start working. I might just HAL this one and let it fail and maybe come back to it when it does.

Thanks for the help in the meantime!

BuongiornoTexas, 2022-12-24:
Given that docker ps, it looks like you sorted it with the influxdb setup script.  Also minor error on my post - my fix was a static ARP.

jasonacox, 2022-12-24:
Hi @nullgel - just to confirm, you used the `setup.sh` script to set this up?  If so, I'll add a TODO to my list to add more checks to the InfluxDB script run to ensure it runs correctly. 

Happy holidays all and Merry Christmas (Eve)! 🎄

nullgel, 2022-12-24:
Indeed, I did use the setup.sh script.

I'll be more than willing to do some additional testing for the project when the time comes. I'll be following along with updates and I'll break out a new machine to see if the problem is gone.

Thanks for your hard work! I love this project. Merry Christmas to you, too! 🎁

Stoobiedoo, 2023-01-07:
I'm running into the same problem. I spun up a Ubuntu 22.10 virtual box instance. It gets stuck at Waiting for InfluxDB to start.
I tried to stop all the containers and found I couldn't kill the influx one. I got the following error:
Error response from daemon: cannot stop container: 9a6fdaf7bccc: permission denied
This lead me to a page that suggested I run:
"sudo aa-remove-unknown"
This allowed me to stop the container, but it did not help with the original problem. Does it mean anything that the pypowerwall and grafana containers keep getting reset?

stuart@stuart-VirtualBox:~/Desktop/Powerwall-Dashboard$ docker ps
CONTAINER ID   IMAGE                          COMMAND                  CREATED        STATUS              PORTS                                                 NAMES
0f8c2dd377bf   jasonacox/weather411:latest    "python3 server.py"      15 hours ago   Up About a minute   0.0.0.0:8676->8676/tcp, :::8676->8676/tcp             weather411
f6637eb9bbeb   telegraf:latest                "/entrypoint.sh tele…"   15 hours ago   Up About a minute   8092/udp, 8125/udp, 8094/tcp                          telegraf
f3ec9e49a9b3   grafana/grafana:9.1.2-ubuntu   "/run.sh"                15 hours ago   Up 3 seconds        3000/tcp, 0.0.0.0:9000->9000/tcp, :::9000->9000/tcp   grafana
ff89450f1e0e   jasonacox/pypowerwall:latest   "python3 server.py"      15 hours ago   Up 2 seconds        0.0.0.0:8675->8675/tcp, :::8675->8675/tcp             pypowerwall
9a6fdaf7bccc   influxdb:1.8                   "/entrypoint.sh infl…"   15 hours ago   Up 2 minutes        0.0.0.0:8086->8086/tcp, :::8086->8086/tcp             influxdb
stuart@stuart-VirtualBox:~/Desktop/Powerwall-Dashboard$ docker ps
CONTAINER ID   IMAGE                          COMMAND                  CREATED        STATUS         PORTS                                                 NAMES
0f8c2dd377bf   jasonacox/weather411:latest    "python3 server.py"      15 hours ago   Up 5 minutes   0.0.0.0:8676->8676/tcp, :::8676->8676/tcp             weather411
f6637eb9bbeb   telegraf:latest                "/entrypoint.sh tele…"   15 hours ago   Up 5 minutes   8092/udp, 8125/udp, 8094/tcp                          telegraf
f3ec9e49a9b3   grafana/grafana:9.1.2-ubuntu   "/run.sh"                15 hours ago   Up 1 second    3000/tcp, 0.0.0.0:9000->9000/tcp, :::9000->9000/tcp   grafana
ff89450f1e0e   jasonacox/pypowerwall:latest   "python3 server.py"      15 hours ago   Up 6 seconds   0.0.0.0:8675->8675/tcp, :::8675->8675/tcp             pypowerwall
9a6fdaf7bccc   influxdb:1.8                   "/entrypoint.sh infl…"   15 hours ago   Up 5 minutes   0.0.0.0:8086->8086/tcp, :::8086->8086/tcp             influxdb
stuart@stuart-VirtualBox:~/Desktop/Powerwall-Dashboard$ docker ps
CONTAINER ID   IMAGE                          COMMAND                  CREATED        STATUS          PORTS                                                 NAMES
0f8c2dd377bf   jasonacox/weather411:latest    "python3 server.py"      15 hours ago   Up 10 minutes   0.0.0.0:8676->8676/tcp, :::8676->8676/tcp             weather411
f6637eb9bbeb   telegraf:latest                "/entrypoint.sh tele…"   15 hours ago   Up 10 minutes   8092/udp, 8125/udp, 8094/tcp                          telegraf
f3ec9e49a9b3   grafana/grafana:9.1.2-ubuntu   "/run.sh"                15 hours ago   Up 8 seconds    3000/tcp, 0.0.0.0:9000->9000/tcp, :::9000->9000/tcp   grafana
ff89450f1e0e   jasonacox/pypowerwall:latest   "python3 server.py"      15 hours ago   Up 7 seconds    0.0.0.0:8675->8675/tcp, :::8675->8675/tcp             pypowerwall
9a6fdaf7bccc   influxdb:1.8                   "/entrypoint.sh infl…"   15 hours ago   Up 10 minutes   0.0.0.0:8086->8086/tcp, :::8086->8086/tcp             influxdb
stuart@stuart-VirtualBox:~/Desktop/Powerwall-Dashboard$

jasonacox, 2023-01-08:
Hi @Stoobiedoo - I'm trying to mimic your setup and narrow in on the issue.

* How did you install docker?
* Those containers should NOT be restarting.  Run `docker logs pypowerwall` to see what error is throwing.  I suspect it is having network trouble reaching your Powerwall and in a recycle loop. 
* I have virtualbox (on MacOS) and Ubuntu 22.10 server (or are you using desktop)? 
* What networking mode are you using for your VM (NAT, bridged, etc.)?
* Have you tried using a Ubuntu 20.04 VM?

Stoobiedoo, 2023-01-08:
Thank you so much for the help. I downloaded the 20.4.5 server and ran through the VM config and steps shared. It worked like a charm the first time through. If I get some more time, I will try to run through the steps again with the 22.10 desktop virtual machine to see if I can get it to work. I'm getting a lightweight server to run these dockers instead of my desktop, so I'll report back if I run into any troubles on that machine.

jasonacox-sam, 2026-08-31:
This tracked installs hanging at "Waiting for InfluxDB to start" on Ubuntu 22.04/22.10 hosts. Debugging covered container restart loops and Docker install questions; the reporter got a fully working install by falling back to Ubuntu 20.04, with 22.10 left unverified.

Closing due to age. If InfluxDB fails to start on a current Ubuntu/Docker combination, please open a fresh issue with `docker logs influxdb` output.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam