jasonacox/Powerwall-Dashboard #137: Update to latest on Synology - Grafana docker container restarting in a loop

mp09, 2022-11-22:
Looks like permission problems unexpectedly. 

[image]

BJReplay, 2022-11-22:
This is similar to the problem that I saw on WSL in #129 - I never worked out what caused this, and a couple of re-boots of the host ended up sorting it out.

jasonacox, 2022-11-22:
A few things to try:

1) Check your compose.env file - it should have a setting for the grafana user.  You should change it to the local user (try `id` or `echo $UID` to see your current user):

```bash
GRAFANAUSER="1000:1000"
```

2) If the above is wrong, you will need to `chown -R` the ./grafana folder to the right user.  If easier, you can delete and rebuild the grafana folder (then follow [setup instructions](https://github.com/jasonacox/Powerwall-Dashboard#grafana-setup)):

```bash
docker stop grafana
docker rm grafana
rm -fr grafana
mkdir grafana
./compose-dash.sh up -d
```

jasonacox, 2022-11-22:
That's great news!  Good job.  

Did you happen to discover what went wrong?  If so, let us know so we can add that to the project or documentation to help others.

closed 2023-01-25 by mp09