jasonacox/Powerwall-Dashboard #142: Powerwall PW cannot include certain special characters

jasonc123456, 2022-12-13:
I have experienced this before with docker setting up MarinaDB. Certain special characters like ".&%$" cannot be used for password.
I don't know any workaround(except using a password without special characters), but I just wanted to raise this issue for you so that this can be documented in the installation guide or the setup.sh

In this scenario, this generated an error on your python script with the following error
"  File "/usr/local/lib/python3.8/site-packages/pypowerwall/__init__.py", line 138, in __init__
    self._get_session()
    raise LoginError("Invalid Powerwall Login")
pypowerwall.LoginError: Invalid Powerwall Login"

jasonacox, 2022-12-16:
Thanks @jasonc123456 !  Good catch. I'll look to see if there is a way we can escape those special characters for the `pypowerwall.env` environment settings file.

jasonacox-sam, 2026-08-31:
This flagged that Powerwall passwords containing special characters like `.&%$` break the generated `pypowerwall.env`, with the suggestion to document the limitation (or escape the characters). The easy workaround is choosing a password without those characters.

Closing due to inactivity, though the doc-suggestion is still fair — re-open or PR if you'd like to add a note to the setup guide.

— Sam 🌊

closed 2026-08-31 by jasonacox-sam