jasonacox/Powerwall-Dashboard #148: Support for Multiple Gateways

jasonacox, 2023-01-04:
There are Tesla Solar + Powerwall installations with 400A service connections, where two separate gateways are required.  Currently, to get a dashboard, you need to run two instances, representing each gateway. I'm opening this issue to track the addition of support for this setup and the addition of a feature to provide a combined view.

Assumptions:
* For the primary "Live Monitoring" energy usage graph and the Power Meters, you would want combined data for solar, grid, home and powerwalls.
* For the power flow animation we would need two instances (side-by-side?)
* The detail graphs below would likely work best as separate breakouts for each gateway, but let me know what you think.

_Originally posted by @jasonacox in https://github.com/jasonacox/Powerwall-Dashboard/discussions/62#discussioncomment-4587716_

nberardi, 2023-01-09:
@jasonacox Hey wanted to let you know that I was able to get this working about 90%. Unfortunately it is a pretty major change, and was easier done in InfluxDB 2.x.  

[x] Fully working dashboard for individual power walls
[x] Live Monitoring and Power Meters is working for combined units
[ ] The rest is a work in progress for combined units

Happy to give you a preview of my progress, however I am not quite sure how to export InfluxDB 2.x work.

jasonacox, 2023-01-09:
Awesome! Thanks for the update @nberardi !  Feel free to share details (or even screenshots). 😁 

What caused you to move to InfluxDB 2.x?  I attempted to upgrade and saw that it was a fairly steep breaking change.  For example, all the continuous queries need to be converted from QL to Flux and moved to tasks.  It also failed to run on older ARM RPis, but that is likely less of an issue now.  Would love any tips as part of your effort!

nberardi, 2023-01-09:
@jasonacox It was actually less of a breaking change than you would expect, and in many respects I was able to eliminate most of the CQ's, and simplify the dashboard queries without loss of functionality.  

The only CQ's I kept were `autogen` and `kwh`. The others were easily represented as standard queries with aggreateWindow mean queries.

jasonacox, 2023-01-09:
Awesome!  Thanks @nberardi !

BuongiornoTexas, 2023-01-09:
Count me in for some beta testing if a move to Flux is afoot! 

I'm hoping this will allow for dynamic tracking of ToU billing and valuing internal consumption (at least a lot more easily than this can be done in InfluxQL).

fasternfaster, 2023-02-27:
if you have a fork/branch of the repo to share, i would love to try it out.

jwvanderbeck, 2024-09-19:
Just wondering where this sits? I just had a system finish install today sat my home with split panels so I had to have two gateways and batteries split between my two electrical panels :( Not ideal but wanted to see if this application can help make monitoring easier.

jasonacox, 2024-09-20:
I know there are several in the community who have made this work with quite a bit of customization. Naturally you could se tit up as separate install (someone used separate RPI's) with no change, but that doesn't seem ideal.

If we built multi-PW gateway support, would 2 be enough?

jwvanderbeck, 2024-09-20:
Two gateways is what I have and would seem to be the most common multi-gateway setup for newer homes with split service.  I think this is becoming more common in new construction.

jwvanderbeck, 2024-09-21:
How would the TEDAPI setup work? Clearly both gateways can't be on the same 192.168.91.1 address

jasonacox, 2026-08-21:
The eventual replacement for pypowerwall proxy (now in maintenance mode) will be [pypowerwall-server](https://github.com/jasonacox/pypowerwall-server) which adds the ability to set up multiple gateways. It is a drop in replacement for pypowerwall, but for multiple powerwalls it will require a a definition file (essentially the list of the different gateways).

@jasonacox-sam  can you help us document the steps we will need to plan to swap out the pypowerwall proxy?

jasonacox-sam, 2026-08-21:
@jasonacox — here's a draft migration plan for swapping the pypowerwall proxy for [pypowerwall-server](https://github.com/jasonacox/pypowerwall-server). The good news: it's designed as a drop-in replacement — all legacy proxy endpoints (`/aggregates`, `/vitals`, `/soe`, `/strings`, `/csv/v2`, `/api/meters/aggregates`, etc.) work unchanged on port 8675, so Telegraf and the Grafana dashboards shouldn't need any query changes.

### Migration steps to plan

**1. Compose service swap**
- Replace the `pypowerwall` service image (`jasonacox/pypowerwall` proxy mode) with `jasonacox/pypowerwall-server`.
- Keep the existing container name / port 8675 mapping so Telegraf's `PW_URL` keeps working. TEDAPI mode requires host networking, which the dashboard compose already uses for the proxy.

**2. Environment / config changes (`pypowerwall.env`)**
- Single gateway (TEDAPI): `PW_HOST=192.168.91.1` + `PW_GW_PWD=<gateway password>` — same variables the dashboard already sets, so for most users this is a no-op.
- Cloud mode: `PW_EMAIL` + `PW_AUTH_PATH` pointing at a mounted volume containing `.pypowerwall.auth` / `.pypowerwall.site` (replaces the proxy's `.auth/` mount). One-time `--setup` flow to generate tokens.
- Multi-gateway (the new capability, per this issue): either `PW_GATEWAYS` JSON env var or a `gateways.yaml` file listing each gateway (`id`, `name`, `host`, `gw_pwd`, optional `port`, `timezone`, `type: inverter` for solar-only sites).

**3. User-facing upgrade path**
- For existing installs: pull latest dashboard, swap the service definition in the compose file (delivered via `upgrade.sh` / repo update), then `./compose-dash.sh down` and `./compose-dash.sh up -d`.
- Verify with `curl http://localhost:8675/stats` and `curl http://localhost:8675/version`, then confirm Telegraf is writing and Grafana panels have live data.
- No InfluxDB, Telegraf, or dashboard JSON changes expected — the API surface is proxy-compatible.

**4. Open design questions before we commit to a plan**
- Should multi-gateway users get *one* combined dashboard (aggregating `/api/aggregate/power`, `/api/aggregate/soe`) or per-gateway dashboards? Aggregated endpoints exist, but the current Grafana schema assumes a single site, so aggregation may need Telegraf changes (per-gateway tags).
- Do we keep `weather411` as-is? (Assumed yes — unrelated to the proxy.)
- Cloud-mode auth expiry/re-auth UX — same token-refresh caveats as the proxy, but worth documenting explicitly in the swap guide.
- Whether `setup.sh` should gain a "multiple gateways?" prompt that generates `gateways.yaml` automatically.

**5. Documentation deliverables**
- A `docs/` migration guide (proxy → pypowerwall-server) with rollback steps (pin the old proxy image tag to roll back).
- README section update + note in the release notes when the swap ships.

One caveat: I haven't tested the swap against the dashboard compose stack end-to-end yet — steps 1–2 are from the pypowerwall-server docs, and the Telegraf/Grafana compatibility claim rests on the proxy-compatible endpoint list. A dry-run on a test install before publishing the guide would be the right next step. Happy to draft the doc and a prototype compose change when you want to move on it.

— Sam 🌊