jasonacox/Powerwall-Dashboard #155: backup on a windows machine

prussell69, 2023-01-12:
After having a disaster on my teslamate which I managed to recover, I decided it would be a good idea to "backup" my powerwall dahsboard too.
When I tried to follow your directions using git bash from my powerwall-dashboard folder, the backup.sh would only return "Must run as root".
Since I'm not very savy, I must be doing something wrong. I tried setting git bash to "run as administator" but that didn't seem to work either. Also sudo -i doesn't work. I even tried using windows terminal.
I'm sure you've probably already answered someone else, but I couldn't find anything similar.
What am I doing wrong???

Please help...

BJReplay, 2023-01-12:
sudo -s

Starts a root shell on WSL

That might be enough to then run backup.sh as root.

exit to exit from your root shell.

prussell69, 2023-01-12:
Tried sudo -s on the command line and I get:
sudo : The term 'sudo' is not recognized as the name of a cmdlet, function, script file, or operable program. Check
the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ sudo -s
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (sudo:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
If I try wsl, I only seem to get various errors, depending on what I try.
Since I'm not a Linux user, I seem to struggle with some of these suggestions.

BJReplay, 2023-01-13:
I assumed that you had installed this under WSL, but it seems you didn't, so I probably can't help, other than to suggest you try again running an elevated command prompt, and changing to your powerwall-dashboard folder to run backup.sh from there.  But even if you start bash from an elevated command prompt or powershell, I still think you need to run sudo -s to run as root within bash.

jasonacox, 2023-01-13:
Hi @prussell69 - the only reason `backup.sh` needs to run as root is to ensure it has access to grab all of the InfluxDB files that get created under the docker root user. Check to see if you have any root owned files:

```bash
ls -lR | grep root
```

If you get anything like this, it means you need to be root to backup this data:
```
drwx------ 2 root root 4096 Jan 10 16:09 1400
drwx------ 2 root root 4096 Jan 11 16:10 1402
drwx------ 2 root root 4096 Jan 12 16:09 1404
drwx------ 2 root root 4096 Jan 12 16:00 1406
ls: cannot open directory './influxdb/wal/powerwall/autogen/1144': Permission denied
```

If nothing comes back, you can backup without root.  Just edit `backup.sh` and remove this section at the top:

```bash
if [ "$EUID" -ne 0 ]
  then echo "Must run as root"
  exit
fi
```

I confess, I have near zero expertise on Windows.  I did manage to finally get a windows workstation so I'll see if I can test or tweak the script to work better on windows.

jasonacox, 2023-01-13:
For my windows setup, here is what I did to get backup to work:

## Command Prompt

I didn't need to run as Administrator.  I ran "command" and then ran "bash":

```
Microsoft Windows [Version 10.0.22000.1335]
(c) Microsoft Corporation. All rights reserved.

C:\Users\jason>bash
jason@WINDOWS:/mnt/c/Users/jason$
```

## Edit the backup.sh script

I updated the backup script to have my correct path:

```bash
DASHBOARD="/mnt/c/Users/jason/Powerwall-Dashboard"
```

## Sudo as Root

I ran sudo and was able to get to a root shell and then ran backup.sh:

```bash
jason@WINDOWS:/mnt/c/Users/jason/Powerwall-Dashboard/backups$ sudo su 
[sudo] password for jason:
root@WINDOWS:/mnt/c/Users/jason/Powerwall-Dashboard/backups# ./backup.sh
bash: ./backup.sh: /bin/bash^M: bad interpreter: No such file or directory
```

## Oops!!  Clean Up Backup.sh

Oops! Somehow the backup.sh file has some end of line garbage; how did this happen?  I had to clean it up:

```bash
root@WINDOWS:/mnt/c/Users/jason/Powerwall-Dashboard/backups# sed -i 's/\r$//' backup.sh

root@WINDOWS:/mnt/c/Users/jason/Powerwall-Dashboard/backups# ./backup.sh

Creating InfluxDB Backup
2023/01/13 04:12:58 backing up metastore to /var/lib/influxdb/backups/meta.00
2023/01/13 04:12:58 backing up db=powerwall
...
grafana/plugins/yesoreyeram-boomtable-panel/tests/BoomSeries.spec.js
grafana/png/
grafana/README.md
Cleaning up old backups
Done
```

And now I have backups:

```bash
root@WINDOWS:/mnt/c/Users/jason/Powerwall-Dashboard/backups# ls
backup.sh  Powerwall-Dashboard.2023-01-12.tgz  README.md
```

prussell69, 2023-01-13:
when I enter "bash" in a windows terminal, I get this:
Processing fstab with mount -a failed.
Failed to mount E:\, see dmesg for more details.

<3>WSL (32) ERROR: CreateProcessEntryCommon:358: getpwuid(0) failed 2
<3>WSL (32) ERROR: CreateProcessEntryCommon:362: getpwuid(0) failed 2
<3>WSL (32) ERROR: CreateProcessEntryCommon:570: execvpe /bin/bash failed 2
<3>WSL (32) ERROR: CreateProcessEntryCommon:579: Create process not expected to return

Arrrrghhh!

BJReplay, 2023-01-13:
Hi @prussell69 - you're running into underlying problems relating to your windows subsystem for linux install.  @jasonacox is trying to help you, but as he has said, he's not a windows expert.  To be fair, you've said you're not a linux expert, so you're both operating in the intersection of your lack of expertise.  I'm stepping in because I'm not a linux expert or a WSL expert, but I'm good at google :)

A quick google of https://www.google.com/search?q=WSL+%22Processing+fstab+with+mount+-a+failed.%22 finds a LOT of hits, and a lot of them are related to people who've installed docker and WSL.  Note if you run the google without quotes, it won't find anything, because the -a is interpreted as find all results without "a" in the results - which, as you can imagine, is not very many at all.

This one https://superuser.com/questions/1760513/wsl-error-processing-fstab-with-mount-a-failed looks pretty much identical and suggests a solution that may or may not work:

`wsl --set-default Ubuntu-22.04`

I'd suggest firstly running

`wsl -l -v`

On my PC running Powerwall-Dashboard, this is what I see

```
PS C:\Users\Optiplex> wsl -l -v
  NAME                   STATE           VERSION
* Ubuntu                 Running         2
  docker-desktop-data    Running         2
  docker-desktop         Running         2
PS C:\Users\Optiplex>
```
The default is highlighted with the asterix - so when I run a bash command, that's the shell it runs under.

I'm guessing you see 

```
  NAME                   STATE           VERSION
  docker-desktop-data    Running         2
* docker-desktop         Running         2
```

You may not see an Ubuntu install, or if you do, it's not default.

If you don't see one, you might need to install one - try

`wsl --install Ubuntu`

Then, (or if you saw an Ubuntu), try

`wsl --set-default Ubuntu`

This may be enough to get you going

prussell69, 2023-01-13:
Thank you sooo much for your help. I had to install Ubuntu to be able to get the sudo command from anywhere!
Once I go that, I followed @jasonacox's directions and finally got a backup.
It looks like the backup is inside my Docker enviroment and not accessible from within windows. Is there a way to export the backup to an external location so I can move it off that machine in case of machine failure?

BJReplay, 2023-01-14:
Again google is your friend!

If you google access WSL files from windows you'll probably find some helpful tutorials.

If you open up windows explorer, you'll probably see a Linux icon, and if you navigate via that, you'll probably find your backup.

You've said the back is inside your docker environment, so, assuming you know where it is, you just need to find it from windows - that's probably not that hard.

Here's what my system looks like:

[image]

Note there is a high level folder for Ubuntu, as well as docker-desktop and docker-desktop-data - you'll find your backup in one of those three.

[image]

prussell69, 2023-01-14:
Thanks again. I finally found the file. It turned up on my c drive after all. In the Linux tree (in Explorer) I counldn't access things like:
\\wsl.localhost\docker-desktop\mnt\host\c. No access. Either way, I found my data.
Unfortunately, I'm have issues with the restore on my test machine. I copied the backup file to my test machine, I ran the "./compose-dash.sh stop" from the command line in the Powerwall-Dashboard directory. It ran fine.
When I ran the restore command (tar -zxvf /backups/Powerwall-Dashboard.xyz.tgz) I kept getting:
tar.exe: Error opening archive: Failed to open '/backups/Powerwall-Dashboard.xyz.tgz'

I finally moved the backup file into the Powerwall-Dashboard folder, then ran "tar -zxvf /Powerwall-Dashboard.xyz.tgz", it ran fine. Finally!!
Thanks again to both of you for all of your help, It means the world to people like me.

prussell69, 2023-01-14:
Well, Scrap that!  I ran the scripts on my test machine, and when I load the Grafana page (localhost:9000), I have to redo the username and password, then I have no old data at all. It only shows the last minute or 2 of current info.
When I ran the stop script it opened up a seperate window and ran some commands, then closed. The Docker showed it had stopped the Powerwall. I then ran the restore command same command line (Just a windows command). So I tried to do it again, but this time I ran the tar command from a bash, sudo su. It looked like it ran (just like before), but when I restart the dashboard, it only resets the login password back to 'admin', and still no data. Actually, it deleted any prior data that the dashboard had.
What next????

prussell69, 2023-01-14:
OMG!
Ya know, sometimes you are your own worst enemy...
I looked a little deeper into what was restored on my test machine, and found data from when I first setup the dashboard as a test. It was installed from my user directory, and I ended up moving the install to the root of c:. So when I ran the backup, I chosse the wrong folder. Arrrggghhh...
I try the backup again, and then did the restore on my test station, and viola, data!!

Just out of curiosity, is there a way to make a backup & restore script thats simple to run? I stuggle to remember what to run, where. Some are command line, some are bash, wsl, with/without sudo su, etc. Enonogh to make your head spin sometimes!.
Anyway, ignore my ignorance.
Thanks!!!!!

prussell69, 2023-01-14:
Darn, another update.
I noticed there were 2 big gaps in the restored data. Both were several weeks worth of data. Not sure why.
Another note, When I restarted the dashboard after the backup, my grafana wouldn't restart. I ended up deleteting the grafana container, and re-runing setup.sh to fix it. I'm not sure if it's an issue with my updated dashboard or not, since I had added the database and dashboard to my Teslamate grafana so I could have both things on one dashbaord, and the other day, my Teslamate's grafan crashed after a windows update & reboot. The teslamate had been running for months that way, and had several reboots without issue, so I don't know what happened. I have a ticket into Telsmate to see if he can help with that.
Just curious why the powerwall's grafana crashed today, too.

jasonacox, 2023-01-14:
> Just out of curiosity, is there a way to make a backup & restore script thats simple to run? 

Great question.  In short, yes.  What is there was a quick hack I haven't touched in quite a while before you opened this issue. I'll add a TODO but I definately welcome any help for anyone who wants to try to tackle it.

> I noticed there were 2 big gaps in the restored data. 

Odd. I've never seen this occur.  I've done hundreds of backup/restores as part of my test pipeline for new releases.  I assume you verified that this data was in fact in the old setup and not missing there too?  Also, and you may not have known this, you need to completely stop the InfluxDB and grafana containers (I stop all the containers) if you do a restore.  If you don't it will collide and data corruption could result.  I would check your influxdb logs (`docker logs influxdb`) to make sure there aren't other errors. 

> When I restarted the dashboard after the backup, my grafana wouldn't restart. 

I've seen this before, it can occur when grafana is running when you do a backup or restore.  Did you see what errors it was throwing (`docker logs grafana`)?  To your point about improving the backup script, it should either omit backing up grafana (my preference) or actually stop grafana before doing the snapshot to ensure state gets recorded accurately. In my testing, I don't back up grafana as I want to replicate a clean install.  I should remove that part of the backup script so that it makes it easy for people to port their data without impacting grafana.

prussell69, 2023-01-14:
1. I noticed there were 2 big gaps in the restored data.
I looked at the data from my running system side-by-side with the test system, and the running system had data going back >year. I'm not sure if it effects the backup/restore process, but when I started the system, I had to import data from my tesla account for the 1st 8 or 9 months. But when I set the dashboard to show the last 12 months, the main graph shows data, where the restored test station shows gaps.
I did stop the container of the whole system in Docker prior to doing both the backup, and the restore. I made sure it showed 'exited' both times. I looked at the influxdb log files, but not knowing exactly what I am looking for, I didn't see anything that looked like an error.

2. When I restarted the dashboard after the backup, my grafana wouldn't restart. 
When the Grafana crashed, I couldn't look at the errors, since the container wasn't running. I only knew how to look at the log file from within Docker. If there is another way, I don't know it. I will have to wait until it happens again once I learn how.

If there is anything I can send you that would help, please let me know. I'd love to find out why Grafana crashed on me on both systems.