jasonacox/Powerwall-Dashboard #164: unRAID Setup

jasonacox, 2023-01-27:
[@drnick5](https://www.reddit.com/user/drnick5/) on [Reddit](https://www.reddit.com/r/TeslaSolar/comments/10ll6mr/comment/j5ypgu4/?utm_source=reddit&utm_medium=web2x&context=3)

Any place have a tutorial on how to set this up? I run unRAID and have Dockers for other things, but couldn't get this to work.

jasonacox, 2023-01-27:
It would be helpful to get any details/error during the "setup" part.  

Also, if docker compose is not available on unRAID, a [manual setup](https://github.com/jasonacox/Powerwall-Dashboard#option-2---manual-install) of each container could work but would require configuration changes.  For example, you would replace the hostnames with `localhost`.   You would skip the `./compose-dash.sh up -d` step and run individual `docker run` commands for each component:

```bash
# weather411
docker run \
-d \
-p 8676:8676 \
-e WEATHERCONF='/var/lib/weather/weather411.conf' \
-v ${PWD}:/var/lib/weather \
--name weather411 \
--restart unless-stopped \
jasonacox/weather411

# pypowerwall
docker run \
-d \
-p 8675:8675 \
-e PW_EMAIL=email@example.com \
-e PW_PASSWORD=password \
-e PW_HOST=10.0.1.2 \
-e PW_TIMEZONE=America/Los_Angeles \
--name pypowerwall \
--restart unless-stopped \
jasonacox/pypowerwall

# etc
```

willyp713, 2024-06-17:
Unraid has an add-on to enable compose support called [Docker Compose Manager](https://forums.unraid.net/topic/114415-plugin-docker-compose-manager/), but it's still a little clunky because it's not native and uses some non-conventional UI tricks. I'm going to try to get this project going that way and will report back if I find anything helpful.

Incidentally, Grafana (and InfluxDB and Telegraf) are both super popular in the Unraid world and have many easy paths to start. Perhaps there's another angle here to create some sort of tutorial or "easy-mode" path to get this project set up for existing installations of Grafana and InfluxDB?

For example: if you already have the Grafana ecosystem working, it would be conceptually easy to add the `pypowerwall` and `weather411` containers and follow the manual config steps in https://github.com/jasonacox/Powerwall-Dashboard/blob/main/README.md

jasonacox, 2026-03-02:
Great news — the solution shared by @znahman and @willyp713 has been added to our new **[Powerwall Dashboard FAQ](https://github.com/jasonacox/Powerwall-Dashboard/discussions/755)**! 🎉

The short answer for unRAID users:
- **With Docker Compose Manager plugin:** install the plugin, then run `./setup.sh` and `./compose-dash.sh up -d` as normal
- **Without the plugin:** follow [Option 2 - Manual Install](https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#option-2---manual-install) and start each container individually

Closing this issue since it is now covered in the FAQ. Feel free to comment there if you run into anything else!