jasonacox/Powerwall-Dashboard #170: Influxdb2 and measurements

dakky, 2023-02-02:
Hello

I need to use a pre existing influxdb2. In Telegraf I configured a new [output](https://github.com/influxdata/telegraf/tree/master/plugins/outputs/influxdb_v2) in telegraf.local and removed the old influxv1 one from telegraf.conf:
```
[[outputs.influxdb_v2]]
  urls = ["http://influxdb:8086"]
  token = "XXXXXXX"
  organization = "myorg"
  bucket = "powerwall"
```

that seems to work fine, telegraf logs no error and i get data in my bucket. 
In order to use the grafana dashboards **without** rewriting everything to `flux` and use `influxQL`  I reconfigured my influxv2 to also accept `influxQL` queries (which basically means to map a _virtual_ database to a bucket and add a user and password with read access to this database). Then I added a influx v1 datasource to grafana. The test of the datasource itself was successful.

Everthing fine until now. And now the Issue: none of the graphs is working. When looking in a simpler panel there is the following statement:
```
SELECT integral("home")  / 1000 / 3600 FROM "autogen"."http" WHERE time >= now() - 5m and time <= now() tz('Europe/Berlin')
```

I do not have a measurement called `home` in my bucket. I have something like `load_instant_power`, which is the exact name of the field from pypowerwall, caught by telegraf and written to influx. Where does this `home` come from? Or that `from pv` and so on?

BJReplay, 2023-02-02:
Did you run influxdb/influxdb.sql to create the CQs?

That's where home and the other measurements used by the graphs are created.

You may need to tweak for your setup.

jasonacox, 2023-02-03:
Hi @dakky - thanks for raising the issue.  I suspect there are others that will want to use an existing InfluxDB v2 installation instead of the project default v1.8.

@BJReplay is correct, the measurements you are looking for are created by the continuous queries (CQs) that are set up as show here: https://github.com/jasonacox/Powerwall-Dashboard/blob/main/influxdb/influxdb.sql

I believe InfluxDB 2.x uses tasks instead of CQs, but perhaps you can figure out a way to make them work.  They are in influxQL.  Please let us know what you discover.

nberardi, 2023-02-27:
InfluxQL only works against InfluxDB1.x.  You need to use Flux against InfluxDB2.x+.  I have a working solution in my branch that I am readying, however it is a breaking change against the current approach.  

https://github.com/nberardi/Powerwall-Dashboard/commit/d0da1cc61039b007e39ba5e14783dd97117f91b5

jasonacox, 2023-02-28:
Thanks for sharing that @nberardi .  Converting to InfluxDB2 is on the roadmap but non-trival for data migration.

nberardi, 2023-02-28:
I was lucky I started from scratch in my case. The data migration is pretty straightforward. The part that made it tricky was the queries and getting them structured in a way that made sense for all the Grafana dashboard graphs. Sent from my iPhone, expect autocorrection issues

dakky, 2023-12-28:
when implementing this feature: please  have in mind users, which are not running the setup.sh but installed this manually ^^

nberardi, 2023-12-28:
@dakky i am not sure what the difference would be, can you share?

dakky, 2023-12-28:
@nberardi The difference is, that `setup.sh` hides what happens inside. So I'm hoping for a **manual guide**, how to switch from influx1 to influx2. In my case i have copied much of the compose files into an existing setup, because i already have a grafana & influx running.

thenoid, 2025-04-15:
Not seeing this merged to master.  Did this die?