jasonacox/Powerwall-Dashboard #179: Net Grid Usage Power Meter

wreiske, 2023-02-11:
I switched over to TOU instead of self-powered and have been keeping an eye on my Net Grid Usage in the Tesla app. Others may find this number valuable too. Would be great to include it on the standard dashboard!

[image]

[image]
[image]

```
{
  "id": 69,
  "gridPos": {
    "h": 2,
    "w": 3,
    "x": 21,
    "y": 15
  },
  "type": "stat",
  "title": "Net Grid Usage",
  "datasource": {
    "uid": "39FBA0KVk",
    "type": "influxdb"
  },
  "pluginVersion": "9.1.2",
  "maxDataPoints": 100,
  "links": [],
  "fieldConfig": {
    "defaults": {
      "mappings": [
        {
          "options": {
            "match": "null",
            "result": {
              "text": "N/A"
            }
          },
          "type": "special"
        }
      ],
      "thresholds": {
        "mode": "absolute",
        "steps": [
          {
            "color": "purple",
            "value": null
          }
        ]
      },
      "color": {
        "mode": "thresholds"
      },
      "decimals": 1,
      "unit": "kwatth"
    },
    "overrides": []
  },
  "options": {
    "reduceOptions": {
      "values": false,
      "calcs": [
        "sum"
      ],
      "fields": ""
    },
    "orientation": "horizontal",
    "textMode": "auto",
    "colorMode": "value",
    "graphMode": "none",
    "justifyMode": "auto",
    "text": {
      "valueSize": 30
    }
  },
  "targets": [
    {
      "datasource": {
        "type": "influxdb",
        "uid": "39FBA0KVk"
      },
      "groupBy": [],
      "measurement": "http",
      "orderByTime": "ASC",
      "policy": "autogen",
      "refId": "A",
      "resultFormat": "time_series",
      "select": [
        [
          {
            "params": [
              "from_grid"
            ],
            "type": "field"
          },
          {
            "params": [],
            "type": "integral"
          },
          {
            "params": [
              " / 1000 / 3600"
            ],
            "type": "math"
          }
        ]
      ],
      "tags": [],
      "query": "SELECT integral(\"from_grid\")  / 1000 / 3600 FROM \"autogen\".\"http\" WHERE $timeFilter",
      "rawQuery": true,
      "hide": true
    },
    {
      "datasource": {
        "uid": "39FBA0KVk",
        "type": "influxdb"
      },
      "refId": "B",
      "hide": true,
      "query": "SELECT integral(\"to_grid\")  / 1000 / 3600 FROM \"autogen\".\"http\" WHERE $timeFilter",
      "rawQuery": true,
      "resultFormat": "time_series"
    },
    {
      "refId": "C",
      "datasource": {
        "type": "__expr__",
        "uid": "__expr__",
        "name": "Expression"
      },
      "type": "math",
      "hide": false,
      "expression": "$A - $B "
    }
  ],
  "transparent": true,
  "interval": ""
}
```

[image]

I'm not sure if this is the best way to grab the details (maybe two sub-selects and doing the math in a single query would be better?), but it works for me!

jasonacox, 2023-02-11:
Thanks @wreiske !  I like it. The simple approach would be to combine it into one query instead of running the expression. I believe this should provide the same result:

```sql
SELECT (integral("from_grid") - integral("to_grid")) / 1000 / 3600 
    FROM "autogen"."http" 
    WHERE $timeFilter
```

wreiske, 2023-02-11:
> ```sql
> SELECT (integral("from_grid") - integral("to_grid")) / 1000 / 3600 
>     FROM "autogen"."http" 
>     WHERE $timeFilter
> ```

Works great!

```
{
  "id": 69,
  "gridPos": {
    "h": 2,
    "w": 3,
    "x": 21,
    "y": 15
  },
  "type": "stat",
  "title": "Net Grid Usage",
  "datasource": {
    "uid": "39FBA0KVk",
    "type": "influxdb"
  },
  "pluginVersion": "9.1.2",
  "maxDataPoints": 100,
  "links": [],
  "fieldConfig": {
    "defaults": {
      "mappings": [
        {
          "options": {
            "match": "null",
            "result": {
              "text": "N/A"
            }
          },
          "type": "special"
        }
      ],
      "thresholds": {
        "mode": "absolute",
        "steps": [
          {
            "color": "purple",
            "value": null
          }
        ]
      },
      "color": {
        "mode": "thresholds"
      },
      "decimals": 1,
      "unit": "kwatth"
    },
    "overrides": []
  },
  "options": {
    "reduceOptions": {
      "values": false,
      "calcs": [
        "sum"
      ],
      "fields": ""
    },
    "orientation": "horizontal",
    "textMode": "auto",
    "colorMode": "value",
    "graphMode": "none",
    "justifyMode": "auto",
    "text": {
      "valueSize": 30
    }
  },
  "targets": [
    {
      "datasource": {
        "type": "influxdb",
        "uid": "39FBA0KVk"
      },
      "groupBy": [],
      "measurement": "http",
      "orderByTime": "ASC",
      "policy": "autogen",
      "refId": "A",
      "resultFormat": "time_series",
      "select": [
        [
          {
            "params": [
              "from_grid"
            ],
            "type": "field"
          },
          {
            "params": [],
            "type": "integral"
          },
          {
            "params": [
              " / 1000 / 3600"
            ],
            "type": "math"
          }
        ]
      ],
      "tags": [],
      "query": "SELECT (integral(\"from_grid\") - integral(\"to_grid\")) / 1000 / 3600 \n    FROM \"autogen\".\"http\" \n    WHERE $timeFilter",
      "rawQuery": true,
      "hide": false
    }
  ],
  "transparent": true,
  "interval": ""
}
```

[image]

jasonacox, 2023-02-12:
Can you think of any other "meter" we should add?   By adding this one, we added some space to the "Power Meters" row.  ;)

wreiske, 2023-02-12:
> Can you think of any other "meter" we should add? By adding this one, we added some space to the "Power Meters" row. ;)

Maybe net power wall usage?

```sql
SELECT (integral("from_pw") - integral("to_pw")) / 1000 / 3600 
    FROM "autogen"."http" 
    WHERE $timeFilter
```

[image]

Here's what that looks like for last 30 days for me:
[image]

jasonacox, 2023-02-12:
@wreiske You have inspired me. :)   

The "Net Grid" is a great metric to help determine how well we are doing in keeping our kWh costs from the utility low (net zero even).  However, I'm also keen on knowing what % of our home is powered by pure renewable energy (% solar powered). That is the main reason I bought powerwalls anyway. 

To experiment, I added a new meter panel next to solar to show "% Solar Powered":

[image]

The assumption is that anything from the grid is "not solar" (either directly or indirectly via Powerwalls).  That would mean that the difference between what the home uses and what came from the grid would be "solar" powered. It ignores any solar energy we generate that we push to the grid because that is for monetary gain (which is fine, don't get me wrong :-), but doesn't contribute toward powering the home with pure renewable solar energy. 

The equation I'm using to determine % solar powered is:

( "Home Usage" - "From Grid" ) / "Home Usage"

The resulting query is fairly simple using that:

```sql
SELECT (home - from_grid) / home FROM 
  (SELECT integral("home")  AS home, integral("from_grid")  AS from_grid 
      FROM "autogen"."http" WHERE $timeFilter)
```

The challenge with any dashboard is figuring out what you should add for everyone vs make optional.  I wonder if "Net Grid Usage" or "% Solar Powered" would resonate with anyone/everyone else?  🤔

wreiske, 2023-02-12:
Very cool! Now we need a % Grid Powered.... 😆

jasonacox, 2023-02-12:
You are correct, it is very close.  It may be close enough.   

The nuance is that the "Self-Powered" pie-chart data doesn't know how much of the Powerwall energy came from Solar vs Grid (e.g. Storm Watch charging or Grid Charing). It basically just gives the breakdown on what is powering the home directly (solar, powerwall or grid). 

The % Solar Powered doesn't care how the powered is delivered (direct or via powerwall) it just cares about what % of the home energy use originated from Solar production.

jasonacox, 2023-02-21:
@wreiske The meter has been added to the default dashboard.json.  Thanks for this great suggestion and help getting it built. 🙏

closed 2023-02-21 by jasonacox