jasonacox/Powerwall-Dashboard #180: Power Flow pane on dashboard used to animate, now just shows code strings

ibmaster, 2023-02-14:
[image]
I can't seem to resolve this, tried reverting to old dashboard.json files without any luck. Running release 2.8.2.

jasonacox, 2023-02-14:
Hi @ibmaster - did you recently upgrade or make a manual change?  The `upgrade.sh` script should have helped update all of the needed bits, but here are a few things to try:

1) Test the animation by going to http://localhost:8675/example.html (or replace localhost with the IP address of the system running the Dashboard)

[image]

2) Check that Grafana is set to render HTML in that panel:

Click on "Power Flow v" and select Edit: 

[image]

Panel options to the right should look like:

[image]

3) Make sure grafana container is running with the right environmental variables:

```bash
docker exec -it grafana sh
set | grep GF
```

You should see:
```
$ set | grep GF
GF_ANALYTICS_CHECK_FOR_UPDATES='false'
GF_ANALYTICS_REPORTING_ENABLED='false'
GF_INSTALL_PLUGINS='grafana-piechart-panel, natel-plotly-panel, blackmirror1-singlestat-math-panel, btplc-trend-box-panel, marcuscalidus-svg-panel, michaeldmoore-multistat-panel, https://github.com/yesoreyeram/yesoreyeram-boomtable-panel/releases/download/v1.5.0-alpha.3/yesoreyeram-boomtable-panel-1.5.0-alpha.3.zip;yesoreyeram-boomtable-panel, ryantxu-ajax-panel, grafana-influxdb-flux-datasource, fetzerch-sunandmoon-datasource, andig-darksky-datasource, simpod-json-datasource'
GF_PANELS_DISABLE_SANITIZE_HTML='true'
GF_PATHS_CONFIG='/etc/grafana/grafana.ini'
GF_PATHS_DATA='/var/lib/grafana'
GF_PATHS_HOME='/usr/share/grafana'
GF_PATHS_LOGS='/var/log/grafana'
GF_PATHS_PLUGINS='/var/lib/grafana/plugins'
GF_PATHS_PROVISIONING='/var/lib/grafana/provisions'
GF_SERVER_HTTP_PORT='9000'
```
The key you are looking for in the above is `GF_PANELS_DISABLE_SANITIZE_HTML='true'`

If you don't see that, check your `grafana.env` file to make sure it is set.  If it is, check your `powerwall.yml` to make sure it looks like this:

```yml
    grafana:
        image: grafana/grafana:9.1.2-ubuntu
        container_name: grafana
        hostname: grafana
        restart: always
        user: "${GRAFANAUSER}"
        volumes:
            - type: bind
              source: ./grafana
              target: /var/lib/grafana
        ports:
            - target: 9000
              published: 9000
              mode: host
        env_file:
            - grafana.env
        depends_on:
            - influxdb

```

ibmaster, 2023-02-14:
Thanks for the help! Steps 1 and 2 look correct to what you listed.
Step 3 errors out when executing the docker command…. which is weird because I can access everything else my Powerwall Dashboard in Grafana

docker exec -it grafana sh
Error response from daemon: Container 7f8df90e9990a16e94a2006fdcf8da47835aa883945ea023a8b8335493410aa5 is restarting, wait until the container is running

ibmaster, 2023-02-14:
Adding the Grafana environment variable GF_PANELS_DISABLE_SANITIZE_HTML='true' fixed this issue. 
Thank you for the assistance!

closed 2023-02-14 by ibmaster