jasonacox/Powerwall-Dashboard #187: Troubleshooting Tool to Help Verify Setup

jasonacox, 2023-02-21:
We seem to have an increasing number of users having setup problems. We spend a lot of time going back and forth doing basic troubleshooting steps to see what part of the stack isn't working correctly.  It would be super amazing if we had a simple `verify.sh` script that tests all of the components for health and reports exactly which parts of the stack are working correctly or where it sees errors.  

_Originally posted by @jasonacox in https://github.com/jasonacox/Powerwall-Dashboard/issues/22#issuecomment-1437650468_

jasonacox, 2023-02-21:
Example run:

[image]

You can disable color with "-no-color" switch:

$ bash verify.sh -no-color

```
Verify Powerwall-Dashboard - Version 2.8.4 - Timezone: America/Los_Angeles
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.6.0 Proxy t24
 - Powerwall State: CONNECTED - Firmware Version: 22.26.4

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.25.2 (git: HEAD@67c08c51)

Checking influxdb
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Config File weather/weather411.conf: GOOD
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 16.26}
 - Version: 0.1.2
```

BJReplay, 2023-02-21:
That is so cool!

```
~/Powerwall-Dashboard$ git pull
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 5 (delta 2), reused 5 (delta 2), pack-reused 0
Unpacking objects: 100% (5/5), 2.12 KiB | 543.00 KiB/s, done.
From https://github.com/jasonacox/Powerwall-Dashboard
   10a202b..c5390a6  main       -> origin/main
Updating 10a202b..c5390a6
Fast-forward
 verify.sh | 243 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 1 file changed, 243 insertions(+)
 create mode 100755 verify.sh
```

Here's what it reports for me (because I'm a non-Weather411-rebel):

```
~/Powerwall-Dashboard$ ./verify.sh
```

[image]

jasonacox, 2023-02-21:
Awesome!  Thanks @BJReplay for helping verify verify.sh. 😂 

It does just the basics. I'm sure we could add more to help validate config settings, etc., but at least it should help with setup troubleshooting. 🙏

commit 24100697 (2023-02-21): v2.8.5 verify tool closes #187
closed 2023-02-21 by jasonacox