jasonacox/Powerwall-Dashboard #193: SolarEdge Inverter Data

jasonacox, 2023-02-25:
**Problem**
Some Powerwall systems use an external inverter like the SolarEdge. Specific data on the inverter does not show up in the Powerwall Gateway APIs. This came up in the discussion https://github.com/jasonacox/Powerwall-Dashboard/discussions/190.

**Enhancement**
Using the SolarEdge local ModBusTCP interface, we could grab the inverter data directly. An optional container extension could be made to the docker-compose setup to launch a service that pushing this SolarEdge data into InfluxDB for a Grafana dashboard. 

Here are some example projects that pull in SolarEdge data:
* https://github.com/ZerxXxes/solaredge_modbus2influx - pulls data from local ModBusTCP endpoint and pushes to InfluxDB - also provides information on how to activate the ModBus TCP option.
* https://github.com/salberin/Solaredge-influxdb - similar to above, pulls from local ModBusTCP endpoint and pushes to InfluxDB - could be converted to a dockerhub hosted container (needs Dockerfile updated)
* https://github.com/ndejong/solaredge-interface - uses the internet based SolarEdge API service so this would not be using the local interface but may be a good/better option for some.

seenhear, 2023-03-16:
OK so I'm only just starting to research this project, and I have a Tesla solar system that uses a SolarEdge inverter.  I also have 2x powerwalls.
So, with that solaredge inverter, what would I get, if anything, from PW-Dashboard?  I'm thinking this wouldn't be worth the effort to set up at this point (I'm not a programmer, and not super facile in *nix OS stuff.)

jasonacox, 2023-03-16:
Hi @seenhear - If you have Powerwalls, you would get the dashboard data as shown [here](https://github.com/jasonacox/Powerwall-Dashboard#powerwall-dashboard).  However, you would not see the Powerwall+ data (string data, voltages, frequencies) but would still get the capacity data and temps.  This issue #193 is to see if we can add additional data from the SolarEdge system.  To your point, it may not be worth the effort and the existing Powerwall-Dashboard may suit your needs for trending data, etc.

For those without Powerwalls, we are exploring ways to capture and graph Tesla Solar data in a similar fashion [here](https://github.com/jasonacox/Powerwall-Dashboard/issues/183).

dkerr64, 2023-03-16:
I also have SolarEdge inverters (2 of them).  The SolarEdge app cannot talk to them, I was told (two years ago) that I needed to call my installer and request a password and/or ask them to enable access to the inverters.  I tried that and Tesla refused to give me access.  Googling at the time I came across comments in reddit and elsewhere that stated that Tesla used to provide this info / unlock access but shortly before I wanted it they decided to stop permitting this.

Has anything changed?  Is it now possible to access the SolarEdge inverters?

Thanks
David

seenhear, 2023-03-17:
Hi David,
I had a Tesla service person out not too long ago (late last fall) to install some missing CTs that were never included in the original installation.  At that time he said Tesla would give me access if I requested it.  So I did, and I got it.  Not sure how recently you've tried, but that's my experience.

seenhear, 2023-03-17:
@jasonacox another question for you, but probably not worth opening a new issue: the windows install instructions say to setup Linux in Windows using WSL, then they say to install both git for windows and docker for windows.  If you have Linux running via WSL, why do you need git for windows?  Couldn't you just run git from the Linux CLI?  Likewise for docker, wouldn't you just install docker via Linux, rather than the windows version?

jasonacox, 2023-03-18:
Hi @seenhear !  Great question! Unfortunately, I know very little about Windows (only use Mac and LInux). I added what was recommended by others who were able to successfully install the Dashboard on Windows.  I made many attempts myself and found Linux on Windows is a bit more mystical art than a science. 🤷  Thankfully, those instructions did work for me. However, I would love to make it easier!  Do you have a set of instructions we could use instead?

BJReplay, 2023-03-18:
Chipping in here re @seenhear's question re git and docker: but based on understanding, not testing...

I'm almost certain that you have to install Docker for Windows to get docker working under WSL - so that part of the instructions is correct.  (in fact, from memory, I might have tried skipping the installing docker for windows, and was unsuccessful until I installed it, and then came across something that confirmed that you had to install it - but I may be mis-remembering).

I'm almost certain that you don't need to install git for windows to get git commands working under WSL, so that part of the command can probably be skipped.  It certainly can't hurt to try :)

However, I haven't tested and don't have a test box or VM handy to test either.

seenhear, 2023-03-18:
> Hi @seenhear ! Great question! Unfortunately, I know very little about Windows (only use Mac and LInux). I added what was recommended by others who were able to successfully install the Dashboard on [...]

Thanks!  I do not have any instructions yet.  If I do bother to get it working, I'll be sure to let you know.  In fact I'm sure I'll be here asking questions.  LOL

jasonacox-sam, 2026-09-06:
This was a backlog enhancement to pull SolarEdge inverter data via its local ModBus TCP interface into the dashboard as an optional container. The discussion surfaced that Tesla locks down SolarEdge access for many installs (installer password required, sometimes granted on request), and no ModBus integration was ever built.

Closing due to inactivity. Powerwall-Dashboard's supported data sources have grown since (see [v5.2.x releases](https://github.com/jasonacox/Powerwall-Dashboard/releases)), but SolarEdge remains out of scope without a community contributor driving it. If you have a SolarEdge system and want to build the exporter, please re-open and we'll help.

— Sam 🌊

closed 2026-09-06 by jasonacox-sam