jasonacox/Powerwall-Dashboard #199: Weather411 unable to write data to InfluxDB v1.8

sumnerboy12, 2023-02-28:
**Problem**
After upgrading to v0.2.0 of weather411 I was not seeing any weather data in my Powerwall dashboard. Enabling debug logging and restarting showed the following;

```
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:Weather411 [0.2.0]
powerwall_weather.1.dgnjryq9xnzr@nuc02    |
powerwall_weather.1.dgnjryq9xnzr@nuc02    | Weather411 Server [0.2.0]
powerwall_weather.1.dgnjryq9xnzr@nuc02    | * Configuration Loaded [/var/lib/weather/weather.conf]
powerwall_weather.1.dgnjryq9xnzr@nuc02    |  + Weather411 - Debug: True, Activate API: True, API Port: 8676
powerwall_weather.1.dgnjryq9xnzr@nuc02    |  + OpenWeatherMap - Key: zzz, Wait: 10, Units: metric
powerwall_weather.1.dgnjryq9xnzr@nuc02    |  + OpenWeatherMap - Lat: xxx, Lon: yyy, Timeout: 10
powerwall_weather.1.dgnjryq9xnzr@nuc02    |  + InfluxDB - Enable: True, Host: db, Port: 8086, DB: powerwall, Field: weather
powerwall_weather.1.dgnjryq9xnzr@nuc02    | * Starting threads
powerwall_weather.1.dgnjryq9xnzr@nuc02    |  + fetchWeather thread
powerwall_weather.1.dgnjryq9xnzr@nuc02    |  + apiServer thread - Listening on http://localhost:8676
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:Starting new HTTPS connection (1): api.openweathermap.org:443
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:https://api.openweathermap.org:443 "GET /data/2.5/weather?lat=xxx&lon=yyy&units=metric&appid=zzz HTTP/1.1" 200 489
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:Weather data loaded
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:Writing to InfluxDB
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:Cannot parse error response to JSON: 404 page not found
powerwall_weather.1.dgnjryq9xnzr@nuc02    | , Extra data: line 1 column 5 (char 4)
powerwall_weather.1.dgnjryq9xnzr@nuc02    | DEBUG:Error writing to InfluxDB
```

**To Reproduce**
Surprised no one else is reporting this. Perhaps something to do with my setup.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Host System**
 - Intel NUC
 - Ubuntu
 - Running in Docker Swarm

**Additional context**
The error "Cannot parse error response to JSON" doesn't show up in any good search, except [here](https://github.com/influxdata/influxdb-client-python/issues/166). I wonder if the weather script is incorrectly attempting to use InfluxDB v2 connection properties, when I am still on v1.8.

My `weather.conf` is unchanged from v0.1.2;
```
[Weather411]
DEBUG = yes

[API]
# Port to listen on for requests (default 8676)
ENABLE = yes
PORT = 8676

[OpenWeatherMap]
# Register and get APIKEY from OpenWeatherMap.org
APIKEY = openweathermap_apikey
# Enter your location in latitude and longitude
LAT = xxx
LON = yyy
WAIT = 10
TIMEOUT = 10
# standard, metric or imperial
UNITS = metric

[InfluxDB]
# Record data in InfluxDB server
ENABLE = yes
HOST = db
PORT = 8086
DB = powerwall
FIELD = weather
# Leave blank if not used
USERNAME = powerwall
PASSWORD = influxdb_password
```

I have reverted back to v0.1.2 and now my weather data is flowing thru to InfluxDB and my dashboard.

jasonacox, 2023-03-01:
Thanks for opening this!  

I cannot duplicate the error on my test systems (all work) including an intel+ubuntu system, but I'm not using docker swarm. 

Would you be willing to test a debug version that should expose more data on the exception stack?

```
jasonacox/weather411:0.2.0beta
```

jasonacox, 2023-03-01:
> USERNAME = powerwall
> PASSWORD = influxdb_password

This seems to be the issue.  The default install for the Dashboard project and my tests use no username or password for InfluxDB.  I just tested using user/pass and I am getting the same DEBUG error you are getting.  While I research, I'm going to revert the latest to v0.1.2 in case there are others who have customized their install to use credentials.

Thanks again for reporting this!

jasonacox, 2023-03-01:
I think I found the problem. The new influxdb-client requires the username and password to be sent via token for connections to InfluxDB 1.8.  I was able to test by activating auth and a user/pass on a test DB.  It is now working in my tests.  I've pushed the changes in this commit: https://github.com/jasonacox/Powerwall-Dashboard/commit/831f88eced955f71b139aa218e04db48fc002e08

You can test it by downloading:

```
jasonacox/weather411:0.2.1
```

I would love to hear if this fixes your problem.  Thanks again for your help.

sumnerboy12, 2023-03-01:
I will test now, thank you for the very prompt action on this!! Love this dashboard, only just got my Powerwall installed a few months ago, but this setup is fantastic!!

sumnerboy12, 2023-03-01:
Yep that seems to be working fine - great stuff - thanks very much!!

closed 2023-03-01 by sumnerboy12

jasonacox, 2023-03-02:
Awesome! Once again, thank you for opening this issue. Good find. 🙏 

I just set v0.2.1 as 'latest'.  Anyone running upgrade.sh or pulling it manually will get the fix. Please report any other issues.

`./verify.sh` output:

[image]