jasonacox/Powerwall-Dashboard #214: Error response from daemon: Cannot restart container weather411: permission denied

kingkaden123, 2023-03-18:
**Problem Description**
Hello! New to docker and trying to install Powerwall-Dashboard. I've gotten to running the setup.sh command but it keeps giving an error (Error response from daemon: Cannot restart container weather411: permission denied) when I go through the setup.

**System Details**
Describe the host system you are trying to use for the Dashboard: Desktop
* Hardware I think a Pentium and 4GB of RAM
* OS Version Ubuntu 22.04.02
* Other Specifics

**Verify Tool Output**
Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware Version: Unknown

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.26.0 (git: HEAD@064650c8)

Checking influxdb
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): ERROR: Not Listening - Logs:
---
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
---
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Config File weather/weather411.conf: GOOD
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": null}
 - Version: 0.2.1

jasonacox, 2023-03-18:
> Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

This seems to indicate a routing problem with your linux host, or something in the docker setup that is not allowing it to open traffic to the public internet or even to your local network.  Do you have a firewall or have you set up iptables on your host?  This would explain the errors in Grafana and the reason that pypowerwall is not listening (it can't reach your Powerwall).

Can you run a few command from your host to help troubleshoot?

```bash
# pypowerwall - verify tool shows that it is running but not listening
docker logs pypowerwall

# install pypowerwall and run a scan to see if connectivity works
pip install pypowerwall
python -m pypowerwall scan

# grafana - check to see if you can access public internet
curl -is https://grafana.com/api/plugins/repo/grafana-piechart-panel | head -12

# weather411 - check logs then manually try to restart to see if you get same permission error
docker logs weather411
docker stop weather411
docker start weather411
```

kingkaden123, 2023-03-19:
Nope, no firewall or iptables. 

Here's the logs from "docker logs pypowerwall"
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.10/site-packages/urllib3/connection.py", line 363, in connect
    self.sock = conn = self._new_conn()
  File "/usr/local/lib/python3.10/site-packages/urllib3/connection.py", line 179, in _new_conn
    raise ConnectTimeoutError(
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPSConnection object at 0x7f7e857e4ca0>, 'Connection to 192.168.68.154 timed out. (connect timeout=10)')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.10/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.68.154', port=443): Max retries exceeded with url: /api/login/Basic (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f7e857e4ca0>, 'Connection to 192.168.68.154 timed out. (connect timeout=10)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 146, in _get_session
    r = self.session.post(url,data = pload, verify=False, timeout=self.timeout)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 635, in post
    return self.request("POST", url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 553, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPSConnectionPool(host='192.168.68.154', port=443): Max retries exceeded with url: /api/login/Basic (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f7e857e4ca0>, 'Connection to 192.168.68.154 timed out. (connect timeout=10)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/server.py", line 107, in <module>
    pw = pypowerwall.Powerwall(host,password,email,timezone,cache_expire,timeout,pool_maxsize)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 138, in __init__
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 151, in _get_session
    raise ConnectionError(err)
pypowerwall.ConnectionError: Unable to connect to Powerwall at https://192.168.68.154
03/19/2023 12:10:19 AM [proxy] [INFO] pyPowerwall [0.6.1] Proxy Server [t24] - HTTP Port 8675
03/19/2023 12:10:19 AM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
  File "/usr/local/lib/python3.10/site-packages/urllib3/util/connection.py", line 95, in create_connection
    raise err
  File "/usr/local/lib/python3.10/site-packages/urllib3/util/connection.py", line 85, in create_connection
    sock.connect(sa)
TimeoutError: timed out

When I tried to run the install command it responded with this, 
Command 'pip' not found, but can be installed with:
apt install python3-pip
Please ask your administrator.
(I tried to install it with sudo before the command as well, but it returned a different error)

Python command resulted in this
Command 'python' not found, did you mean:
  command 'python3' from deb python3
  command 'python' from deb python-is-python3

However, grafanna resulted in a positive connection. 

When running the commands for weather411, I got the following output:
 Error response from daemon: cannot stop container: weather411: permission denied

Sorry, I'm sure there are a lot of beginner-level mistakes in all of this - I'm really new to all of this. Thanks so much for trying to help!

jasonacox, 2023-03-19:
> urllib3.exceptions.ConnectTimeoutError

This means that pypowerwall is not able to connect to your Powerwall.  A few things to try

```bash
# Ping your Powerwall from the Ubuntu system - change the IP to the address of the Powerwall
ping 10.0.1.x

# Install python
sudo apt install python3
sudo apt install python3-pip
sudo pip install pypowerwall
python -m pypowerwall scan
```

> Error response from daemon: cannot stop container: weather411

I wonder if you have docker permission on your user account. You can try this:

```bash
# Add your user to docker group
sudo usermod -aG docker $USER

# reboot or log out and log back in
exit
```

The biggest issue I see is the connectivity (pypowerwall not seeing the Powerwall) which of course does cause problems with other containers also needing connectivity.