jasonacox/Powerwall-Dashboard #221: Remove Grafana login requirement (optional)

jasonacox, 2023-03-21:
Since most of us use this grafana dashboard for our home system, there is very little need to create a login requirement.  If you want to remove that requirement, you can add these lines to the `grafana.env` file and recreate grafana with `./compose-dash.sh up -d`:

```bash
# Disable login requirement
GF_AUTH_DISABLE_LOGIN_FORM=true
GF_AUTH_ANONYMOUS_ENABLED=true
GF_AUTH_ANONYMOUS_ORG_NAME=Main Org.
GF_AUTH_ANONYMOUS_ORG_ROLE=Admin
GF_USERS_ALLOW_SIGN_UP=false
```

I'm opening the issue to implement this as an optional setup.  I'm going to add it to the base installation but commented out.

jasonacox, 2023-03-21:
Or should we make this the DEFAULT?

mp09, 2023-03-21:
Generally, I think it's better to make loosened security an opt-in item. it's pretty standard that if you want to create a no-auth setup that isn't the default and you have to go out of the way to get it. (not sure I personally have a concern with data that could be exposed here, but it feels non-standard.)

dkerr64, 2023-03-21:
I agree with @mp09 as I feel it a bit dangerous to allow edit/write access without authentication.  I do want view access without authentication which was addressed in #130 
Thanks

jasonacox, 2023-03-22:
Thanks all! I'm leaving this as an opt-in (uncomment) as part of [grafana.env.sample](https://github.com/jasonacox/Powerwall-Dashboard/blob/0b8396adc2a87cfc3f4dfe6df7fb512ebb9a96da/grafana.env.sample#L11-L22).

closed 2023-03-22 by jasonacox