jasonacox/Powerwall-Dashboard #224: Power meter reverse proxy

dngray, 2023-03-23:
Hi,

I'm currently using [Powerwall-Dashboard on Docker](https://github.com/dngray/containers/blob/main/compose/docker/powerwall/powerwall.yml#L69) behind a [Traefik reverse proxy](https://doc.traefik.io/traefik/). Configuration is [here](https://github.com/dngray/containers/blob/main/compose/docker/traefik/compose.yml).

My configuration works fine everywhere in all my [other services](https://github.com/dngray/containers/tree/main/compose/docker). The only problem I've been having is with the Power Meter when I enabled HTTPS. The frame shows:

> Secure Connection Failed
> 
> An error occurred during a connection to grafana.my_host.home.arpa:8675. SSL received a record that exceeded the maximum permissible length.
> 
> Error code: SSL_ERROR_RX_RECORD_TOO_LONG

Note I'm not using HTTPS for Grafana. I've got Traefik running in front of all that as a reverse proxy so that my server can use a single port 443 for all services.

When I disabled HTTPS on Traefik the power meter worked.

Any ideas how I might fix this?

jasonacox, 2023-03-23:
Hi @dngray - Yes the iFrame creates some fun challenges with https.  Have you tried a different browser?  Chrome, Safari, etc, may give a different error but from a quick search, that error is unique to Firefox.

Also, take a look at this discussion: https://github.com/jasonacox/Powerwall-Dashboard/discussions/215 This is for [lightttpd](https://www.lighttpd.net/) instead of [Traefik reverse proxy](https://doc.traefik.io/traefik/) but might be helpful.

dngray, 2023-03-23:
Sure I tried in Chromium and just got

> **grafana.my_host.home.arpa** sent an invalid response.

I will check out those other threads.

dngray, 2023-11-11:
Solution here: https://github.com/jasonacox/Powerwall-Dashboard/discussions/215#discussioncomment-7540307

closed 2023-11-11 by dngray