jasonacox/Powerwall-Dashboard #229: Powerwall temp set to Celsius ignoring config

VenomrushCT, 2023-03-30:
Despite config is set in Imperial, the Powerwall temp on the default Dashboard is always in Celsius.

jasonacox, 2023-03-31:
Hi @VenomrushCT - thanks for opening this.  The metric/imperial config is only related to the weather data. The Powerwall temp data is only available in Celsius.  I'll add a note to update the setup so that it is clear that it is only for weather data.  

However, there is a way to have Grafana convert the Powerwall temp data to Fahrenheit:

[image]

VenomrushCT, 2023-03-31:
Thanks!

Added math(* 1.8 + 32) and changed Y axis label.

closed 2023-03-31 by VenomrushCT