jasonacox/Powerwall-Dashboard #230: Add per string total string power generated.

brianmsailing, 2023-03-31:
It would be great to see the total per string power generated for the time period on display as a numerical table tally.

jasonacox, 2023-04-02:
Hi @brianmsailing !  Thanks for this suggestion. I'll add this as an enhancement request.  However, the challenge is that there are different numbers of strings per installation (I have 3 but I know others who have 8).  Currently the graphs support up to 16 but if the data is not there, it doesn't show. We would want something similar for the numerical table.  

Do you have some idea or have you attempted to add a panel to explore how we could represent that?

brianmsailing, 2023-04-06:
Jason, thank you. Great development and amazing work by your contributors.  My suggestion was potentially simpler than I might have explained. I have not attempted to add a panel as my software development skills are not sufficient to make any changes.  My visions would be as you described a table that contains a power tally in watt's that added up the area under the curve for the displayed time adding up the total watts generated for a specific string allowing comparison between the strings for the given time.

jasonacox, 2023-04-07:
I think I might know where you are going... I know this isn't exactly where you want to go, but this graph basically stacks all the strings in a per-day bar graph and creates totals to the right based on the duration (this is last 30 days):

[image]

Last 1 year:
[image]

You set up each string like this:

[image]

brianmsailing, 2023-04-07:
thank you, I was able to create based on your guidance 

SELECT integral("A_Power")  /1000/3600 FROM "strings"."http" WHERE $timeFilter
[image]

jasonacox, 2023-04-08:
Awesome! Nice job @brianmsailing 

One minor note, I see the labels "Front STring KW" - but if you are using `integral()` you are actually getting the product of energy over a period of time (kW * time) = kWh.

closed 2023-04-11 by brianmsailing