jasonacox/Powerwall-Dashboard #231: Git pull fails because setup process modifies files under change control

dkerr64, 2023-03-31:
I get the following error message...
```
david@powerwall:~/Powerwall-Dashboard$ git pull
Updating a2e5e29..0f8468b
error: Your local changes to the following files would be overwritten by merge:
	dashboard.json
	dashboards/dashboard.json
Please commit your changes or stash them before you merge.
Aborting
```

This is caused by the setup process making modifications to local copies of these files (in my case changing every "America/Los_Angeles" to "America/New_York"

This is bad practice.  There should be default versions of these files in GIT and the setup process should not modify them, rather the modified files should be given a new name.

I suspect that there are several other files that may be similarly impacted, just have not come across them yet because changes have not been made to the default versions.

jasonacox, 2023-04-01:
HI @dkerr64 - that's correct. It gets messy and worse, it makes it very complex to get a clean "upgrade" if you need to process all the files every time. For everything we could, the files are treated as you say and copying a localized versions of the file. This works for most situations as those files only contain localized data and are not needing to be upgraded (e.g. `pypowerwall.env`).  Others, like `dashboard.json`, are intended to be upgraded since the only customization is timezone.  

The `upgrade.sh` script was built to help address this.  It specifically reverses the timezone changes before `git pull`:

```bash
# Remember Timezome and Reset to Default
echo "Resetting Timezone to Default..."
DEFAULT="America/Los_Angeles"
TZ=`cat tz`
if [ -z "${TZ}" ]; then 
    TZ="America/Los_Angeles"
fi
./tz.sh "${DEFAULT}"

# Pull from Github
echo ""
echo "Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes..."
echo ""
git stash
git pull --rebase
echo ""

...

# Set Timezone 
echo ""
echo "Setting Timezone back to ${TZ}..."
./tz.sh "${TZ}"
```

dkerr64, 2023-04-01:
So you are saying that I should not use git pull, but instead use the upgrade.sh script?  That is a workable work-around I suppose.

Thanks
David

jasonacox, 2023-04-01:
I confess, I don't love the upgrade process. It would be clean if it were only git pull but the changes we make often include things like InfluxQL updates which the upgrade.sh script accommodates by running those queries. It also pulls the latest container updates and restarts the stack. In other words,  a git pull would require some manual work an update script anyway.

For now, this methods seems to work well for most of the community.  Key goal is to make it easy to install and upgrade.  BUT, I'm always open to better ways.  🙏 

Thanks for opening this issue. I do think it highlights the need for better instructions. I'll flag this for documentation.

jasonacox-sam, 2026-09-06:
This flagged that `git pull` fails because setup modifies version-controlled files (dashboard.json timezone edits). The agreed answer: use `upgrade.sh` rather than raw `git pull` — it handles localized files, InfluxQL schema updates, and container refreshes in one step. The reporter accepted the workaround.

Closing as resolved (documented workflow: `upgrade.sh`). The upgrade tooling has been further improved since — see [v5.2.0](https://github.com/jasonacox/Powerwall-Dashboard/releases/tag/v5.2.0) and [v5.2.1](https://github.com/jasonacox/Powerwall-Dashboard/releases/tag/v5.2.1).

— Sam 🌊

closed 2026-09-06 by jasonacox-sam