jasonacox/Powerwall-Dashboard #234: Minimize Grafana Plugins

jasonacox, 2023-04-03:
## Proposal

As @ThePnuts has pointed out, there are several Grafana plugins in the default install that are no longer used.  I suggest that we remove these vestigial plugins to minimize the startup time for Grafana.  Since plugins for Grafana are specified in the localized `grafana.env` file, this will not impact existing installations (you will need to manually prune), but all new installs will have only the plugins required for the Powerwall Dashboard.

## Current Plugins

* grafana-piechart-panel
* natel-plotly-panel
* blackmirror1-singlestat-math-panel
* btplc-trend-box-panel
* marcuscalidus-svg-panel
* michaeldmoore-multistat-panel
* yesoreyeram-boomtable-panel (forced [updated version](https://github.com/yesoreyeram/yesoreyeram-boomtable-panel/releases/tag/v1.5.0-alpha.3) to fix "no data" bug)
* ryantxu-ajax-panel
* grafana-influxdb-flux-datasource
* fetzerch-sunandmoon-datasource
* andig-darksky-datasource
* simpod-json-datasource

## Proposed
* grafana-piechart-panel
* yesoreyeram-boomtable-panel (forced [updated version](https://github.com/yesoreyeram/yesoreyeram-boomtable-panel/releases/tag/v1.5.0-alpha.3) to fix "no data" bug)
* fetzerch-sunandmoon-datasource

## grafana.env.sample

```bash
# Existing 
GF_INSTALL_PLUGINS=grafana-piechart-panel, natel-plotly-panel, blackmirror1-singlestat-math-panel, btplc-trend-box-panel, marcuscalidus-svg-panel, michaeldmoore-multistat-panel, https://github.com/yesoreyeram/yesoreyeram-boomtable-panel/releases/download/v1.5.0-alpha.3/yesoreyeram-boomtable-panel-1.5.0-alpha.3.zip;yesoreyeram-boomtable-panel, ryantxu-ajax-panel, grafana-influxdb-flux-datasource, fetzerch-sunandmoon-datasource, andig-darksky-datasource, simpod-json-datasource

# Proposed
GF_INSTALL_PLUGINS=grafana-piechart-panel, https://github.com/yesoreyeram/yesoreyeram-boomtable-panel/releases/download/v1.5.0-alpha.3/yesoreyeram-boomtable-panel-1.5.0-alpha.3.zip;yesoreyeram-boomtable-panel, fetzerch-sunandmoon-datasource
```

BuongiornoTexas, 2023-04-03:
Can I ask that we keep simpod json for now. I'm using it in the ToU system I'm slowly pulling together and I'm moderately hopeful this will be generally useful.

jasonacox, 2023-04-03:
Absolutely!  Are there any others that we would have a compelling case to use in the future?  I would prefer to err on including more than we need as long as we have a case for them, as it it will be headache to upgrade later (will need to add some logic to upgrade.sh to add them to the local grafana.env).

BuongiornoTexas, 2023-04-03:
In which case, its also probably worth keeping the flux client on the basis that it's going to make sense to go to InfluxdB 2.0 at some point (@ThePnuts  can't be an island forever!). 

(I'm already using basic flux queries in the ToU system via the old InfluxDB datasource - compared to InfluxQL, it is just so much nicer to work with for basic queries. Completely horrible for complex stuff, but then I can't even start on the complex stuff with InfluxQL)

ThePnuts, 2023-04-03:
Proposed list looks great.

Might be for another discussion, but the [grafana-piechart-panel is depreciated](https://github.com/grafana/piechart-panel) and has been replaced with a new version as of Grafana 8.0. Its replacement is part of Grafana Core and auto installed with Grafana.

Old:
[image]

Current:
[image]

ThePnuts, 2023-04-03:
I believe this is the live dashboard by default right? https://github.com/jasonacox/Powerwall-Dashboard/blob/main/dashboards/dashboard.json

This plugin update should be seamless for everyone since the new version is part of core and auto installed. If anyone wants to test this updated dashboard, it should just work using the new plugin, just looks like it lost a tiny bit of weight.

[Powerwall Power Flow.json.zip](https://github.com/jasonacox/Powerwall-Dashboard/files/11142701/Powerwall.Power.Flow.json.zip)

Are all of the dashboards there part of the normal install? I can do the rest of them as well if its just this simple.

jasonacox, 2023-04-07:
Thanks @ThePnuts !  I know this is going to sound crazy, but I'm not a fan of the new piechart. They add way to much blank space and the donut is off balance/alignment with the text. It drives me crazy (yes, I have OCD issues).  I have been on a hunt to figure out how I can adjust the padding or render locations on it, but it doesn't seem to offer up as many parameters as the old graph.  I should probably find he code and submit a PR to grafana.

In any event, thanks for helping sort out what we really need in terms of plug-ins.  I think we are getting down to a good list.

ThePnuts, 2023-04-07:
Ha, no worries @jasonacox, I like the older version better tbh. My OCD issue is wanting to be on the latest/newest version of everything, hence why I stuck with re-creating everything in Flux so I could be. I've been busy making other Flux dashboards for my server and such as well, so I learned a bit more since, but still no way to fatten up the donut.

PR #246 (2023-04-15): 2.9.1
# Release v2.9.1

* Minimize the number of default Grafana plutins as defined in the `grafana.env.sample` file (installed as localized `grafana.env`) for new installations by @jasonacox in https://github.com/jasonacox/Powerwall-Dashboard/issues/234
* Update power flow animation panel code to display Grafana loading image while power flow graphics are loaded by @dkerr64 as discussed in https://github.com/jasonacox/Powerwall-Dashboard/discussions/216#discussioncomment-5573094

Closes #234
closed 2023-04-15 by jasonacox