jasonacox/Powerwall-Dashboard #235: Podman Support

oneguynick, 2023-04-06:
**Problem**

I have slowly migrated most of my personal container usage to podman. My last remaining stack was Powerwall. The transition was pretty simple, given the project's aim for rootless docker containers. 

**Enhancement**

Ideally, the setup code would account for both docker and podman environments. My edits were, in most cases, as simple as sed docker for podman. I additionally needed to chown some directories to account for the permissions within podman, but outside of that and small QOL script updates for the compose, it worked first try.

**Additional context**

You can see the work here: https://github.com/oneguynick/Powerwall-Dashboard

jasonacox, 2023-04-07:
Thanks @oneguynick !

Here is an option on how we can add "podman" support.

I suggest that we change all the scripts to check for and pull in a `PLATFORM` file that contains either “docker” or “podman”.  This file would NOT exist in the repo but would be created during setup:

During `setup.sh`: 

- 	Add logic to determine if podman or docker are available - if both available, let user decide with a prompt, if only one is available, use that. Default: docker.
- 	Create a file `PLATFORM` in install directory to be used by other scripts (contains only “docker” or “podman”)
- 	Replace all references to “docker” command with “${PLATFORM}"

For `upgrade.sh`, `compose-dash.sh`, `verify.sh`, `weather.sh`, etc.

- 	Add logic:  If `PLATFORM` file exist, set platform variable to what is in that file. If it does not exist, default to “docker”.
- 	Replace all references to “docker” with “${PLATFORM}”

The advantage of the above approach is that existing installations (all docker) would not be impacted as docker will be default. All new installs will have option to pick between docker or podman (if installed).

@oneguynick we would need to add logic for the podman-only things you had to do. 

Community: concerns, thoughts, ideas?

dkerr64, 2023-04-10:
Interesting idea. This might make worthwhile reading...

https://www.redhat.com/en/topics/containers/what-is-podman

That makes me wonder whether podman should be the default rather than docker?

jasonacox-sam, 2026-09-06:
This proposed Podman support as an alternative to Docker, with @oneguynick reporting that a sed-swap of docker→podman plus some chown fixes worked, and Jason sketching a `PLATFORM` file approach for setup.sh. No implementation PR was ever submitted, and Docker remains the supported platform.

Closing due to inactivity. If Podman support still matters to you and you're willing to drive the PLATFORM-file refactor, please re-open with a PR plan — the design sketch in this thread is still a solid starting point.

— Sam 🌊

closed 2026-09-06 by jasonacox-sam