jasonacox/Powerwall-Dashboard #237: upgrade.sh's image removal causes conflicts with existing third party containers sharing same dependency

GrimmiMeloni, 2023-04-09:
I am just going through my first upgrade of the Dashboard.
I am running into problems as the removal of the images (`docker rmi` step) fails.
I am using some of the images also in different projects (in my case it is telegraf). So it is not simply possible to perform the docker image removal like the upgrade script tries to because of these dependencies.

I have worked around this, by removing (and recreating) my other projects, but this feels very invasive. In a nutshell the `upgrade.sh `in its current form assumes exclusivity, which it should not.
I would recommend a different approach for the upgrade, like for example using an explicit `docker pull telegraf:latest` instead.

jasonacox, 2023-04-09:
Hi @GrimmiMeloni ! Thanks for opening this. You are right!  It would be good to make this change. Are you interested in submitting a PR to fix it?  I'll tag this as a bug.

GrimmiMeloni, 2023-04-10:
Hey Jason, happy to help.
Before I start hacking away, let's agree on what the end result should look like.

I would suggest the following solution:

1. remove the image removal steps out of upgrade.sh. My rationale is both the problem stated in my initial report, but also after more thought, I think Powerwall-Dash even without such a problem, should not touch the local image registry and toss things away. I think it should be left to the system's operator to decide when a good time for pruning has come.
 
2. Change the compose file, to move away from using `latest` images, and use explicit versioning of the dependencies. It is anyways good style (repeatable builds/installs), and will also ensure that all users of Powerdash have the same experience. A good summary on all the shenanigans using latest bring can be found [here](https://vsupalov.com/docker-latest-tag/). 
This change removes the need for explicitly tossing the local images away to fetch something "fresh". Compose can take care of pulling the specified images.
I suggest the following explicit versions, based on what my local host pulled during yesterday's upgrade.

- Telegraf: 1.26.1
- pypowerwall: 0.6.2t25
- weather411: 0.2.1

jasonacox, 2023-04-11:
> 1. remove the image removal steps out of upgrade.sh

Agree!  Plus, the current upgrade is way too slow, mostly due to this image churn.

> 2. Change the compose file, to move away from using latest images

Agree!  This is brilliant. This makes the version of Powerwall-Dashboard consistent.  We should rev the project version (in files `VERSION` and `upgrade.sh`)  if we change any of the pins. As you pointed out so well, current state makes the project version meaningless and everyone is potentially using a different combination. That makes it difficult to troubleshoot and manage.

Thank you for your help @GrimmiMeloni !  🙏 This will be a great enhancement.

jasonacox, 2023-04-12:
Thanks @GrimmiMeloni !  Merged and released as v2.9.0.

Please report any issues.

jasonacox-sam, 2026-09-06:
This reported that `upgrade.sh`'s `docker rmi` cleanup broke setups where images (e.g. telegraf) were shared with other projects. @GrimmiMeloni proposed removing the image-removal step and pinning image versions in compose — both accepted, implemented as PR #240, and shipped in [v2.9.0](https://github.com/jasonacox/Powerwall-Dashboard/releases/tag/v2.9.0).

Closing as resolved.

— Sam 🌊

closed 2026-09-06 by jasonacox-sam